SpinalHDL/SpinalHDL #505: Mem Black Box discuss

jijingg, 2021-08-30:
@Dolu1990 , The current MEM blackbox looks like several fixed interface forms(`Ram_1w_1ra` ,`Ram_1w_1rs`,`Ram_1wrs`,`Ram_2wrs`). It may not be suitable for ASIC design
Because of different teams or companies, MEM wrapper has different moudule names and interface names.
And The same HDL may use different memwrap on FPGA or ASIC, or different Tech of ASIC(UMC16nm/TSMC8nm ... )
Of course, spinal cannot make all vendor MEM wrappers or future MEM wrappers into a blackbox.

[image]
So we should design a way to easily replace memwrap. 
**There are two principles in this:**
- **User-transparent**(The replacement of the memwrap should not modify the HDL code. Vendor parameter is recommended)
- **Forward compatibility**( Adding a new vendor memwrap should not modify the spindle source code )
 
[image]

so it is hoped that spinal will provide a unified and simple MEM API to select specific blackbox implementations by defining vendors

now we provide a common 3 ram API and 1 rom API for users . 
The following is my specific practice, which has worked well in my local project . hope you can provide some advice. 

https://github.com/jijingg/SpinalHDL/tree/ram/lib/src/main/scala/spinal/lib/memory/ram/blackbox 
```scala
case class Ram1rw(mc: MemConfig, memName: String = "") extends Component with MemWrap{
    ...
    val ram = mc.vendor.build(this)  // here the Ram1rw will call vendor.build() method auto connect port which vendor or users  defined
    ...
}
case class Ram1r1w(mc: MemConfig, memName: String = "") extends Component with MemWrap{...}
case class Ram2rw(mc: MemConfig, memName: String = "") extends Component with MemWrap{...}
case class Rom(mc: MemConfig, memName: String = "") extends Component with MemWrap{...}
```
Usage: 
```scala
  class Top(vendor: Vendor) extends Component{
//    this.setDefinitionName(s"Top${vendor}")
    val io = new Bundle{
      val bus = slave(sPort(8, Bits(32 bit)))
    }

    val ram = Ram1r1w(MemConfig(32, 256, vendor = vendor))
   
    ram.io.wr     := io.bus.wr
    ram.io.waddr  := io.bus.addr
    ram.io.wdata  := io.bus.wdata
    ram.io.rd     := io.bus.cs && !io.bus.wr
    ram.io.raddr  := io.bus.addr
    io.bus.rdata  := ram.io.rdata
}
```
now wen can add the spinal exists mem wrapper 
```
case object SpinalMem extends Vendor{
  override def build(mw: Ram1rw): MemBlackBox = new spinalmem.Ram_1w_1rs(mw).Build()
  override def build(mw: Ram1r1w): MemBlackBox = new spinalmem.Ram_1wrs(mw).Build()
  override def build(mw: Ram2rw): MemBlackBox = new spinalmem.Ram_2wrs(mw).Build()
  override def build(mw: Rom): MemBlackBox = new spinalmem.Rom(mw).Build()
}
```
Other Vendor or  Users define blackbox themselvs.
```scala
trait Vendor {
  // umc mem_wrapper as default vendor 
  def build(mw: Ram1rw): MemBlackBox = new umc.mbb1rw(mw).Build()
  def build(mw: Ram1r1w): MemBlackBox = new umc.mbb1r1w(mw).Build()
  def build(mw: Ram2rw): MemBlackBox = new umc.mbb2rw(mw).Build()
  def build(mw: Rom): MemBlackBox = new umc.mbbrom(mw).Build()
}

case object Intel extends Vendor{
  override def build(mw: Ram1rw): MemBlackBox = new intel.mbb1rw(mw).Build()
  override def build(mw: Ram1r1w): MemBlackBox = new intel.mbb1r1w(mw).Build()
  override def build(mw: Ram2rw): MemBlackBox = new intel.mbb2rw(mw).Build()
  override def build(mw: Rom): MemBlackBox = new intel.mbbrom(mw).Build()
}
//For different companies and different teams, different wrappers can be defined on the premise of extensions vendor
case object SPRD extends Vendor{...}
case object FishSemi extends Vendor{...}
case object AMD extends Vendor{...}
case object HuaWei extends Vendor{...}
case object HuaWei10nm extends Vendor{...}
case object ZTE extends Vendor{...}
case object Invida extends Vendor{...}
....
```

Dolu1990, 2021-08-31:
Hi,

I agree with the chalange, so there was the initial plan : 

Default blackboxing is just a default. People could create their own variation of the https://github.com/SpinalHDL/SpinalHDL/blob/c8234fadf6bd7cfecc3faead198144098f35fc49/core/src/main/scala/spinal/core/internals/Phase.scala#L717 class to transform the SpinalHDL tree and add blackbox to match their synthesis backend.

We could imagine another implementation of that blackboxification phase which would implement a aproache as you suggest.

One thing i'm not realy for is "val ram = Ram1r1w(MemConfig(32, 256, vendor = vendor))"

I would realy prefer allowing the user to keep using the memory agnostic Mem class. and then have a phase to blackboxify them, eventualy using something like the Vendor trait.

How does that sound ?

sebastien-riou, 2021-09-11:
In my use case, I needed to add additional control bits to handle sleep mode and test mode. I also added ROM black box.
https://github.com/Tiempogithub/SpinalHDL
For ASIC, I replace the inside of the following generic description with an instance of the vendor specific macro 
[generic_memories.zip](https://github.com/SpinalHDL/SpinalHDL/files/7147654/generic_memories.zip)

sebastien-riou, 2021-09-11:
In the long run, the ideal would be to allow a mem vendor to provide a set of files which add support for its product. Within one project, it shall be possible to use memories from different vendors "at the same time" (meaning that the resulting ASIC may include one memory from vendor A and another one from vendor B)

jijingg, 2021-09-11:
> sleep mode and test mode

@sebastien-riou  i agree, some test mode or BIST signal is also a very important topic，In the traditional IC flow, it is usually inserted through script or manually in the later stage. These signals are also independent of mem function, but related to the back-end flow.
so we should give a simple and safe way to insert those signals.   

**There are two challenges：**

- First, how  to  make BIST or test signal cross multiple hierarchies to the top
- Second: how to safely merge BIST logic of multiple mems into one path in every hierarychies.

I have no very clear idea how to implement in a elegant way right now . This is also the reason why my MEM implementation has not submitted PR for a long time.

>it shall be possible to use memories from different vendors "at the same time" 

yes, These are to be taken into account

Dolu1990, 2021-09-13:
@sebastien-riou @jijingg 

> In my use case, I needed to add additional control bits to handle sleep mode and test mode

One thing which is fesable to handle vendor specific stuff, is to add a SpinalHDL compilation phase to rework the blackboxed memories. The  that way you can progamaticaly manage all those specific things, including brining all the test wire to a specific spot in the design.

Would that be a good way to handle those cases ?

> First, how to make BIST or test signal cross multiple hierarchies to the top

somethwere.deep.in.hearchy.mySignal.pull()   <3

sebastien-riou, 2021-09-14:
> One thing which is fesable to handle vendor specific stuff, is to add a SpinalHDL compilation phase to rework the blackboxed memories. The that way you can progamaticaly manage all those specific [...]

Sounds good, things to consider:
- Does it improves over my quick and dirty approach, for example: get the name of vendor specific control signals rather than just raw bits
- It shall not add complexity from user point of view (users which just target FPGA) 
- It shall not add new error case that are hard to debug

Dolu1990, 2021-09-15:
> It shall not add complexity from user point of view (users which just target FPGA)

So, they will not be impacted, as that additional phase would be something specific to the people which need the feature. I was thinking about a compilation phase that would not be there by default.

> Does it improves over my quick and dirty approach, for example: get the name of vendor specific control signals rather than just raw bits

I'm not sure to understand, "rather than just raw bits" ?

> It shall not add new error case that are hard to debug

I would say it won't add hard to debug bugs, as if there is a bug, that bug would be very reproducable.

sebastien-riou, 2021-09-15:
> I'm not sure to understand, "rather than just raw bits" ?

I mean that I put all bits into a signal "miscIn", so I do not provide meaningful names. It would be better to be able to define a bundle.

> I would say it won't add hard to debug bugs, as if there is a bug, that bug would be very reproducable.

My concern is about things liker fiber and Handles: errors are reproducable but most people don't have a clue on how to fix them. 

Anyway I realize now that the ability to add things on all memories in a design without modifying the blocks that instanciate them would be great. Some memories have "triming" values that shall be set depending on power supply voltage for example. With such memories, it is mandatory to drive those signals. So for example the RAM for the cache in VexRisc shall have such controls... If adding a compilation phase allow to support this without modifying VexRisc code, that's  worth the effort in my opinion.

Dolu1990, 2021-09-17:
> I mean that I put all bits into a signal "miscIn", so I do not provide meaningful names. It would be better to be able to define a bundle.

Hooo right, to keep things clean.

> My concern is about things liker fiber and Handles: errors are reproducable but most people don't have a clue on how to fix them.

Yes they maybe hard to understand bugs, especialy for people not used with how to manipulate the netlist API. but thing is, Handles are to be manipulated by users constantly, while the Mem blackboxing stuff is more a "implement once, reuse many times".

> So for example the RAM for the cache in VexRisc shall have such controls... If adding a compilation phase allow to support this without modifying VexRisc code, that's worth the effort in my opinion.

Right, automation would be better than handworking all the time :)

Dolu1990, 2021-10-30:
Here is an example which replace all Mem which have multiple write port into XOR based rams : 
https://github.com/SpinalHDL/NaxRiscv/blob/main/src/main/scala/naxriscv/compatibility/MultiportRam.scala
(a few assertion / if filter are missing, ex being sure there is now readSync port, stuff like this, but that's the idea)

wswslzp, 2021-12-06:
Would you like to provide some doc about the netlist API mentioned above?

Dolu1990, 2021-12-07:
The netlist API =>  All the low level SpinalHDL functionalities which doesn't have lipstick ^^
So it is very large, there is not much doc about it :
https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Developers%20area/spinalhdl_datamodel.html#spinalhdl-internal-datamodel

For instance, here is that custom phase which wrap all ram which have multiple writes into a xor based one : 
https://github.com/SpinalHDL/NaxRiscv/blob/main/src/main/scala/naxriscv/compatibility/MultiportRam.scala#L105

wswslzp, 2021-12-07:
Thanks, It's really helpful to better understand the core of SpinalHDL!

It turns out that customized phases are a bit like customized FIRRTL transformations in Chisel. Developers can make their own phases to replace some existing scripts in the current development flow, e.g. scripts of inserting the wire connection of BIST port of SRAM they mentioned above if we have more detailed doc and examples about the netlist API. This potential of SpinalHDL is also amazing!

Dolu1990, 2021-12-07:
> It turns out that customized phases are a bit like customized FIRRTL transformations in Chisel.

Right, the main diference at that level maybe that SpinalHDL has a mutable graph while FIRRTL internals has a immutable one.

Then another aspect on which i'm not realy sure about Chisel is about the user modifying the netlist directly durring the user elaboration time, so modifying stuff without the need of FIRRTL custom physes. 
In SpinalHDL :
https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Developers%20area/spinalhdl_datamodel.html#modifying-a-netlist-as-a-user-without-plugins
There is also tools to look for the latency of some paths (durring user elaboration time too) : 
https://github.com/SpinalHDL/SpinalHDL/blob/b61b3039ae41264fb0059e36e6be13b28aac9738/lib/src/main/scala/spinal/lib/Utils.scala#L659