SpinalHDL/SpinalHDL #507: Null pointer access with inout in blackboxes

dnltz, 2021-09-05:
Hi  @Dolu1990,

I made a design which includes blackboxes for some Artix7 IO. The outgoing pins are defined as `inout(Analog(Bool))`. In a next step I want to define a simulation for my board which includes the design as blackbox. The goal for this approach is to allow to add either the generated or synthesized verilog file.

This is a minimal demo of my code:
```
package trigger

import spinal.core._
import spinal.core.sim._
import spinal.lib._

object Board {
  def apply() = Board()

  def main(args: Array[String]) {

    val config = SpinalConfig(noRandBoot = false)
    val compiled = SimConfig.withConfig(config).withWave.compile {
      val board = Board()
      board
    }
    compiled.doSimUntilVoid("board") { dut =>

      fork {
        dut.io.foo #= true
        sleep(10)
        dut.io.foo #= false
      }

    }
  }

  case class Top() extends BlackBox {
    val io = inout(Analog(Bool))
  }

  case class Board() extends Component {
    val io = new Bundle {
      val foo = in(Bool)
    }
    val top = Top()
    //top.io := foo
    top.io := False
  }
}
```
Well, if I want to run this code I receive a null pointer access which will not occur when I remove `top.io := ...`. Moreover, If I change the `inout` to `in`, the error will not occur, too.

Do you have any idea why this error occurs?

Error log:
```
$ sbt "runMain trigger.Board"
[info] welcome to sbt 1.4.4 (Private Build Java 14.0.2)
[info] loading project definition from /home/daniel/work/elements-sdk/zibal/project
[info] loading settings for project root from build.sbt ...
[info] loading settings for project vexriscv-build from plugins.sbt ...
[info] loading project definition from /home/daniel/work/elements-sdk/vexriscv/project
[info] loading settings for project root from build.sbt ...
[info] set current project to Zibal (in build file:/home/daniel/work/elements-sdk/zibal/)
[info] compiling 1 Scala source to /home/daniel/work/elements-sdk/zibal/target/scala-2.11/classes ...
[info] running (fork) trigger.Board 
[info] [Runtime] SpinalHDL v1.6.0    git head : 73c8d8e2b86b45646e9d0b2e729291f2b65e6be3
[info] [Runtime] JVM max memory : 7996.0MiB
[info] [Runtime] Current date : 2021.09.05 09:41:06
[info] [Progress] at 0.000 : Elaborate components
[info] [Progress] at 0.130 : Checks and transforms
[info] asd
[info] ERROR !
[info] A null pointer access has been detected in the JVM.
[info] This could happen when in your SpinalHDL description, you access an signal which is only defined further.
[error] Exception in thread "main" java.lang.NullPointerException
[info] For instance :
[info]   val result = Bool()
[info]   result := a ^ b  //a and b can't be accessed there because they are only defined one line below (Software rule of execution order)
[info]   val a,b = Bool()
[info]           
[error] 	at spinal.core.internals.PhaseCheckCombinationalLoops.spinal$core$internals$PhaseCheckCombinationalLoops$$walk$2(Phase.scala:1338)
[error] 	at spinal.core.internals.PhaseCheckCombinationalLoops$$anonfun$spinal$core$internals$PhaseCheckCombinationalLoops$$walk$2$5.apply(Phase.scala:1365)
[error] 	at spinal.core.internals.PhaseCheckCombinationalLoops$$anonfun$spinal$core$internals$PhaseCheckCombinationalLoops$$walk$2$5.apply(Phase.scala:1365)
[error] 	at spinal.core.internals.Statement$class.walkParentTreeStatementsUntilRootScope(Statement.scala:239)
[error] 	at spinal.core.internals.AssignmentStatement.walkParentTreeStatementsUntilRootScope(Statement.scala:301)
[error] 	at spinal.core.internals.PhaseCheckCombinationalLoops.spinal$core$internals$PhaseCheckCombinationalLoops$$walk$2(Phase.scala:1365)
[error] 	at spinal.core.internals.PhaseCheckCombinationalLoops$$anonfun$impl$34.apply(Phase.scala:1386)
[error] 	at spinal.core.internals.PhaseCheckCombinationalLoops$$anonfun$impl$34.apply(Phase.scala:1384)
[error] 	at spinal.core.internals.ScopeStatement$$anonfun$walkStatements$1.apply(Statement.scala:142)
[error] 	at spinal.core.internals.ScopeStatement$$anonfun$walkStatements$1.apply(Statement.scala:141)
[error] 	at spinal.core.internals.ScopeStatement.foreachStatements(Statement.scala:136)
[error] 	at spinal.core.internals.ScopeStatement.walkStatements(Statement.scala:141)
[error] 	at spinal.core.internals.PhaseContext$$anonfun$walkStatements$1.apply(Phase.scala:129)
[error] 	at spinal.core.internals.PhaseContext$$anonfun$walkStatements$1.apply(Phase.scala:129)
[error] 	at spinal.core.internals.GraphUtils$.walkAllComponents(Misc.scala:262)
[error] 	at spinal.core.internals.PhaseContext.walkStatements(Phase.scala:129)
[error] 	at spinal.core.internals.PhaseCheckCombinationalLoops.impl(Phase.scala:1384)
[error] 	at spinal.core.internals.PhaseContext.doPhase(Phase.scala:181)
[error] 	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2$$anonfun$apply$120.apply(Phase.scala:2531)
[error] 	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2$$anonfun$apply$120.apply(Phase.scala:2529)
[error] 	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error] 	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2529)
[error] 	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2466)
[error] 	at spinal.core.ScopeProperty$.sandbox(ScopeProperty.scala:32)
[error] 	at spinal.core.internals.SpinalVerilogBoot$.singleShot(Phase.scala:2466)
[error] 	at spinal.core.internals.SpinalVerilogBoot$.apply(Phase.scala:2436)
[error] 	at spinal.core.Spinal$.apply(Spinal.scala:355)
[error] 	at spinal.core.SpinalConfig.generateVerilog(Spinal.scala:161)
[error] 	at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:593)
[error] 	at trigger.Board$.main(trigger.scala:14)
[error] 	at trigger.Board.main(trigger.scala)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 4 s, completed Sep 5, 2021, 9:41:07 AM
```

Dolu1990, 2021-09-06:
Hi,

I looked at why it happen, basicaly, SpinalHDL create island of analog connection inside each component where there is analog stuff.
The issue happening with your case, is that it try to create the island in the blackbox itself.

So, there is a "easy" workaround if you have no island in the parent component : 

```scala
  case class Board() extends Component {
    val io = new Bundle {
      val foo = in(Bool)
    }
    val top = Top()
    val miaou = Analog(Bool)
    top.io := miaou
    miaou := io.foo
  }
```

Else i would say, just go inout still the whole way : 

```scala
  case class Board() extends Component {
    val io = new Bundle {
      val foo = inout(Analog(Bool))
    }
    val top = Top()
    top.io := io.foo
  }
```

The inout feature in SpinalHDL has its limitation, i'm not sure it is realy worth it to try fixing them, as inout isn't something that much important, as long there is a way to get it works. (without too much suffering)

dnltz, 2021-09-06:
Thanks, that explained it a little bit and helped! Actually, another issue was using Bool only. I need to create a `analogFalse` signal:
```
val analogFalse = Analog(Bool))
analogFalse := False
```
Will close this issue, thanks!

closed 2021-09-06 by dnltz