SpinalHDL/SpinalHDL #512: Register report latch error?

dxsm, 2021-09-15:
```scala
class top extends Component{
  val sel = Bits(2 bits)
  val b = Reg(Bits(8 bits))
  b := sel.mux(
    0 -> B"8'h30",
    1 -> B"8'h43",
    2 -> B"8'h47"
  )
}
```
As show in the example, b is a register, but encounter the error: 
```
LATCH DETECTED from the combinatorial signal (toplevel/??? :  Bits[8 bits]), defined at
    top.<init>(test.scala:7)
    main$$anonfun$main$1.apply(test.scala:16)
    main$$anonfun$main$1.apply(test.scala:16)
    spinal.sim.JvmThread.run(SimManager.scala:51)
```
I think register does not need default branch, and my desire rtl is
```verilog
always @(*) begin
    case(sel)
      2'b00 : begin
        b <= 8'h30;
      end
      2'b01 : begin
        b <= 8'h43;
      end
      2'b10 : begin
        b <= 8'h47;
      end
    endcase
  end
```
Could it be realized with `sel.mux` which I think is the shorter way than `switch`?

Dolu1990, 2021-09-15:
So, the behaviour SpinalHDL showed is the one i'm expecting.

The issue is that 

```scala
sel.mux(
    0 -> B"8'h30",
    1 -> B"8'h43",
    2 -> B"8'h47"
  )
 ```
 
 in itself create a combinatorial signal (which is the latch reported)
 
 And than your code take that signal (latch) and assign it to the register.
 
 You realy need to use the switch statement if you want to conditionaly assign a register (in your case, all the time but not in the case of sel === 3.
 
 if you don't realy care about the case were sel === 3, then 
 
 ```scala
   b := sel.mux(
    0 -> B"8'h30",
    1 -> B"8'h43",
    default -> B"8'h47"
  )
  ```
  
  That verilog code create a latch on b, right ? :
  
  ```verilog
  always @(*) begin
    case(sel)
      2'b00 : begin
        b <= 8'h30;
      end
      2'b01 : begin
        b <= 8'h43;
      end
      2'b10 : begin
        b <= 8'h47;
      end
    endcase
  end
  ```

dxsm, 2021-09-15:
Oh, my fault, that verilog code I desire should be with clock which not create a latch:
```verilog
always @(posedge clk) begin
  case(sel)
    2'b00 : begin
      b <= 8'h30;
    end
    2'b01 : begin
      b <= 8'h43;
    end
    2'b10 : begin
      b <= 8'h47;
    end
  endcase
end
```

You mean `sel.mux` always create a combinatorial signal  first even though the left is a register, then I think it can only be commonly limited to the use of combinatorial logic. Because sequential logic usually has no default branch for power saving purpose.

Dolu1990, 2021-09-17:
> You mean sel.mux always create a combinatorial signal first even though the left is a register,

Right

> then I think it can only be commonly limited to the use of combinatorial logic.

yes

> Because sequential logic usually has no default branch for power saving purpose.

Right

So, all good ^^ ?

dxsm, 2021-09-18:
Why not extend to sequential logic？`sel.mux` is a simple way than switch. How can I extend or implement this to sequential logic? Thanks in advance.

Dolu1990, 2021-09-18:
So, we could add another API to have conditional assignement as you would like it, but not with the keyword "mux". I think it is better to avoid having one thing behaving differently depending how you use its result.

What's about adding :

 ```scala
 myReg.swichAssign(sel)(
   0 -> B"8'h30",
   1 -> B"8'h43",
   2-> B"8'h47"
 )
 ```
 
  ?

dxsm, 2021-09-19:
yeah，that looks good. We do not need to write `myReg := ???` repeatedly.

Dolu1990, 2021-09-20:
Implemented ^^

```scala
      val sel = in UInt(8 bits)
      val r  = out(Reg(UInt(8 bits)))
      r.swichAssign(sel)(
        0 -> U(32),
        1 -> U(44),
        4 -> U(55),
        default -> U(66)
      )
```

closed 2021-09-20 by Dolu1990