SpinalHDL/SpinalHDL #515: BitVector.range returns confusing value

martonbognar, 2021-10-04:
Is it intentional that [BitVector.range](https://github.com/SpinalHDL/SpinalHDL/blob/dev/core/src/main/scala/spinal/core/BitVector.scala#L51) returns a range `[0, width)`?

I feel like based on the name, it would be more logical to return `[0, maxValue]` (e.g. when using `Counter.range`).

martonbognar, 2021-10-04:
Maybe my confusion stems from the [documentation](https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/miscelenea/core/elements.html#range). What do the columns "Description" and "Width" mean here?

Dolu1990, 2021-10-05:
yes it is the expected behaviour `[0, width-1]`

But i see what you mean.
I guess it was designed like this by analogy to VHDL (https://peterfab.com/ref/vhdl/vhdl_renerta/mobile/source/vhd00054.htm#:~:text=A%20range%20is%20descending%20if,otherwise%20the%20range%20is%20null).&text=A%20range%20can%20be%20undefined,is%20illustrated%20in%20example%202.)

> What do the columns "Description" and "Width" mean here?

Width mean number of bits

martonbognar, 2021-10-12:
I'm not sure I agree that the VHDL description justifies this behavior. It says "The range specifies a subset of values of a scalar type.". I would argue that the values of a bitvector are the numbers it can represent (so a 5-bit bitvector can represent numbers 0-31, which is the range I would expect to get).

Regarding the documentation, I expected to see something like `log2(y-x)` for the width, but I guess this is just related to the above, if the range means `lsb_index to msb_index`, then it makes sense.

martonbognar, 2021-10-13:
>  the only solution i see is adding an alternative explicite API 

I think that would be great! And maybe deprecating / updating the description of `range` to make sure people don't accidentally use the wrong one (I'm not sure what your approach usually is).

> what is it ? y x ?

To give you a concrete example, I just meant that if you have a `(Value)Range`, then I would expect the table in the documentation to look something like this:

Syntax | Description | Width
----------|-----------------|--------
(0 until 32) | [0:32[, 0 < 32 | ceil(log2(32 - 0)) = 5 bits