SpinalHDL/SpinalHDL #526: Consider providing stand-alone, all-batteries-included, no-network-required installer or container

wjl, 2021-11-03:
I've been in the use-languages-to-generate-netlists game for a long time, starting with my involvement in developing BYU's JHDL more than 20 years ago, and continuing with lots of similar tools including a lot of my own hobby and internal corporate systems. I only came across SpinalHDL pretty recently, and I'm extremely impressed. 

**SpinalHDL works extremely well, but has some problems when used in many environments because it generally requires network access to install and run.** This isn't *really* SpinalHDL's fault, of course; most of this is an artifact of how SBT works, etc. We're a very networked world and toolchains love to be able to just download updates willy-nilly. I get it; it's convenient and user friendly to a large extent. However, from an end-user perspective, this means it is extremely difficult or impossible to use SpinalHDL in some very common situations, and most RTL designers could care less if SpinalHDL is part of the cool Scala ecosystem -- they just want a tool.

**Fixing this lack and making SpinalHDL more accessible in this way could massively help SpinalHDL's popularity and adoption.**

- On a normal desktop machine when behind a heavy-handed corporate firewall.
  - Corporate firewalls are often super heavy-handed, don't correctly follow internet standards, use buggy MITM SSL interception, etc.
  - Very savvy people can work around this, but good luck telling your coworker who's a pro at EE but not IT: "SpinalHDL is super cool, give it a shot real quick!" (They are not going to get past figuring out all the weird  proxy and and SSL cert magic they have to do, and will probably blame their problems on SpinalHDL. "Meh, I couldn't even get it installed, nevermind.")
- In a non-networked standalone configuration, like a canonical build server.
  - Once somebody savvy has created, say, a Docker container of SpinalHDL with no network dependencies, this is pretty usable.
  - But right now there is no canonical SpinalHDL all-in-one Docker container, and all of the unofficial SpinalHDL Docker containers I was able to find still required a network connection to actually execute because they didn't actually include all dependencies in the container.
- In other non-networked or non-fully-networked environments, such as:
  - Completely standalone systems (e.g. due to cleanroom rules, security/classification, etc).
  - Intermittently connected systems which are only connected to the internet periodically for updates, etc.

Anyway, from my perspective, any one of these would do the job, but higher is better in my opinion:
- An official all batteries included SpinalHDL installer that does not need network access for any step of installation or runtime.
- An official Docker container that at runtime doesn't need network access for any reason (but could require the network to rebuild the container).
- Official instructions that walk someone through how to install SpinalHDL in a way that after its done, doesn't require any kind of network access at runtime. (These instructions could be then used by savvy end users to generate their own Docker containers, a standalone VMs, etc).

In summary, thanks again for making SpinalHDL; **please consider making it possible to use for non-IT savvy engineers stuck behind egregious corporate firewalls or otherwise prevented by technical limitations or policies to not require internet access when installing or running SpinalHDL.** =)

soronpo, 2021-11-04:
I think most of the problem is mainly dependent on Scala toolflow than SpinalHDL.

jiegec, 2021-11-04:
Maybe publish them to a local ivy2 repo and copy the files around?

Dolu1990, 2021-11-04:
@wjl Thanks :)

As i'm nearly never leaving my cave. I didn't had to face those issues much, but the few time i had to go to compagnies, networking was a paine and suffering.

So i agree.

Then there is three factors : 
- The environnement (native, vm, container)
- The buildscripts (sbt, ...)
- The depedencies (jar)

My guess is that the most important for deployement would be to have a sbt running natively. (I'm not so much into containers and don't realy want to lock people with docker).

So, maybe we could setup a repository with a few script to : 
- Clean all the caches (.sbt  ,ivy2)
- Build a dummy sbt project which many usefull depedencies, whch will force the caches to fill (online).
- Package all those caches into some archive 
- Deploy a arhive of those cache on a offline machine, which will then never need internet.

Hopefully that would be workable first step ? (This follow the jiegec idea)
Or there is something flawed ?

wjl, 2021-11-04:
So, I've been aware of Scala for years, but I've never seriously used it, and SBT in particular is pretty black-magical to me right now, so I'm probably not really going to be able to help do this the **right way**. BUT, I did want to report that I did get something working for my day job with docker that I proved could work without any network access after it was built. It turned out REALLY BIG (almost 2 GiB) but it does work with both VexRiscv and the example SpinalHDL template repository, including Verilator simulation support.

I can't post the exact Dockerfile I used, but I'm going to summarize here what it did. I plan to recreate this as an open-source Dockerfile, so I can use it personally and share it with you guys, but I haven't done this yet. Keep in mind these are all automated steps, but I'm just describing them there informally.

- Started with a Ubuntu 20.04 LTS image that I had previously customized to be able to work with our corporate proxy (blood and tears, I promise you, and they change stuff often so it breaks all the time).
  - I picked 20.04 both because it's "approved" for use at our company and using an Ubuntu LTS release seems like a sensible baseline. I did make sure that it had versions of the JDK, Verilator, etc, that were compatible with SpinalHDL.
  - As an aside, I tried this with the official OpenJDK images (8, 11, and 17) but for some reason none of those worked with SBT, and ended up giving inscrutable Java backtraces (from SBT, it didn't get far enough to even download SpinalHDL).
- On top of the Ubuntu image I installed OpenJDK 11 from the normal Ubuntu 20.04 LTS repositories, as well as a few other prerequisites to build the rest of the image, like curl, git, ca-certificates, gpg, etc.
- I also included verilator, gcc, bison, etc to make sure SpinalHDL could run the example simulation, and also pulled in ghdl and icarus, but I didn't test simulating with those.
- I added the SBT repository to APT, imported the GPG key, etc, and installed SBT from the repository.
- Then I cloned both the official SpinalHDL template repository, and the VexRiscv repository.
- In each of those, I ran an sbt build to emit VHDL, emit Verilog, and run a Verilator simulation, which forced SBT to download various versions of scala, different versions of SBT, SpinalHDL and its dependencies, etc.
- After running on both of these repositories, I tested the container against separately checked out copies of these repositories outside the container while preventing any network access on the host.
  - I originally only did this with the example repository, but then when I tried to use it on VexRiscv, it failed because it wanted to download a different version of SBT and a bunch of other stuff.
  - When no network is available, a couple Java backtraces get spit out by SBT, but in the end everything keeps trucking and I get my nicely generated VHDL and/or Verilog and/or Verilator simulation results.

So I think I've proved this is possible without TOO much pain and suffering. Here are some takeaways:
- I did this by "brute force", by just saving *everything* that SBT & SpinalHDL cache during the run of two example repositories.
  - I don't know what or where SBT or SpinalHDL squirrel away things (it's not where it's documented, as there is definitely not a ~/.ivy2, but there is ~/.cache/sbt/ and possibly other things.
  - With this technique, I don't really care -- but it does mean the resulting image is pretty huge (not a show stopper for me, but probably a bit of a waste).
- It seems that even after making this all-in-one SpinalHDL container, it still will fail if trying to run against any SpinalHDL projects that point to a different version of SBT than what was used in the two example repositories.
  - The SBT version I installed is 1.5.5.
  - But the SpinalHDL example template makes it pull down SBT 1.4.7, and
  - The VexRiscv repository pulls down something else (1.2.x, I forget which), and seemed to also pull a whole lot more SpinalHDL jars than just running the template design did (but I may have misunderstood what was going on).
- Even when it is all working, I get lots of SBT warnings about possibly incompatible libraries telling me to look at "evictions" and so forth -- I just ignored all this and still got my VHDL. =) 

Anyway, hope this writeup helps. As I said, I will probably attempt to remake this Dockerfile with the above steps on my own time now that I know what I did, but I certainly won't claim that it's perfect. If I do, I will post about it here. But in general, I would love to see this handled by someone who actually knows more what they are doing with SBT and SpinalHDL. =)

wjl, 2021-11-05:
Okay, here is an open-source version of a docker container build script. Each build step is documented in the file. Please feel free to do whatever you want with it!

[Dockerfile.txt](https://github.com/SpinalHDL/SpinalHDL/files/7479548/Dockerfile.txt)

This actually should be called "Dockerfile" and not "Dockerfile.txt", but I had to give it an extension to be able to upload it.

wjl, 2021-11-05:
Here is an example of running the VexRiscv "GenFull" demo with this container. I don't know if all these warnings are normal, but in the end it spits out a perfectly fine Verilog file.

```
% podman run -it --rm --network=none --volume=$(pwd):/data spinalhdl sbt "runMain vexriscv.demo.GenFull"
[info] Loading settings for project data-build from plugins.sbt ...
[info] Loading project definition from /data/project
[info] Updating ProjectRef(uri("file:/data/project/"), "data-build")...
[info] Done updating.
[warn] There may be incompatibilities among your library dependencies; run 'evicted' to see detailed eviction warnings.
[info] Loading settings for project root from build.sbt ...
[info] Set current project to VexRiscv (in build file:/data/)
[info] Updating ...
[info] Done updating.
[warn] There may be incompatibilities among your library dependencies; run 'evicted' to see detailed eviction warnings.
[info] Compiling 88 Scala sources to /data/target/scala-2.11/classes ...
[warn] /data/src/main/scala/vexriscv/VexRiscv.scala:35:17: match may not be exhaustive.
[warn] It would fail on the following input: None
[warn]     plugins.find(_.getClass == clazz) match {
[warn]                 ^
[warn] /data/src/main/scala/vexriscv/demo/smp/VexRiscvSmpLitexMpCluster.scala:1:1: 
[warn] Found names but no class, trait or object is defined in the compilation unit.
[warn] The incremental compiler cannot record the dependency information in such case.
[warn] Some errors like unused import referring to a non-existent class might not be reported.
[warn]     
[warn] package vexriscv.demo.smp
[warn] ^
[warn] there were 38 deprecation warnings; re-run with -deprecation for details
[warn] there were two feature warnings; re-run with -feature for details
[warn] four warnings found
[info] Done compiling.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.protobuf.UnsafeUtil (file:/root/.sbt/boot/scala-2.12.7/org.scala-sbt/sbt/1.2.7/protobuf-java-3.3.1.jar) to field java.nio.Buffer.address
WARNING: Please consider reporting this to the maintainers of com.google.protobuf.UnsafeUtil
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Packaging /data/target/scala-2.11/vexriscv_2.11-2.0.0.jar ...
[info] Done packaging.
[info] Running (fork) vexriscv.demo.GenFull 
[info] [Runtime] SpinalHDL v1.6.0    git head : 73c8d8e2b86b45646e9d0b2e729291f2b65e6be3
[info] [Runtime] JVM max memory : 1960.0MiB
[info] [Runtime] Current date : 2021.11.05 00:07:35
[info] [Progress] at 0.000 : Elaborate components
[info] [Progress] at 4.659 : Checks and transforms
[info] [Progress] at 6.264 : Generate Verilog
[info] [Warning] 216 signals were pruned. You can call printPruned on the backend report to get more informations.
[info] [Done] at 7.716
[success] Total time: 294 s, completed Nov 5, 2021, 12:07:44 AM

```

Dolu1990, 2021-11-07:
Thanks :D
So maybe i should open a centralized with all the methods we have so far ?
Something like SpinalNomad ?

Dolu1990, 2021-11-07:
Got sbt to work on a machine never connecter to internet, using file transfer by using : 
- https://github.com/sbt/sbt/releases/download/v1.5.5/sbt-1.5.5.tgz
- ~/.cache/coursier
- ~/.cache/JNA
- ~/.sbt

Just running SpinalTemplateSbt.git, all those cache are about 250 MB

wjl, 2021-11-07:
> Thanks :D So maybe i should open a centralized with all the methods we have so far ? Something like SpinalNomad ?

That seems like a good idea. It could contain various ways of using SpinalHDL stand-alone or offline, starting with the Dockerfile I made and any other kinds scripts or instructions that make sense, and then and would be a specific place where such kind of things could be contributed / maintained.

Dolu1990, 2021-11-08:
There it is : 
https://github.com/SpinalHDL/SpinalNomad.git

numero-744, 2022-08-18:
I think this issue can be closed as done

closed 2022-08-19 by Dolu1990