SpinalHDL/SpinalHDL #529: Found an error about SpinalEnum staticEncoding

DoubiZhizun, 2021-11-10:
when I use SpinalEnum staticEncoding in my design and do sim with verilator, it told me that a function named binary_sequential_to_staticEncoding not found, I think you may forget to generate it when generate verilog?

Dolu1990, 2021-11-10:
Hi,

Hmm it is possible, there was a few changes in the verilog enum generation not so long ago.

Could you provide a self contained SpinalHDL example which reproduce the broken verilog ?
Thanks :)

DoubiZhizun, 2021-11-10:
[example.zip](https://github.com/SpinalHDL/SpinalHDL/files/7511157/example.zip)
Here's an example, and I also found the same problem when generate VHDL.

Dolu1990, 2021-11-10:
Thanks :D

That example.zip realy helped, because to recreate the issue it realy has to hit some specific condition : 
A enum mux which has a enum operator as condition.

This should now be fixed in dev. Thanks :)

commit 420fa77e (2021-11-10): Prevent enum's mux normalizeInputs being applied to the selection exception
(fix #529)
closed 2021-11-10 by Dolu1990