SpinalHDL/SpinalHDL #530: component alias

dxsm, 2021-11-10:
Hi，is there any plan to create an alias for `Component` as `Module`？This is more friendly to the newers who are not familiar with VHDL when promoting to them. I tried `type Component = Module` but throw error.Is there any way to achieve it.

Dolu1990, 2021-11-10:
Hmm, i tried to modifiy the spinal.core package object itself with that, and i didn't had issues with for instance modifying the StreamFifo to be a Module : 

```scala
package object core extends BaseTypeFactory with BaseTypeCast {

  import languageFeature.implicitConversions

  type Module = spinal.core.Component
  
}

....
class StreamFifo[T <: Data](dataType: HardType[T], depth: Int) extends Module {

}
```

What error did you had ? what was your test which broke ?

dxsm, 2021-11-10:
```scala
import spinal.core._
import spinal.lib._

type Module = Component
class top extends Module {
  val a = B"2'd1"
}

object main {
  def main(args: Array[String]) {
    SpinalVerilog(new top())
  }
}
```
This code throws error:
```
expected class or object definition
type Module = Component
```

Dolu1990, 2021-11-10:
Ahhh that's normal, i will commit the feature addition tomorrow ^^

dxsm, 2021-11-11:
Great，thanks!

Dolu1990, 2021-11-11:
pushed

closed 2021-11-11 by Dolu1990