SpinalHDL/SpinalHDL #535: Consider using code formatting tools

jiegec, 2021-11-23:
The current code is pretty inconsistent with regard to indents, spaces etc. Scalafmt can be used to format the scala sources.

Dolu1990, 2021-11-24:
We can try it, I'm a bit scared that as as SpinalHDL is a idsl, it may trigger quite some false positive

jiegec, 2022-04-13:
Closed by #539

Dolu1990, 2022-10-18:
> formatting all files at once; might be annoying if there are currently big PRs modifying lots of existing files (related to https://github.com/SpinalHDL/SpinalHDL/pull/539#issuecomment-1083185452 I guess)

I think this is a no go, it would also make others people forks quite hard to merge

> formatting files not modified by currently open PRs (I hope there is a way to automate this)

This doesn't has the visbility on fork not being PR yet.

> make the lint return warnings so that GitHub sorts them out, to only show those related to the current PR (I'm not 100% sure about it, and having a hard pass/fail would be better IMO).

This would be ok i think, as long as it isn't too rigid and allows hand crafted code alignment, i guess.

>  (IMO 120 chars per line is a lot, the maximum is usually between 80 and 100)

I think here there is a few factors which make 120 necessary : 
- Hardware description is a lot about nested scope
- Hardware description require often a lot of chained doted notation

"Random" example from the NaxRiscv LSU : 
```scala
          setup.cacheStore.cmd.mask :=  AddressToMask(setup.cacheStore.cmd.address, size, widthOf(setup.cacheStore.cmd.mask))
```

numero-744, 2022-10-18:
I 80% agree your first point, and 100% the second one.

> This would be ok i think, as long as it isn't too rigid and allows hand crafted code alignment, i guess.

It would just be a check, we can always ignore it. If it is set to be a warning, I think it will just print stuff in the diff and not even put a red cross on the actions at the end of PR discussion. About code alignment, if you are talking about things like aligning assignment tokens, IMHO it is really useful for verbose languages like VHDL and Verilog, but not for modern DRY languages like Scala (and SpinalHDL).

To make the code base slowly improve, we could imagine formatting files as we modify them, sometimes, as a side-effect of other PRs, (in a specific commit so that it is cleaner and easier to review), for those who want/care. Because if someone else modifies the same file, there will probability be conflicts anyway. I mean, there is always an increased risk of conflict but here as the initial risk is > 0 then the increase is lower XD

About nested scope, indeed, I should check but I have noticed it does not consider indentation and code the same (something like we can have indentation + code < 100 but code itself < 80).

About notations, for this example, I indeed find the line too long XD

I would prefer:

```scala
          setup.cacheStore.cmd.mask := AddressToMask(
            setup.cacheStore.cmd.address,
            size,
            widthOf(setup.cacheStore.cmd.mask)
          )
// or (easier to understand IMO)
          val cmd = setup.cacheStore.cmd
          cmd.mask := AddressToMask(cmd.address, size, widthOf(cmd.mask)) // <- 74 chars, indentation included
```

Dolu1990, 2022-10-20:
> If it is set to be a warning

Fair ^^

> About code alignment, if you are talking about things like aligning assignment tokens, IMHO it is really useful for verbose languages like VHDL and Verilog

Sure, it is less usefull than in VHDL/Verilog, but still a necessity usefull. For instance for class / function parameters : 
https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/CsrPlugin.scala#L94

> To make the code base slowly improve, we could imagine formatting files as we modify them, sometimes, as a side-effect of other PRs,

Only modification themself may be reformated. but all the code aren't shouldn't as it make tracking changes a paine ^^

> About notations, for this example, I indeed find the line too long XD

If you look at this line alone, yes, but when you look at it its contexts, not necessarly.
Look at this magnificent chunky block of code, managing the cacheStore.cmd bus <3 : 
https://github.com/SpinalHDL/NaxRiscv/blob/1e66c0fd7937b14c303883043f6306e0198745b6/src/main/scala/naxriscv/lsu/LsuPlugin.scala#L1644

In some ways, allowing long lines allows to preserve the "macro" shape of the code, avoiding "unecessary" line returns for things which aren't so important. Making the navigation through the code easier.

numero-744, 2022-10-20:
For aligning I'm okay to keep it but I would not have enabled it because I still not like it much :smile: except for VHDL (actually even for Verilog I do not like it much :thinking:).

> > To make the code base slowly improve, we could imagine formatting files as we modify them, sometimes, as a side-effect of other PRs,
>
> Only modification themself may be reformated. but all the code aren't shouldn't as it make tracking changes a paine ^^

It is really painful to apply scalafmt on a file then un-format (more or less by hand) everything we did not modify, for really little benefit:

* since formatting has been done in its own commit (without modifying anything else) it is not more difficult to track changes;
* for the probability to have conflicts, I have already provided an argument, but I can add one more: with the way Scala splits everything into files, it reduces the probability that two people are working on the same file in parallel.

For your example:

```scala
        val feed = new Area{
          val holdPrefetch = False
          val prediction = prefetch.predictor.io.prediction.cmd.stage().haltWhen(holdPrefetch).toFlow
//          prediction.valid clearWhen(True) //TODO remove

          //WARNING, setupCacheStore is also used by the peripheral controller to know what to do
          val io = sq.mem.io.readAsync(ptr.writeBackReal)
          val size = regs.map(_.address.size).read(ptr.writeBackReal)
          val data = mem.word.readAsync(ptr.writeBackReal)
          val skip = False //Used for store conditional
          val doit = ptr.writeBack =/= ptr.commit && waitOn.ready && !prediction.valid
          val fire = CombInit(doit)

          setup.cacheStore.cmd.valid := doit
          setup.cacheStore.cmd.address := mem.addressPost.readAsync(ptr.writeBackReal)
          setup.cacheStore.cmd.mask :=  AddressToMask(setup.cacheStore.cmd.address, size, widthOf(setup.cacheStore.cmd.mask))
          setup.cacheStore.cmd.generation := generation
          setup.cacheStore.cmd.data.assignDontCare()
          setup.cacheStore.cmd.io := io
          setup.cacheStore.cmd.prefetch := False
          switch(size){
            for(s <- 0 to log2Up(widthOf(setup.cacheStore.cmd.data)/8)) is(s){
              val w = (1 << s)*8
              setup.cacheStore.cmd.data.subdivideIn(w bits).foreach(_ := data(0, w bits))
            }
          }

          when(prediction.valid){
            setup.cacheStore.cmd.valid := True
            setup.cacheStore.cmd.address := prediction.payload
            setup.cacheStore.cmd.io := False
            setup.cacheStore.cmd.prefetch := True
          }

          ptr.writeBack := ptr.writeBack + U(fire)
        }
```

vs (added `val cmd`, moved WARNING line, removed TODO line, added empty line before switch and scalafmt'ed in a repo without custom config).

```scala
        val feed = new Area {
          val holdPrefetch = False
          val prediction =
            prefetch.predictor.io.prediction.cmd
              .stage()
              .haltWhen(holdPrefetch)
              .toFlow
          val io = sq.mem.io.readAsync(ptr.writeBackReal)
          val size = regs.map(_.address.size).read(ptr.writeBackReal)
          val data = mem.word.readAsync(ptr.writeBackReal)
          val skip = False // Used to store conditional
          val doit =
            ptr.writeBack =/= ptr.commit && waitOn.ready && !prediction.valid
          val fire = CombInit(doit)
          // WARNING, setup.CacheStore is also used by the peripheral controller to know what to do
          val cmd = setup.cacheStore.cmd

          cmd.valid := doit
          cmd.address := mem.addressPost.readAsync(ptr.writeBackReal)
          cmd.mask := AddressToMask(cmd.address, size, widthOf(cmd.mask))
          cmd.generation := generation
          cmd.data.assignDontCare()
          cmd.io := io
          cmd.prefetch := False

          switch(size) {
            for (s <- 0 to log2Up(widthOf(cmd.data) / 8)) is(s) {
              val w = (1 << s) * 8
              cmd.data.subdivideIn(w bits).foreach(_ := data(0, w bits))
            }
          }
          when(prediction.valid) {
            cmd.valid := True
            cmd.address := prediction.payload
            cmd.io := False
            cmd.prefetch := True
          }
          ptr.writeBack := ptr.writeBack + U(fire)
        }
```

Readon, 2022-10-21:
> For aligning I'm okay to keep it but I would not have enabled it because I still not like it much 😄 except for VHDL (actually even for Verilog I do not like it much 🤔).
> 
> > > To make the code base slowly improve, we could imagine formatting files as we modify them, sometimes, as a side-effect of other PRs,
> > 
> > 
> > Only modification themself may be reformated. but all the code aren't shouldn't as it make tracking changes a paine ^^
> 
> It is really painful to apply scalafmt on a file then un-format (more or less by hand) everything we did not modify, for really little benefit:
> 
> * since formatting has been done in its own commit (without modifying anything else) it is not more difficult to track changes;
> * for the probability to have conflicts, I have already provided an argument, but I can add one more: with the way Scala splits everything into files, it reduces the probability that two people are working on the same file in parallel.
> 
> For your example:
> 
> ```scala
>         val feed = new Area{
>           val holdPrefetch = False
>           val prediction = prefetch.predictor.io.prediction.cmd.stage().haltWhen(holdPrefetch).toFlow
> //          prediction.valid clearWhen(True) //TODO remove
> 
>           //WARNING, setupCacheStore is also used by the peripheral controller to know what to do
>           val io = sq.mem.io.readAsync(ptr.writeBackReal)
>           val size = regs.map(_.address.size).read(ptr.writeBackReal)
>           val data = mem.word.readAsync(ptr.writeBackReal)
>           val skip = False //Used for store conditional
>           val doit = ptr.writeBack =/= ptr.commit && waitOn.ready && !prediction.valid
>           val fire = CombInit(doit)
> 
>           setup.cacheStore.cmd.valid := doit
>           setup.cacheStore.cmd.address := mem.addressPost.readAsync(ptr.writeBackReal)
>           setup.cacheStore.cmd.mask :=  AddressToMask(setup.cacheStore.cmd.address, size, widthOf(setup.cacheStore.cmd.mask))
>           setup.cacheStore.cmd.generation := generation
>           setup.cacheStore.cmd.data.assignDontCare()
>           setup.cacheStore.cmd.io := io
>           setup.cacheStore.cmd.prefetch := False
>           switch(size){
>             for(s <- 0 to log2Up(widthOf(setup.cacheStore.cmd.data)/8)) is(s){
>               val w = (1 << s)*8
>               setup.cacheStore.cmd.data.subdivideIn(w bits).foreach(_ := data(0, w bits))
>             }
>           }
> 
>           when(prediction.valid){
>             setup.cacheStore.cmd.valid := True
>             setup.cacheStore.cmd.address := prediction.payload
>             setup.cacheStore.cmd.io := False
>             setup.cacheStore.cmd.prefetch := True
>           }
> 
> 
> 
>           ptr.writeBack := ptr.writeBack + U(fire)
>         }
> ```
> 
> vs (added `val cmd`, moved WARNING line, removed TODO line, added empty line before switch and scalafmt'ed in a repo without custom config).
> 
> ```scala
>         val feed = new Area {
>           val holdPrefetch = False
>           val prediction =
>             prefetch.predictor.io.prediction.cmd
>               .stage()
>               .haltWhen(holdPrefetch)
>               .toFlow
>           val io = sq.mem.io.readAsync(ptr.writeBackReal)
>           val size = regs.map(_.address.size).read(ptr.writeBackReal)
>           val data = mem.word.readAsync(ptr.writeBackReal)
>           val skip = False // Used to store conditional
>           val doit =
>             ptr.writeBack =/= ptr.commit && waitOn.ready && !prediction.valid
>           val fire = CombInit(doit)
>           // WARNING, setup.CacheStore is also used by the peripheral controller to know what to do
>           val cmd = setup.cacheStore.cmd
> 
>           cmd.valid := doit
>           cmd.address := mem.addressPost.readAsync(ptr.writeBackReal)
>           cmd.mask := AddressToMask(cmd.address, size, widthOf(cmd.mask))
>           cmd.generation := generation
>           cmd.data.assignDontCare()
>           cmd.io := io
>           cmd.prefetch := False
> 
>           switch(size) {
>             for (s <- 0 to log2Up(widthOf(cmd.data) / 8)) is(s) {
>               val w = (1 << s) * 8
>               cmd.data.subdivideIn(w bits).foreach(_ := data(0, w bits))
>             }
>           }
>           when(prediction.valid) {
>             cmd.valid := True
>             cmd.address := prediction.payload
>             cmd.io := False
>             cmd.prefetch := True
>           }
>           ptr.writeBack := ptr.writeBack + U(fire)
>         }
> ```

What I am doing is manually select the required modification and ignore other formatting changes.

numero-744, 2022-10-23:
Today I've tried to get formatting errors printed like diffs, and to integrate it into GitHub actions to annotate modified files in diff section of PRs, nothing worked :disappointed:

* Tried the 4 github actions in the market matching "scalafmt"
* Tried getting `scalafmt` binary by several ways, I think I should try again later
* Tried to run sbt scalafmtCheck locally, with most recent version and with appropriate config, but cannot make it print warnings :sob: 

Any idea?

Dolu1990, 2022-10-24:
@numero-744 

> val cmd

Yes it is a good way to reduce verbosity ^^

> Today I've tried to get formatting errors printed like diffs, and to integrate it into GitHub actions to annotate modified files in diff section of PRs, nothing worked disappointed

Hmmm i never explored those lands, maybe the easiest would be to find another github scala repo already setuped and copy past from it ?

numero-744, 2022-10-24:
Repos that seem to be configured this way are using `scalafmt` as a binary file into the repository but I did not reproduce yet. Also I don't know if it brings annotations.

For now, all I can get (locally, should be easy to put into actions) is "this file is not formatted", not "this line is not formatted".

closed 2022-11-02 by Dolu1990

numero-744, 2022-11-22:
Another example why I don't like aligning things by-hand: it looks beautiful when you have just done it (maybe when you do it you find it even more beautiful because of auto-satisfaction to have done that fancy thing by-hand) but it makes the line more dependent on other lines around, and over time we as we modify it we can choose to update all lines (actually makes maintenance harder) ([booooring.gif](https://media.tenor.com/BqGv6rHYMCcAAAAd/killing-eve-this-is-so-boring.gif)), or update none:

```scala
                        scopeProperties                : mutable.LinkedHashMap[ScopeProperty[_], Any] = mutable.LinkedHashMap[ScopeProperty[_], Any](),
                        private [core] var _withEnumString : Boolean = true,
                        var enumPrefixEnable                 : Boolean = true,
                        var enumGlobalEnable                 : Boolean = false,
                        bitVectorWidthMax              : Int = 4096
```

I strongly think we should stop "playing with spaces and tabs as monkeys" ([expression source](https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/miscelenea/regular_hdl.html#meta-hardware-description-capabilities)) and just use tools to do good things for us.

Using code formatting tools is a really tiny bit less beautiful, but much easier to maintain on the long term.

Dolu1990, 2022-11-23:
It probably depend the context. In some ways, not having manual code alignement can transform structured code into spagetti noddle mess.

```scala
        val physRd      = create(decoder.PHYS_RD)
        val regfileRd   = create(decoder.REGFILE_RD)
        val robId       = create(ROB.ID)
        val robIdMsb    = create(ROB.MSB)
        val pc          = create(PC)
        val sqAlloc     = create(UInt(log2Up(sqSize)+1 bits))
        val io          = create(Bool())
        val writeRd     = create(Bool())
        val lr          = create(Bool())
        val unsigned    = create(Bool())
        val doSpecial   = create(Bool())
        
        val physRd = create(decoder.PHYS_RD)
        val regfileRd = create(decoder.REGFILE_RD)
        val robId = create(ROB.ID)
        val robIdMsb = create(ROB.MSB)
        val pc = create(PC)
        val sqAlloc = create(UInt(log2Up(sqSize) + 1 bits))
        val io = create(Bool())
        val writeRd = create(Bool())
        val lr = create(Bool())
        val unsigned = create(Bool())
        val doSpecial = create(Bool())        
``` 

But yeah, when you have long line (private [core] var _withEnumString) mixed with short line (var enumPrefixEnable), manual alignement lose some of it niceness XD

> done that fancy thing by-hand
> and over time we as we modify it we can choose to update all lines 

Well, there is semi automated ways to do it with multi line selection. That steal some of the satisfaction.

> or update none

Who dare ! ^^

PR #936 (2022-11-02): ci: add scalafmtCheck step
Fixes #535 

### This PR avoids double workflow

Double workflow makes annotations appear twice if PR and source branch are on the same repository, and consume twice more energy.

### This PR updates `sbt-scalafmt` plugin and configures it

Only modified files (compared to `dev` branch) are checked

### This PR adds a bash script to run scalafmt and interpret the diff to create parsable "problems"

### This PR adds a "problem" regular expression so that GitHub can parse the problems to turn them into PR annotations

### This PR adds a step to Scala CI workflow to che [... cut]