SpinalHDL/SpinalHDL #536: Will there be axilite4 crossbar and axilite4decode in lib in the future?

qgzln, 2021-11-23:
In the  Spinal lib，I find there is axi4 crossbar,axi4 decode ,Will there be axilite4 crossbar and axilite4decode in lib in the future?

Dolu1990, 2021-11-23:
Hi,

I'm not planning to implement it myself, but if somebody has usage of it, and is motivated into implementing it, it can be added

Readon, 2021-12-16:
According to the definition of Axi4Lite it's a subset of AXI, lacking burst access capability. I suppose that Axi4CrossBar would work with that. However you have to test it yourself. :)

Dolu1990, 2021-12-17:
Main issue is that the current AXI4 arbiter extends the number of ID bits in order to route things back.

Readon, 2021-12-18:
Then there will always be a problem on determine the back stream no matter in which implementation. Perhaps the proper way is select a right policy by use id property. For the Stream without id the only way is to process one transaction once.