SpinalHDL/SpinalHDL #544: DecoderSimplePlugin assertIllegalInstruction false positive

piegamesde, 2021-12-11:
I tried to set the `assertIllegalInstruction` option of `DecoderSimplePlugin` and my simulation instantly failed although it shouldn't. Looking at the wave trace, this already happens in the first cycle after the reset goes down.  Looking some more, the problem may be related to the fact the the `decodeExceptionPort.valid` signal is kept high for one cycle after a reset, which I think triggers the false positive. Note that `catchIllegalInstruction` appears to be working just fine somehow.

wswslzp, 2021-12-12:
Is this a vexriscv issue?

piegamesde, 2021-12-12:
Oops. Yes it is, please move to the VexRiscv repository (I think GitHub has this feature)

piegamesde, 2021-12-13:
Moved to https://github.com/SpinalHDL/VexRiscv/issues/225

closed 2021-12-13 by piegamesde