SpinalHDL/SpinalHDL #546: 1.6.1 use addStandardMemBlackboxing(blackboxAll) and StreamFifo error

aiit-linry, 2021-12-13:
[image]

how can I avoid this error and blackbox rams in StreamFifo?

aiit-linry, 2021-12-13:
By the way, use mem( addr ) := data to write a mem is error too when addStandardMemBlackboxing(blackboxAll). Unfortunately, it    is used in StreamFifo for write ram.

aiit-linry, 2021-12-13:
This didn't happen in 1.6.0, but I love most of features in 1.6.1, looking forward to your answer : )

aiit-linry, 2022-01-07:
I cant resolve it, just avoid it via jijingg's mem way. But I hope I can get an answer.

Dolu1990, 2022-01-09:
Hi,

Sorry, i missed the issue, basicaly using my email to see new threads and messages, but it happen that i miss them XD

Good you pinged again ^^

I can reproduce the issue with 

```scala
  case class TopLevel() extends Component {
    val a = slave(Stream(UInt(8 bits)))
    val b = slave(Stream(UInt(8 bits)))

    val x = master(a.queue(16).combStage())
    val y = master(b.queue(16).combStage())
  }

  def main(args: Array[String]): Unit = {
    SpinalConfig().addStandardMemBlackboxing(blackboxAll).generateVerilog(new TopLevel)
  }
  ```
I'm currently trying to see what broke

Dolu1990, 2022-01-09:
So, the seconde issue you had is now fixed in dev, it was because the mem(x) := y syntax has to delete the mem.readAsync(x) that it produce (as a side effect), but it wasn't cleaning up enough of the SpinalHDL AST, there was some left over. And those leftover where triggering some other checks that i added in the same release.

About the first issue, i can't reproduce it, How do you produce it ? Can you share some code ?

Thanks ^^
And sorry for the delay

Dolu1990, 2022-01-09:
I just pushed a potential fix for the first issue, which relax the check for blackboxes.

aiit-linry, 2022-01-09:
It is an honor to have your reply ^^

I can reproduce it under some in some scenarios :
1. change StreamFifo write logic to write function like this picture( just use to avoid second issue )
[image]

2. use **some different FIFO size** and open **oneFilePerComponent** config , with blackbox.

my test code here
`class streamfifo_test extends Component {

    val io = new Bundle {
        val a1 = slave(Stream(UInt(8 bits)))
        val b1 = master(Stream(UInt(8 bits)))
        val a2 = slave(Stream(UInt(16 bits)))
        val b2 = master(Stream(UInt(16 bits)))
    }

    noIoPrefix

    val st_fifo_1 = stream_fifo(UInt(8 bits), 8)
    val st_fifo_2 = stream_fifo(UInt(16 bits), 8)

    io.a1            >> st_fifo_1.io.push
    st_fifo_1.io.pop >> io.b1

    io.a2            >> st_fifo_2.io.push
    st_fifo_2.io.pop >> io.b2
}

object gen_streamfifo_test extends App {

    SpinalConfig(
        oneFilePerComponent = true,
    )
        .addStandardMemBlackboxing(blackboxAll)
        .generateVerilog(new streamfifo_test())

}
`

this two stream_fifo is repackage myself from StreamFifo, my code skill is poor XD

Dolu1990, 2022-01-09:
So with the fixes pushed in the SpinalHDL dev branch, that code is now passing.
You can use the dev branch as 1.6.2 using : 
https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/About%20SpinalHDL/faq.html?highlight=publish#how-to-use-an-unreleased-version-of-spinalhdl-but-committed-on-git

I will release 1.6.2 soon too ^^

Thanks for the issue :D

closed 2022-01-09 by Dolu1990

aiit-linry, 2022-01-09:
wow, I am  looking forward to the new version very much, it sounds cool ^^