SpinalHDL/SpinalHDL #548: CC fifo reset handeling

Dolu1990, 2021-12-16:
Adding some API to ease the handeling of resets on cross clock component : 

```scala
object ResetCtrl{
  def asyncAssertSyncDeassertCreateCd(resetCd : ClockDomain,
                                      clockCd : ClockDomain,
                                      bufferDepth : Int = BufferCC.defaultDepth.get) : Unit = {
    val cd = ClockDomain(clock = clockCd.clock, reset = Bool(), config = clockCd.config)
    cd.reset := asyncAssertSyncDeassert(
      input = resetCd.reset,
      clockDomain = clockCd,
      inputPolarity = HIGH,
      bufferDepth = bufferDepth
    )
    cd
  }
  ```

Readon, 2021-12-16:
The combination of function method name and arguments name seems not well suited? Because there are no signal to be Assert and Deassert according to arguments. 
How about add a resetBy method to ClockDomain class?
```
def resetBy(resetCd : ClockDomain, bufferDepth : Int = BufferCC.defaultDepth.get): ClockDomain = {
    withAsyncReset(ResetCtrl.asyncAssertSyncDeassert(
             input = resetCd.reset,
             clockDomain = this,
             inputPolarity = HIGH,
             bufferDepth = bufferDepth
         )
    )
}
```
When it is applied, the code could be.
```
val popCC = new ClockingArea(popClock.resetBy(pushClock))
```

Dolu1990, 2021-12-18:
>  function method name 

Yes it sucks ^^

> Because there are no signal to be Assert and Deassert according to arguments.

The name was reflecting how the reset is buffered, trying to be explicit in the low level api, to avoid clashing with future alternatives

> val popCC = new ClockingArea(popClock.resetBy(pushClock))

Sure, this is the end goal. But only "resetBy" is not explicit enough about how things are done, for instance, some could directly wire the reset signal without buffering.

What's about :

```scala
 val popCC = new ClockingArea(popClock.withBufferedReset(pushClock))
 ```

"with" meaning it create a new version of popClock
"buffered" meaning the asyncAssertSyncDeassert

?

Readon, 2021-12-18:
good enough

Readon, 2021-12-18:
how about withBufferedResetBy or withBufferedResetFrom， because the argument is not a reset signal，it's a clock domain. Might be good to distinguish them.

Dolu1990, 2021-12-18:
withBufferedResetFrom is good to me ^^

Dolu1990, 2021-12-18:
added with https://github.com/SpinalHDL/SpinalHDL/commit/694e44adc0d360f4804743f5ceb007bd11966d66#diff-ed2f3bf8226589313fa21882e2539100a05c7ae4120a7fcf03571799e2c395c9R1147

But without the withAsyncReset so far.
@Readon Why withAsyncReset in your code ?

Readon, 2021-12-18:
> added with [694e44a#diff-ed2f3bf8226589313fa21882e2539100a05c7ae4120a7fcf03571799e2c395c9R1147](https://github.com/SpinalHDL/SpinalHDL/commit/694e44adc0d360f4804743f5ceb007bd11966d66#diff-ed2f3bf8226589313fa21882e2539100a05c7ae4120a7fcf03571799e2c395c9R1147)
> 
> But without the withAsyncReset so far. @Readon Why withAsyncReset in your code ?

The withAsyncReset is a method of ClockDomain. It's in line 395 of ClockDomain.scala

Dolu1990, 2021-12-18:
yes right, but i mean, why  enforcing asyncronus reset ?

Readon, 2021-12-18:
> 

eh, as I understand the output of ResetCtrl is a async signal to the target ClockDomain ("this" object here.)

Dolu1990, 2021-12-18:
withAsyncReset will affect the topology of the registers reset,forcing the flipflop to not wait the rising edge of the clock to change their value with the reset is high.

It isn't realy related to the nature of the reset signal itself.

Readon, 2021-12-18:
> withAsyncReset will affect the topology of the registers reset,forcing the flipflop to not wait the rising edge of the clock to change their value with the reset is high.
> 
> It isn't realy related to the nature of the reset signal itself.

oh, then it is usable. Why the generated ClockDomain is called clockDomainPimped. Seems it's not easy to get what it refers to.

Dolu1990, 2021-12-20:
@Readon  @yuqi-feng So i guess now the issue is for component designed without care of this, how to use them in a design which has those reset properties (no asserted together)

Either we can use a ScopeProperty which will trigger the low level cross clock fifo / buffer implementation to reset the pop side using the push side reset, either we have to patch components that use those CC elements to have the option.

Readon, 2021-12-21:
> @Readon @yuqi-feng So i guess now the issue is for component designed without care of this, how to use them in a design which has those reset properties (no asserted together)
>
> Either we can use a [...]

I am not familiar with ScopeProperty. :) However, I guess it's hard to detect pop side and push side automatically. 
Following explicit is better than implicit rule, a patch or patches would be better. The number of components that need to be patched will be limited.

yuqi-ali, 2021-12-21:
it's fine

yuqi-ali, 2021-12-21:
I think it can be used in StreamCCFifo and StreamCCByToggle. the push clockDomain has a responsibility to reset the whole Component to avoid unexpected data while asserting the reset， and the bufferDepth is better to same as the bufferdepth used in the component

Dolu1990, 2021-12-22:
> the push clockDomain has a responsibility to reset the whole Component 

@yuqi-ali Would there be scenario, where this would not be the ideal implementation ?

Dolu1990, 2021-12-22:
(seems ok to me to implement )

yuqi-ali, 2021-12-23:
both of the intel and xilinx recommend the method in project,and it's ok in system design

Dolu1990, 2021-12-23:
I will try to implement it, in a way which avoid duplication of that reset buffer

Dolu1990, 2021-12-23:
I pushed the changes in a separated branch via 
https://github.com/SpinalHDL/SpinalHDL/commit/4defba80fa39ce044ce14981df0bf5d28696e3e9
https://github.com/SpinalHDL/SpinalHDL/commit/9674cdff7b569f7ea13979ef9cb6f466df2edb8b
I will merge it in dev if no concerns are raised  ^^

closed 2021-12-30 by Dolu1990

Pyromuffin, 2022-08-17:
this change broke our design, as now streamfifocc implicitly creates a buffercc component for the reset. We don't supply constraints for buffercc in our builds, so we were confused why everything was now failing timing. 

simple enough to go through and change all our streamfifocc to use a false argument at the end, but it took us a long time to hunt down this change as what was causing the break.

Dolu1990, 2022-08-19:
Ahhh damned :/

On which synthesis / place / route toolchain you had the issue ? what FPGA  / ASIC ?

PR #567 (2021-12-30): Cross block elements from the lib now use the push side to provide the pop side reset #548
Fix #548