SpinalHDL/SpinalHDL #551: A Register in Bool.fall() is not Initialized

ZimingYuan, 2021-12-17:
In `SpinalHDL/core/src/main/scala/spinal/core/Bool.scala`, line 137:
```
 def fall(): Bool = ! this && RegNext(this)
```
The register is never initialized in the code. Sometimes it will be mistakenly initialized to True and this will cause wrong result.
I think this overload of the function is misleading. It's better to delete it or initialize the register to the default value False.

wswslzp, 2021-12-17:
[image]
There is a `fall()` version with an initialization value.

ZimingYuan, 2021-12-17:
> [image] There is a `fall()` version with an initialization value, just at line 135.

I know. I mean someone (careless people like me) may mistakenly use the version in line 137. Since it's misleading, it would be better to just delete or modify it.

Dolu1990, 2021-12-17:
Hmm, maybe renaming it into fallWithoutInit, but then things become soooo verbose.

numero-744, 2022-10-21:
I agree I think we can do the parallel with `RegNext` which would become `RegNextWithoutInit`.

This kind of mistake can be detected, I think. How? Is there a `Phase` do detect registers without `init`? If so, how to use it? I cannot find this on RTD.

But this is going a little bit off-topic for an old issue so I close it ^^' (but I'm still curious about the answer).

closed 2022-10-21 by numero-744

Dolu1990, 2022-10-24:
>  Is there a Phase do detect registers without init

There is a phase which check if registers aren't driven (no init / no assignement).
Also there is a phase which translate register with only init into literals.

> This kind of mistake can be detected,

User side mistake, but legal hardware, so not realy catchable ?

numero-744, 2022-10-24:
I mean detect it as an option. Maybe with #916 it can be easier, using `allReg` and check for each reg if it has an `init`?

Dolu1990, 2022-10-24:
Sure #916 can help looking at it.
But overall, i would say having all registers with an init value is a bad practice, so not sure it is so helpfull.
We could generate a list of all the registers with init for the user to look at, but it seems to me that there will just be too much to look at very fast (ex vivado warning ocean ^^)

Dolu1990, 2022-10-24:
On FPGA : 
- Consume the clr thing, which could have been used for other usages. So use more LUT and decrease FMAX. It can realy be a big deal, as the FMax impact isn't necessarly small. (from my experience) (also, more lut used => less compact design => more time to travel between LUT)

On ASIC : 
- I'm not an expert at all, but likely having the reset increase the area and add some delay. Also, increasing the area by itself can reduce the FMax of the design, as it mean less compact logic => bigger distance to travel. But i have no data to back any of those claims ^^