SpinalHDL/SpinalHDL #552: add replication operator?

wmj0607, 2021-12-18:
I'm trying to replicate some bits several times, in verilog `{repetition_number{expr1, expr2, ..., exprN}}` will make it.

A trick is, you can replicate a single bit by `S(bit, n.bits)`, but I can't figure out a convenient way to replicate multiple bits.

So maybe add methods like `bits.repl(n)` or `Repl(n, bits)`?

PS: when generate verilog, `S(bit, n.bits)` will give you:
``` verilog
  wire       [0:0]    _zz_out1;

  assign _zz_out1 = in1;
  assign out1 = {{7{_zz_out1[0]}}, _zz_out1};
```

Dolu1990, 2021-12-18:
I added a Repeat(that, times) in spinal.lib

Is it good ?

wmj0607, 2021-12-18:
looks good to me, thanks.
haven't thought of doing that way :P

closed 2021-12-18 by wmj0607