SpinalHDL/SpinalHDL #554: An error about HDMI

DoubiZhizun, 2021-12-19:
in spinal.lib.graphic.hdmi, component VgaToHdmiEcp5, forget to add crossClockDomain tag behind val shift_R,shift_G,shift_B and shift_C.

Dolu1990, 2021-12-20:
Hi,

In which situation do you have the issue ?

DoubiZhizun, 2021-12-20:
When I use different ClockDomain as parameters VgaCd and hdmiCd, eg:

SpinalVhdl(VgaToHdmiEcp5(ClockDomain.external("VGA"),ClockDomain.external("HDMI")))

It will return me clock crossing violation error, eg:

 Error detected in phase PhaseCheckCrossClock
********************************************************************************
********************************************************************************

CLOCK CROSSING VIOLATION :
- Source            : (toplevel/encode_R/io_TMDS : out Bits[10 bits]) spinal.lib.graphic.hdmi.TmdsEncoder$$anon$1$$anonfun$1.apply(TmdsEncoder.scala:22)
- Source clock      : (VGA_clk :  Bool)
- Destination       : (toplevel/shift_R :  Bits[10 bits]) spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$6$$anonfun$apply$3.apply(VgaToHdmi.scala:27)
- Destination clock : (HDMI_clk :  Bool)
- Source declaration :
    spinal.lib.graphic.hdmi.TmdsEncoder$$anon$1$$anonfun$1.apply(TmdsEncoder.scala:22)
    spinal.lib.graphic.hdmi.TmdsEncoder$$anon$1$$anonfun$1.apply(TmdsEncoder.scala:22)
    spinal.lib.graphic.hdmi.TmdsEncoder$$anon$1.<init>(TmdsEncoder.scala:22)
    spinal.lib.graphic.hdmi.TmdsEncoder.<init>(TmdsEncoder.scala:18)
    spinal.lib.graphic.hdmi.TmdsEncoder$.apply(TmdsEncoder.scala:8)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$1.apply(VgaToHdmi.scala:17)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$1.apply(VgaToHdmi.scala:17)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5.<init>(VgaToHdmi.scala:17)
    top.main$$anonfun$main$1.apply(main.scala:8)
    top.main$$anonfun$main$1.apply(main.scala:8)
    spinal.sim.JvmThread.run(SimManager.scala:51)

- Destination declaration :
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$6$$anonfun$apply$3.apply(VgaToHdmi.scala:27)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$6$$anonfun$apply$3.apply(VgaToHdmi.scala:27)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$6.apply(VgaToHdmi.scala:27)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$6.apply(VgaToHdmi.scala:27)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5.<init>(VgaToHdmi.scala:27)
    top.main$$anonfun$main$1.apply(main.scala:8)
    top.main$$anonfun$main$1.apply(main.scala:8)
    spinal.sim.JvmThread.run(SimManager.scala:51)

- Connection path :
      >>> (toplevel/encode_R/io_TMDS : out Bits[10 bits]) at spinal.lib.graphic.hdmi.TmdsEncoder$$anon$1$$anonfun$1.apply(TmdsEncoder.scala:22) >>>
      >>> (toplevel/TMDS_red :  Bits[10 bits]) at spinal.lib.graphic.hdmi.VgaToHdmiEcp5.<init>(VgaToHdmi.scala:15) >>>
      >>> (toplevel/shift_R :  Bits[10 bits]) at spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$6$$anonfun$apply$3.apply(VgaToHdmi.scala:27) >>>

             
********************************************************************************
********************************************************************************

CLOCK CROSSING VIOLATION :
- Source            : (toplevel/encode_G/io_TMDS : out Bits[10 bits]) spinal.lib.graphic.hdmi.TmdsEncoder$$anon$1$$anonfun$1.apply(TmdsEncoder.scala:22)
- Source clock      : (VGA_clk :  Bool)
- Destination       : (toplevel/shift_G :  Bits[10 bits]) spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$7$$anonfun$apply$4.apply(VgaToHdmi.scala:27)
- Destination clock : (HDMI_clk :  Bool)
- Source declaration :
    spinal.lib.graphic.hdmi.TmdsEncoder$$anon$1$$anonfun$1.apply(TmdsEncoder.scala:22)
    spinal.lib.graphic.hdmi.TmdsEncoder$$anon$1$$anonfun$1.apply(TmdsEncoder.scala:22)
    spinal.lib.graphic.hdmi.TmdsEncoder$$anon$1.<init>(TmdsEncoder.scala:22)
    spinal.lib.graphic.hdmi.TmdsEncoder.<init>(TmdsEncoder.scala:18)
    spinal.lib.graphic.hdmi.TmdsEncoder$.apply(TmdsEncoder.scala:8)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$2.apply(VgaToHdmi.scala:18)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$2.apply(VgaToHdmi.scala:18)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5.<init>(VgaToHdmi.scala:18)
    top.main$$anonfun$main$1.apply(main.scala:8)
    top.main$$anonfun$main$1.apply(main.scala:8)
    spinal.sim.JvmThread.run(SimManager.scala:51)

- Destination declaration :
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$7$$anonfun$apply$4.apply(VgaToHdmi.scala:27)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$7$$anonfun$apply$4.apply(VgaToHdmi.scala:27)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$7.apply(VgaToHdmi.scala:27)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$7.apply(VgaToHdmi.scala:27)
[... 23 lines cut ...]
    spinal.lib.graphic.hdmi.TmdsEncoder.<init>(TmdsEncoder.scala:18)
    spinal.lib.graphic.hdmi.TmdsEncoder$.apply(TmdsEncoder.scala:8)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$3.apply(VgaToHdmi.scala:19)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$3.apply(VgaToHdmi.scala:19)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5.<init>(VgaToHdmi.scala:19)
    top.main$$anonfun$main$1.apply(main.scala:8)
    top.main$$anonfun$main$1.apply(main.scala:8)
    spinal.sim.JvmThread.run(SimManager.scala:51)

- Destination declaration :
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$8$$anonfun$apply$5.apply(VgaToHdmi.scala:27)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$8$$anonfun$apply$5.apply(VgaToHdmi.scala:27)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$8.apply(VgaToHdmi.scala:27)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$8.apply(VgaToHdmi.scala:27)
    spinal.lib.graphic.hdmi.VgaToHdmiEcp5.<init>(VgaToHdmi.scala:27)
    top.main$$anonfun$main$1.apply(main.scala:8)
    top.main$$anonfun$main$1.apply(main.scala:8)
[... cut ...]
      >>> (toplevel/encode_B/io_TMDS : out Bits[10 bits]) at spinal.lib.graphic.hdmi.TmdsEncoder$$anon$1$$anonfun$1.apply(TmdsEncoder.scala:22) >>>
      >>> (toplevel/TMDS_blue :  Bits[10 bits]) at spinal.lib.graphic.hdmi.VgaToHdmiEcp5.<init>(VgaToHdmi.scala:15) >>>
      >>> (toplevel/shift_B :  Bits[10 bits]) at spinal.lib.graphic.hdmi.VgaToHdmiEcp5$$anonfun$8$$anonfun$apply$5.apply(VgaToHdmi.scala:27) >>>

I think this error may because lack of crossClockDomain tag behind val shift_R,shift_G,shift_B and shift_C.

Dolu1990, 2021-12-20:
Ahhh, this is on purpose, basicaly, the cross clock domain do the assumption that the two clock are syncronus, so there is they way to specify it : 

```scala
  SpinalVhdl {
    val vgaCd = ClockDomain.external("VGA")
    val hdmiCd = ClockDomain.external("HDMI")
    hdmiCd.setSynchronousWith(vgaCd)
    VgaToHdmiEcp5(vgaCd, hdmiCd)
  }
```

DoubiZhizun, 2021-12-21:
Ok, thank you for your guide, I never realize that I can syncronus the two clock domain. My problem has resolved.

closed 2021-12-21 by DoubiZhizun