SpinalHDL/SpinalHDL #556: Read Only type register issue and negedge sampled register

yyxxrr, 2021-12-22:
I am designing a small module that contains a register bank, but it seems like the RO type register has a problem. In the RO method _(spinalhdl-lib_2.12-1.6.1-sources.jar\spinal\lib\bus\regif\RegInst.scala line 139)_  line 139 the register is not initialized.
`  protected def RO(bc: BitCount): Bits = Bits(bc)`

I would like to define a register that is sampled at the negative clock edge, can anyone tell me how to do this?

XiaoruiYin, 2021-12-23:
I noticed that the address increment of register in BusIf `spinalhdl-lib_2.12-1.6.1-sources.jar\spinal\lib\bus\regif\BusIfBase.scala line 46` must be multiple of wordAddressInc (dataWidth / 8), but this makes no sense to me. Is there a way to create continuous addresses?

Dolu1990, 2021-12-26:
@jijingg ^^

Readon, 2022-12-18:
I think it's related with #904

closed 2023-03-21 by xiaorui-yin