SpinalHDL/SpinalHDL #560: How to set the name of Stream.queue() ?

dahuwei, 2021-12-26:
Hi, when I use Stream.queue() to generate stream fifo. `val streamInQueue = io.flowIn.toStream.throwWhen(!io.enable).queue(256)`
The generated fifo and signal name is streamFifo_1.
Is there any method to use the name of val (streamInQueue) instead of streamFifo_n? This name is difficult to find when the code size increase. 

scala code:
`case class Test(useQueue: Boolean = true) extends Component {
  val io = new Bundle {
    val enable = in Bool()
    val flowIn = slave(Flow(UInt(32 bits)))
    val streamOut = master(Stream(UInt(32 bits)))
  }
  if(useQueue) {
    val inputQueue = new Area {
      val streamInQueue = io.flowIn.toStream.throwWhen(!io.enable).queue(256)
      io.streamOut << streamInQueue
    }.setCompositeName(this, "inputQueue", weak = true)
  } else {
    io.streamOut << io.flowIn.toStream.throwWhen(!io.enable)
  }
}`

generated verilog code:
`module Test (
  input               io_enable,
  input               io_flowIn_valid,
  input      [31:0]   io_flowIn_payload,
  output              io_streamOut_valid,
  input               io_streamOut_ready,
  output     [31:0]   io_streamOut_payload,
  input               clk,
  input               resetn
);
  wire                streamFifo_1_io_push_ready;
  wire                streamFifo_1_io_pop_valid;
  wire       [31:0]   streamFifo_1_io_pop_payload;
  wire       [8:0]    streamFifo_1_io_occupancy;
  wire       [8:0]    streamFifo_1_io_availability;
  wire                when_Stream_l408;
  reg                 _zz_io_push_valid;

  StreamFifo streamFifo_1 (
    .io_push_valid      (_zz_io_push_valid             ), //i
    .io_push_ready      (streamFifo_1_io_push_ready    ), //o
    .io_push_payload    (io_flowIn_payload             ), //i
    .io_pop_valid       (streamFifo_1_io_pop_valid     ), //o
    .io_pop_ready       (io_streamOut_ready            ), //i
    .io_pop_payload     (streamFifo_1_io_pop_payload   ), //o
    .io_flush           (1'b0                          ), //i
    .io_occupancy       (streamFifo_1_io_occupancy     ), //o
    .io_availability    (streamFifo_1_io_availability  ), //o
    .clk                (clk                           ), //i
    .resetn             (resetn                        )  //i
  );
  assign when_Stream_l408 = (! io_enable);
  always @(*) begin
    _zz_io_push_valid = io_flowIn_valid;
    if(when_Stream_l408) begin
      _zz_io_push_valid = 1'b0;
    end
  end

  assign io_streamOut_valid = streamFifo_1_io_pop_valid;
  assign io_streamOut_payload = streamFifo_1_io_pop_payload;

endmodule`

dahuwei, 2021-12-26:
Scala code:
[image]

Generated verilog code:
[image]

Dolu1990, 2021-12-26:
Hi ^^

To past code, you can do triple `

\`\`\`scala
myScalaCode
\`\`\`

About the issue itself,

There is one solution which works with the current released SpinalHDL : 

```scala
  case class Test(useQueue: Boolean = true) extends Component {
    val io = new Bundle {
      val enable = in Bool()
      val flowIn = slave(Flow(UInt(32 bits)))
      val streamOut = master(Stream(UInt(32 bits)))
    }
    val withQueue = useQueue generate new Area {
      val str = io.flowIn.toStream
      val streamInQueue = str.throwWhen(!io.enable).queue(256)
      io.streamOut << streamInQueue
    }
    val withoutQueue = !useQueue generate new Area {
      io.streamOut << io.flowIn.toStream.throwWhen(!io.enable)
    }
  }
  ``` 
  
  I just pushed a fix, which allow toStream to propagate names, which allow (in dev branch) : 
  
  ```scala
  case class Test(useQueue: Boolean = true) extends Component {
    val io = new Bundle {
      val enable = in Bool()
      val flowIn = slave(Flow(UInt(32 bits)))
      val streamOut = master(Stream(UInt(32 bits)))
    }
    val withQueue = useQueue generate new Area {
      val streamInQueue = io.flowIn.toStream.throwWhen(!io.enable).queue(256)
      io.streamOut << streamInQueue
    }
    val withoutQueue = !useQueue generate new Area {
      io.streamOut << io.flowIn.toStream.throwWhen(!io.enable)
    }
  }
  ```

dahuwei, 2021-12-28:
> Hi ^^
>
> To past code, you can do triple `
>
> ```scala myScalaCode ```
>
> About the issue itself,
>
> There is one solution which works with the current released SpinalHDL :
>
> ```scala
>   case class [...]

Hi Dolu
Flow.toStream can propagate names now. 
But the generated code use the propagated name, not the variable name "streamInQueue".
Is there any option to force Stream.queue() to use the variable name ?  
```
  val withQueue = useQueue generate new Area {
//    val streamInQueue = io.flowIn.toStream.throwWhen(!io.enable).queue(256)
    val streamInQueue = StreamFifo(UInt(32 bits), 256)
    streamInQueue.io.push << io.flowIn.toStream.throwWhen(!io.enable)
    io.streamOut << streamInQueue.io.pop
  }
```

Dolu1990, 2021-12-30:
The only way would be to manualy instanciate the fifo Component, or to add a argument to the queue function to say that the fifo should be named from the returned stream (but this isn't implemented yet)

godenfreemans, 2022-04-19:
[image]

Dolu1990, 2022-04-20:
@godenfreemans Ahhhh tricky tricky ^^, didn't thinked about it.

a similar way could be , Component.current.children.last.setDefinitionName("rawrrr") ^^

closed 2022-10-23 by numero-744