SpinalHDL/SpinalHDL #561: setXxxName usage ?

jijingg, 2021-12-27:
hi @Dolu1990 
is there any document or introduction of various `setname`  method ?
I often feel confused about those method and thier paramter：
``` setDefinitionName(name: String)
    setName(name: String)
    setName(name: String, namePriority: Byte)
    setName(name: String, weak :Boolean)
    setCompositeName(nameable: Nameable)
    setCompositeName(nameable: Nameable, weak: Boolean)
    setCompositeName(nameable: Nameable, namePriority: Byte)
....
    setPartialName(name: String)
    setPartialName(owner: Nameable, name: String, namePriority: Byte)
    setPartialName(owner: Nameable, name: String, weak: Byte)
....
```
What are the differences between them, what they are designe for ?

Dolu1990, 2021-12-30:
Hi ^^ 

There is very little bit of doc here : 
https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Structuring/naming.html?highlight=setcompositename#nameable-base-class

setName force the full name
setCompositeName allow to use another Nameable as prefix
setPartialName is used in SpinalHDL data structure, to use the thing which possess "this" as prefix.

weak = true =>  it is a proposal which will not override a stronger name source
A bit to namePriority

In general, as a user, both setName(string) and setCompositeName(other, string) are made to be used.

Also this was recently added : 
https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Structuring/naming.html#composite-in-a-function

jijingg, 2021-12-30:
> There is very little bit of doc here :
https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Structuring/naming.html?highlight=setcompositename#nameable-base-class
...
Also this was recently added :
https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Structuring/naming.html#composite-in-a-function

Great ! thanks

closed 2021-12-30 by jijingg

Dolu1990, 2022-01-05:
Just found some user usages to setPartialName("asdasd")
For instance if you use it on the element of a bundle, this will change the postfix to asdasd (instead of the val xxxx name)

Dolu1990, 2022-01-05:
Fixed, thanks ^^