SpinalHDL/SpinalHDL #563: add `.allowOverrideOnce` function?

jonnykl, 2021-12-29:
I want to define a default value for a signal without totally disabling the check for assignment overlaps. Using `.allowOverride` the check will be disabled totally for the signal. So would a `.allowOverrideOnce` function be possible?

Example:

I have a signal `x` which should default to `0`.

```scala
// define the signal and set default value
val x = Bits(2 bits)
x := 0
x.allowOverride

...

// assign x
x := 1

...

// accidentally assign x again
x := 2
// will pass all elaboration checks
```

I think it would be useful to have a `.allowOverrideOnce` function to allow the signal to be overriden only once.

Or is there already a way to do this?

jijingg, 2021-12-30:
@jonnykl  the right way maybe x.removeAssignment instead instead set x.allowOverride when your need override . it's more safe 
```
// define the signal and set default value
val x = Bits(2 bits)
x := 0

...

// assign x
x.removeAssignment()
x := 1

...

// accidentally assign x again
x := 2
// will pass all elaboration checks
```

jonnykl, 2022-01-12:
This does not work for me. If the signal is not assigned after the call to `removeAssignment`, SpinalHDL will raise a `NO DRIVER ON` error. I need a way to mark an assignment as default if no assignment to this signal follows.

Dolu1990, 2022-01-13:
@jonnykl there is the Data.default(Data) feature which would do that. Normaly it is used to have default value to component inputs, but, should work there too ?

jonnykl, 2022-01-13:
This seems to work only on signals, but I want to assign a default value to a component input:

```scala
import spinal.core._
import spinal.core.sim._
import spinal.sim._

case class Foo() extends Component {
  val io = new Bundle {
    val x = in Bool()
    val y = out Bool()
  }

  io.y := io.x
}

case class Test() extends Component {
  val io = new Bundle {
    val a = out Bool()
  }

  val foo = Foo()
  io.a := foo.io.y

  // ok
  //foo.io.x.default(True)
  //foo.io.x := False

  // NO DRIVER ON
  //foo.io.x.default(True)

  // ok
  //val x = Bool()
  //x.default(True)
  //foo.io.x := x
  //x := True

  // ok
  //val x = Bool()
  //x.default(True)
  //foo.io.x := x
}

object GenTest {
  def main(args: Array[String]): Unit = {
    SimConfig.compile(Test()).doSim { dut =>
    }
  }
}
```

Dolu1990, 2022-01-17:
Hooo the default feature was made to be called within the component which host the Data.

I just pushed a fix in dev

jonnykl, 2022-01-17:
This only fixes a part of the problem. Now using `.default()` outside the component it is used for works. But `.default()` still does not work when overwriting the signal conditionally:

```scala
case class Foo() extends Component {
  val io = new Bundle {
    val x = in Bool()
    val y = out Bool()
  }

  io.y := io.x
}

case class Test() extends Component {
  val io = new Bundle {
    val a = out Bool()
    val b = in Bool()
  }

  val foo = Foo()
  io.a := foo.io.y

  // ok
  //foo.io.x.default(True)
  //foo.io.x := False

  // fixed, now ok
  //foo.io.x.default(True)

  // LATCH DETECTED
  //foo.io.x.default(True)
  //foo.io.x.setWhen(io.b)

  // ok
  //val x = Bool()
  //x.default(True)
  //foo.io.x := x
  //x := True

  // ok
  //val x = Bool()
  //x.default(True)
  //foo.io.x := x

  // LATCH DETECTED
  //val x = Bool()
  //x.default(True)
  //foo.io.x := x
  //x.clearWhen(io.b)
}
```

Maybe this is the intended behavior of `.default()` but then there should be another function implementing my proposed behavior.

Dolu1990, 2022-01-20:
>   // LATCH DETECTED
  //val x = Bool()
  //x.default(True)
  //foo.io.x := x
  //x.clearWhen(io.b)
  
Yes it is expected

In which scénario  allowOverrideOnce  is wished ?

jonnykl, 2022-01-20:
I have a component with many inputs and want to set a default value because some of the inputs may be unused or are only set conditionally. Example:

```scala
case class ManyInputsComponent(n: Int) extends Component {
  val io = new Bundle {
    val inputs = in Vec(Bool(), n)
    ...
  }

  ...

  def ioDefaults(): Unit = {
    io.inputs.foreach(_.setDefaultValue(False))
    // setDefaultValue should assign the passed data (False) and allow it to be overwritten once

    ...
  }
}

case class TopLevelComponent() extends Component {
  val io = new Bundle {
    val a = in Bool()
  }

  val manyInputs = ManyInputsComponent(8)
  manyInputs.ioDefaults()

  manyInputs.io.inputs(0) := True
  manyInputs.io.inputs(1) := True
  manyInputs.io.inputs(7) := False

  when(io.a) {
    manyInputs.io.inputs(5) := True
  }
}
```

Without such a `setDefaultValue()` method I would have to do the following:

```scala

case class TopLevelComponent() extends Component {
  val io = new Bundle {
    val a = in Bool()
  }

  val manyInputs = ManyInputsComponent(8)
  
  // assign defaults
  //manyInputs.io.inputs(0) := False
  //manyInputs.io.inputs(1) := False
  manyInputs.io.inputs(2) := False
  manyInputs.io.inputs(3) := False
  manyInputs.io.inputs(4) := False
  manyInputs.io.inputs(5) := False
  manyInputs.io.inputs(6) := False
  //manyInputs.io.inputs(7) := False

  // (yes in this simple example the following would also be possible)
  manyInputs.io.inputs(2 until 7).foreach(_ := False)

  // -> I have to commet some default values out (or else SpinalHDL would throw an ASSIGNMENT OVERLAP error)
  // this is OK for a few inputs, but really annoying if you have many inputs

  manyInputs.io.inputs(0) := True
  manyInputs.io.inputs(1) := True
  manyInputs.io.inputs(7) := False

  when(io.a) {
    manyInputs.io.inputs(5) := True
  }
}
```

Dolu1990, 2022-01-20:
Ahhhhh, then what you want maybe the following : 

The ability to assign the input of the current component from within, not realy a allowOverrideOnce. 
Right ?

jonnykl, 2022-01-20:
No, `ioDefaults()` is just a helper method and gets only called outside the component it is defined in.

Example without `ioDefaults()`:

```scala
case class ManyInputsComponent(n: Int) extends Component {
  val io = new Bundle {
    val inputs = in Vec(Bool(), n)
    ...
  }

  ...
}

case class TopLevelComponent() extends Component {
  val io = new Bundle {
    val a = in Bool()
  }

  val manyInputs = ManyInputsComponent(8)

  io.inputs.foreach(_.setDefaultValue(False))
  // setDefaultValue should assign the passed data (False) and allow it to be overwritten once

  manyInputs.io.inputs(0) := True
  manyInputs.io.inputs(1) := True
  manyInputs.io.inputs(7) := False

  when(io.a) {
    manyInputs.io.inputs(5) := True
  }
}
```

But a way to assign input defaults within the component would also be a solution, maybe something like this:

```scala
case class ManyInputsComponent(n: Int) extends Component {
  val io = new Bundle {
    val inputs = in Vec(Bool(), n) setDefaultValue(Vec.fill(n)(False))
    ...
  }

  ...
}

case class TopLevelComponent() extends Component {
  val io = new Bundle {
    val a = in Bool()
  }

  val manyInputs = ManyInputsComponent(8)

  manyInputs.io.inputs(0) := True
  manyInputs.io.inputs(1) := True
  manyInputs.io.inputs(7) := False

  when(io.a) {
    manyInputs.io.inputs(5) := True
  }
}
```

Dolu1990, 2022-01-25:
hmm so this is quite a bit more tricky
So the best i could propose so far is a way to have a callback outside the given component to assign things.

numero-744, 2022-10-13:
Something like this?

```scala
class MyComponent extends Component {
  // Call it after instantiating if you want
  def assignDefaults(): Unit = { io.inputs defaults … }
  ...
}
```

closed 2022-10-26 by numero-744