SpinalHDL/SpinalHDL #568: ScopeProperty issue

jijingg, 2021-12-30:
@Dolu1990 , I found FixPointProperty not work properly aft doSimUntilVoid,  I'm not sure what happen in simcfg.compile().doSim
and I found you have rewirte ScopProperty and remove stack way 
```scala
  class Top extends Component
  FixPointConfig(RoundType.FLOOR, false){
    println(0, getFixRound()) //(0,FLOOR)   OK!
    FixPointConfig(RoundType.ROUNDUP, false){
      println(1, getFixRound()) //(0, ROUNDUP)   OK!
      SimConfig.compile(new Top).doSimUntilVoid("xx"){dut =>
        println(2, getFixRound()) //(2, ROUNTOINF) FAIL!  ,  except ROUNDUP but get FixPoint default value ROUNDTOINF
                                  //it's seems FixPointProperty back to default
      }
      FixPointConfig(RoundType.CEIL, false){
        println(3, getFixRound()) //(3, CEIL)  OK!
      }
      DefaultFixPointConfig{
        println(4, getFixRound()) //(4, ROUNDTOINF)  OK!
      }
    }
  }
```

Dolu1990, 2021-12-30:
Hi,

Yes the integrated stack behaviour was removed, and is now implemented via some the API itself.

Dolu1990, 2021-12-30:
I will check what happend with your example

commit 08670cf8 (2022-01-05): SimThread now restore the original ScopeProperty state
fix #568
closed 2022-01-05 by Dolu1990