SpinalHDL/SpinalHDL #569: Clock and Reset synthesis tags

kleinai, 2022-01-02:
Synthesis tags on clock and reset don't appear in the resulting VHDL. I do not know if this also effects the Verilog output.
My specific use case is attaching tags for Vivado's block design tool.
(There's actually a small second bug with the stream ready "bufferize" stage and synthesis tags)

Sample component:
```
class MyComponent extends Component {
  val io = new Bundle {
    val axis_in = slave(Stream(Bits(8 bit)))
    val axis_out = master(Stream(Bits(8 bit)))
  }

  io.axis_out << io.axis_in.stage()

  io.axis_in.valid.addAttribute("X_INTERFACE_INFO", "xilinx.com:interface:axis:1.0 axis_in TVALID")
  io.axis_in.ready.addAttribute("X_INTERFACE_INFO", "xilinx.com:interface:axis:1.0 axis_in TREADY")
  io.axis_in.payload.addAttribute("X_INTERFACE_INFO", "xilinx.com:interface:axis:1.0 axis_in TDATA")

  io.axis_out.valid.addAttribute("X_INTERFACE_INFO", "xilinx.com:interface:axis:1.0 axis_out TVALID")
  io.axis_out.ready.addAttribute("X_INTERFACE_INFO", "xilinx.com:interface:axis:1.0 axis_out TREADY")
  io.axis_out.payload.addAttribute("X_INTERFACE_INFO", "xilinx.com:interface:axis:1.0 axis_out TDATA")

  clockDomain.clock.addAttribute("X_INTERFACE_INFO", "xilinx.com:signal:clock:1.0 clock clk")
  clockDomain.reset.addAttribute("X_INTERFACE_INFO", "xilinx.com:signal:reset:1.0 reset rst")
  clockDomain.clock.addAttribute("X_INTERFACE_PARAMETERS", "ASSOCIATED_BUSIF axis_in:axis_out, ASSOCIATED_RESET reset")

  noIoPrefix()
}
```

Dolu1990, 2022-01-05:
Hi,

```scala
    clockDomain.readClockWire.addAttribute("X_INTERFACE_INFO", "xilinx.com:signal:clock:1.0 clock clk")
    clockDomain.readResetWire.addAttribute("X_INTERFACE_INFO", "xilinx.com:signal:reset:1.0 reset rst")
    clockDomain.readClockWire.addAttribute("X_INTERFACE_PARAMETERS", "ASSOCIATED_BUSIF axis_in:axis_out, ASSOCIATED_RESET reset")

```

Should fix it, basicaly, clockDomain.clock refer to the original clock, it may be somewere else in the hearchy (even outside the toplevel)
while clockDomain.readClockWire will pull the clock domain to your current location, and refer to that pulled signal

kleinai, 2022-01-06:
That does fix that issue. Thank you!
There is a strange second issue where buffered signals don't get their proper attributes. For example the `io.axis_in.ready` from my example.
I expected attributes on `axis_in_ready` but got attributes on `axis_in_ready_read_buffer` instead.

Dolu1990, 2022-01-06:
I don't understand, i have no axis_in_ready_read_buffer from your code example
For me the generated verilog is : 

```verilog
module MyComponent (
  (* X_INTERFACE_INFO = "xilinx.com:interface:axis:1.0 axis_in TVALID" *) input               axis_in_valid,
  (* X_INTERFACE_INFO = "xilinx.com:interface:axis:1.0 axis_in TREADY" *) output reg          axis_in_ready,
  (* X_INTERFACE_INFO = "xilinx.com:interface:axis:1.0 axis_in TDATA" *) input      [7:0]    axis_in_payload,
  (* X_INTERFACE_INFO = "xilinx.com:interface:axis:1.0 axis_out TVALID" *) output              axis_out_valid,
  (* X_INTERFACE_INFO = "xilinx.com:interface:axis:1.0 axis_out TREADY" *) input               axis_out_ready,
  (* X_INTERFACE_INFO = "xilinx.com:interface:axis:1.0 axis_out TDATA" *) output     [7:0]    axis_out_payload,
  (* X_INTERFACE_INFO = "xilinx.com:signal:clock:1.0 clock clk" , X_INTERFACE_PARAMETERS = "ASSOCIATED_BUSIF axis_in:axis_out, ASSOCIATED_RESET reset" *) input               clk,
  (* X_INTERFACE_INFO = "xilinx.com:signal:reset:1.0 reset rst" *) input               reset
);

  wire                axis_in_m2sPipe_valid;
  wire                axis_in_m2sPipe_ready;
  wire       [7:0]    axis_in_m2sPipe_payload;
  reg                 axis_in_rValid;
  reg        [7:0]    axis_in_rData;
  wire                when_Stream_l342;

  always @(*) begin
    axis_in_ready = axis_in_m2sPipe_ready;
    if(when_Stream_l342) begin
      axis_in_ready = 1'b1;
    end
  end

  assign when_Stream_l342 = (! axis_in_m2sPipe_valid);
  assign axis_in_m2sPipe_valid = axis_in_rValid;
  assign axis_in_m2sPipe_payload = axis_in_rData;
  assign axis_out_valid = axis_in_m2sPipe_valid;
  assign axis_in_m2sPipe_ready = axis_out_ready;
  assign axis_out_payload = axis_in_m2sPipe_payload;
  always @(posedge clk or posedge reset) begin
    if(reset) begin
      axis_in_rValid <= 1'b0;
    end else begin
      if(axis_in_ready) begin
        axis_in_rValid <= axis_in_valid;
      end
    end
  end

  always @(posedge clk) begin
    if(axis_in_ready) begin
      axis_in_rData <= axis_in_payload;
    end
  end

endmodule

``` 

Not the same for you ?

kleinai, 2022-01-06:
That all looks correct. Could you try generating VHDL from that sample?

Dolu1990, 2022-01-09:
Hoo right, so the reason of this is that the axis_in_ready  output signal is readed in the VHDL component, which isn't allowed by VHDL (before 2008), so to fix that, SpinalHDL is wrapping the signal into a buffer.

Is it a issue ?

kleinai, 2022-01-09:
I understand why it's buffering signals. I primarily use VHDL myself. The problem is that I've requested (or thought I requested) for attributes to be placed on the component I/O. Vivado looks at the I/O signals and their attributes for things like their block designs. Although I can't see how to differentiate that some attributes need to be on the unbuffered signal versus moving them to a buffered signal if generated.
I'll settle on thin wrappers for my top level components if I need them with port attributes.

Dolu1990, 2022-01-12:
Hmmm,
That's a tricky case, because from a abstract point of view, either the attribut is wished for the signal driven by some logic, either for the io itself, both case may make sense in some cases.

So, yes, either using a wrapper, either not reading the output signal. would workaround it.

closed 2022-01-13 by kleinai