SpinalHDL/SpinalHDL #571: Please, open a issue to discuss, before implementing changes

Dolu1990, 2022-01-05:


numero-744, 2022-10-14:
https://github.com/SpinalHDL/SpinalHDL/issues/892 is not pinned but it seems it should.

Could the title of this (#571) issue be added on top of https://github.com/SpinalHDL/SpinalHDL/issues/619 and the issue be deleted?

Dolu1990, 2022-10-17:
> https://github.com/SpinalHDL/SpinalHDL/issues/892 is not pinned but it seems it should.

Yes right XD i was sad when i have seen that there is a limitation for the number of pinned issues to 3 XD

> Could the title of this (https://github.com/SpinalHDL/SpinalHDL/issues/571) issue be added on top of https://github.com/SpinalHDL/SpinalHDL/issues/619 and the issue be deleted?

Hmm i think you are right. It can be done.

closed 2022-10-17 by Dolu1990