SpinalHDL/SpinalHDL #572: Axi4Downsizer Implementation.

Readon, 2022-01-07:
I am planning to submit my implementation of Axi4Downsizer which is missing compare to Axi4Upsizer.
It is targeting on supporting a backpresure-enabled Axi4 bus.
There are many stream based features has been adopted.

Both read and write only address bus involves splitting a transaction into several sub-transactions, so that an Axi4DownsizerSubTransactionGenerator is required. This module should compute the start address, sub-transaction number, size for each sub-transaction.
If the size of each beat of input transaction is smaller than the total size of one transaction for output, then the sub-transaction should have the same transaction len and size as inputs'. On other side, the transaction splitting is required, which can be implemented by StreamTransactionExtender.

Dolu1990, 2022-01-09:
Look good to me ^^

For a downSizer you can also relax timings a git by having some dead cycle between bursts, this may help the implementation.

Readon, 2022-01-10:
> Look good to me ^^
> 
> For a downSizer you can also relax timings a git by having some dead cycle between bursts, this may help the implementation.

It's werid that when I just name the StreamReady Randomizer as this [action](https://github.com/Readon/SpinalHDL/runs/4758906436?check_suite_focus=true) do it will cannel the test.

Readon, 2022-01-11:
> It's werid that when I just name the StreamReady Randomizer as this [action](https://github.com/Readon/SpinalHDL/runs/4758906436?check_suite_focus=true) do it will cannel the test.

The problem is limited to assign StreamReadyRandomizer to a variable like
```
val rDriver = StreamReadyRandomizer(r, clockDomain)
```
in Axi4Agent.scala then r will not be drived. This is only happened in Axi4ReadOnlyMasterAgent part, in Axi4WriteOnlyMasterAgent the similar assignment is good enough.

closed 2022-01-19 by Readon