SpinalHDL/SpinalHDL #573: Function to support assign the unassigned signals in bundle.

Readon, 2022-01-07:
The usual cases is that if a Axi4 bus need to be assigned from the other, then assignSomeByName companion with removeAssignments, and reassign the signals wanted is a good choice, like
```
      targetBundle.assignSomeByName(srcBundle)
      targetBundle.wantedSignal.removeAssignments()
      targetBundle.wantedSignal := newSignal
```

However, there are cases that srcBundle have slightly differences compared to targetBundle, such as dataWidth of data channels of Axi4. In that case, assignSomeByName would report errors. Assigning all signals manually is not a good idea, and the main reason I personally do not like verilog, so that an assignUnassignedByName is preferred to be used as
```
      targetBundle.wantedSignal := newSignal
      targetBundle.assignUnassignedByName(srcBundle)
```
where wantedSignal would not be assigned again.

Readon, 2022-01-08:
First, need to add hasDataAssignment to BaseType by foreach the statements in DslScopeStack.get and retrieve all DataAssignmentStatement to see if any statement is a data assigned one.
Second, add assignUnassginedByName method to Bundle.

Dolu1990, 2022-01-09:
Feature added :)

Works with : 

```scala
    val x = in(Rgb(8,7,6))
    val y = out(Rgb(8,7,6))

    y.g := 0
    y.assignUnassignedByName(x)
```

commit 53068e50 (2022-01-09): Add MultiData.assignUnassignedByName and BaseType.hasDataAssignment
fix  #573
closed 2022-01-09 by Dolu1990