SpinalHDL/SpinalHDL #575: `when` statement in Verilog format discuss

jijingg, 2022-01-08:
@Dolu1990  I see that the recent version has greatly improved the naming of variables. It's great. but on some point, I have some different views .
I found now spinal will give a new wire instead inline verilog expression. 
```scala
    when(io.wr && (io.addr === 0x3f00)){
      regs(0)  := io.wdata
    }
    when(io.wr && (io.addr === 0x3f01)){
      regs(1)  := io.wdata
    }
```
now spianl will give 
```verilog
  wire                when_play_l502;
  wire                when_play_l505;

  assign when_play_l502 = (io_wr && (io_addr == 16'h3f00));
  assign when_play_l505 = (io_wr && (io_addr == 16'h3f01));
  always @(posedge clk) begin
    if(when_play_l502) begin
      regs_0 <= io_wdata;
    end
    if(when_play_l505) begin
      regs_1 <= io_wdata;
    end
  end
```
sometimes it's useful for debug. bug most time the designer want print In situ, In particular, thousands of such cases will **generate too much  code lines** , so  Inline expression maybe a better choice  with better readability
```verilog
  always @(posedge clk) begin
    if(io_wr && (io_addr == 16'h3f00)) begin
      regs_0 <= io_wdata;
    end
    if(io_wr && (io_addr == 16'h3f01)) begin
      regs_1 <= io_wdata;
    end
  end
```
When the when condition is complex, let user  handle themselves ，and the Spinal should give WYSIWYG(what you see whant you get ) as much as possible
```scala
when(io_wr && (io_addr = 0x3f00) && ! io.wdata[1] balabala ....) {
}
```
Handle myself if i want
```scala
val hit_addr = io_wr && (io_addr = 0x3f00) 
val when_cond = hit_addr && !io.wdata[1] .....
when(when_cond){
}
```
at least I think there should an option on SpinalConfig  to chose which style be used.

Dolu1990, 2022-01-09:
Hi ^^

> at least I think there should an option on SpinalConfig to chose which style be used.

ok for me ^^

So the reason why the default is now to always generate when_xxx_lyyy statements is mainly because unless you explicitly refer to a single named signal, (when(when_cond)), the boolean used as condition will be unamed.
This maybe fine in some cases where things can be inligned, but for instance, as soon as your when statement mix register and combinatorial assignement, this will mess things up : 

```scala
    val x = False
    val y = Reg(False)

    val cond = True
    when(cond){
      x := True
      y := True
    }
    cond.unsetName()
```

```verilog
  wire                _zz_when;
  reg                 x;
  reg                 y;

  assign _zz_when = (1'b1 && 1'b1);
  always @(*) begin
    x = 1'b0;
    if(_zz_when) begin
      x = 1'b1;
    end
  end

  always @(posedge clk) begin
    if(_zz_when) begin
      y <= 1'b1;
    end
  end
```

Figuring out what is the _zz_when is a paine

note the case also happen if you assign multiple combinatorial signals (and no reg).

So by enforcing a name on the final bool, we also ensure that all anonymus signals which drive it will also get a name composed of _zz_when_xxx_lyyy. One reason where it realy help is in synthesis timing reports.

Unless we have a better solution, we add the option ?

jijingg, 2022-01-10:
sorry , mybe I 'm not get your point .  My doubt is，why directly generate inline
```verilog
  reg                 x;
  reg                 y;

  always @(*) begin
    x = 1'b0;
    if(1'b1 && 1'b1) begin
      x = 1'b1;
    end
  end

  always @(posedge clk) begin
    if( 1'b1 && 1'b1) begin
      y <= 1'b1;
    end
  end
```
are you worried about two of the same logic？I don't think this is a problem. WYSIWSG may be more important for read,
The synthesis tool is smart enough that it will optimize it.
So at the very least, If the user still doesn't trust the Synthesis-tool, you can handle it manually
```scala
val  my_cond = balabal & balala || babalba  .. 
when(my_cond){
}
```

Dolu1990, 2022-01-12:
> are you worried about two of the same logic？

Yes, not only twice, in some case it can be much more

> The synthesis tool is smart enough that it will optimize it.

This is very likely, yes, but may not be the case for all. (voodoo magic XD)

Not realy related, but was easy to check, there is some converted code from verilator : 
```scala
    x := 0
    y := 0
    z := 0
    when(a+b+c > 0x42){
      x := 0
      y := 1
      z := 2
    }
```

```verilog
  assign _zz_when_Play1_l65 = (_zz_when_Play1_l65_1 + c);
  assign _zz_when_Play1_l65_1 = (a + b);
  always @(*) begin
    x = 8'h0;
    if(when_Play1_l65) begin
      x = 8'h0;
    end
  end

  always @(*) begin
    y = 8'h0;
    if(when_Play1_l65) begin
      y = 8'h01;
    end
  end

  always @(*) begin
    z = 8'h0;
    if(when_Play1_l65) begin
      z = 8'h02;
    end
  end

  assign when_Play1_l65 = (8'h42 < _zz_when_Play1_l65);
```

```c++
    vlTOPp->x = 0U;
    if ((0x42U < (IData)(vlTOPp->unamed__DOT___zz_when_Play1_l65))) {
        vlTOPp->x = 0U;
    }
    vlTOPp->y = 0U;
    if ((0x42U < (IData)(vlTOPp->unamed__DOT___zz_when_Play1_l65))) {
        vlTOPp->y = 1U;
    }
    vlTOPp->z = 0U;
    if ((0x42U < (IData)(vlTOPp->unamed__DOT___zz_when_Play1_l65))) {
        vlTOPp->z = 2U;
    }
```

Funny that it duplicate the 0x42U <  xxx XD
(expected that it doesn't merge the if statements)

> So at the very least, If the user still doesn't trust the Synthesis-tool, you can handle it manually

It shouldn't be handled manualy, don't want poeple to have to bother about it, and things also need to be portable.

jijingg, 2022-01-14:
> It shouldn't be handled manualy, don't want poeple to have to bother about it, and things also need to be portable.

ok, I agree. for the  condition used more than once , spinal automatic handle this make sense,  It is reasonable.

A little bad， renaming all conds to `when_XX` may be a little radical. For some regular code, such as register interface, it will lead to large number of code,  shuttle  debug from thousands of lines of code is not a good experience.   

this case only use once.  still give `when_xxx` name
```scala
    when(wdata(0) && wr && waddr  === 0x3f0){
      reg1 :=  wdata
    }
    when(wdata(2)){
      reg3 :=  wdata
    }
```
```verilog 
  assign when_play_l530 = ((wdata[0] && wr) && (waddr == 16'h03f0));
  assign when_play_l533 = wdata[2];
  always @(posedge clk or posedge reset) begin
    if(reset) begin
      reg1 <= 32'h0;
      reg3 <= 32'h0;
    end else begin
      if(when_play_l530) begin
        reg1 <= wdata;
      end
      if(when_play_l533) begin
        reg3 <= wdata;
      end
    end
```

so , mybe we have tow choice 

- keep inline the case of cond only be used once.  an give new name of `when_xx` for case be used more than once 
- give an option on SpinalConfig to chose renaming cond or not .

Dolu1990, 2022-01-17:
Ultimately, it would require some decision to be made in the verilog/vhdl backend, because "often" expresion tree need to be split. For instance : 
```scala
    val a,b = in UInt(8 bits)
    val x = out (False)
    val cond = a+b > 10 //As a val, to make the later unsetName
    when(cond){
      x := True
    }
    cond.unsetName()
    ```
    
```verilog
  wire       [7:0]    _zz_when;

  assign _zz_when = (a + b); //Enforce bit width
  always @(*) begin
    x = 1'b0;
    if((8'h0a < _zz_when)) begin
      x = 1'b1;
    end
  end
  ```

jijingg, 2022-01-17:
`(a + b)`  is an exception. It is acceptable to create new wire for this.
But I still think **WYSIWYG** is a very sexy feature for HDL-DSL,  although it is impossible to achieve completely,  we should try best to keep that .  😃

closed 2022-05-03 by jijingg

Dolu1990, 2022-05-04:
There is also one bad aspect of the current implement (when_xxx_yyy)
Each time the scala code move a bit around, the generated netlist will change a bit in the naming of things (as it depend on scala line number XD)

So, i would say we can implement a SpinalHDL option which disable the feature. (can be usefull for coverage verification)

jijingg, 2022-05-04:
> #575

@Dolu1990  I love this option <3

another question .   is it easy to add option `inlineExpression: Boolean = true` to off expression, defualt is false.

I know your original intention is not to repeat the same logic， it's good . 
but If the user wants close it for some personal purpose, and spinal give an option to close that will be better . 

[image]

Dolu1990, 2022-05-04:
It is pretty easy to do. I just push another option, so now if you do nameWhenByFile = false, inlineConditionalExpression = true  it should follow what you wished for ^^

jijingg, 2022-05-04:
> I just push another option, so now if you do nameWhenByFile = false

great，it' work well.  thanks 
[image]