SpinalHDL/SpinalHDL #576: I2C Slave would send ACK even when the address didn't hit

Ncerzzk, 2022-01-09:
Hi, I use the I2C_Ctrl in spinalhdl lib in my project(which contains a i2c slave and a PWM component)

I enable the address filter and set the address of filter 0 to 0x20.

I use a soft master to test it. I send Start signal followed by the address byte(contains R/W bit).
when the address byte is 0x40 which is correct, the slave would pull down the SDA to ACK.

while when the address byte is 0x20 which is uncorrect, the slave would still pull down the SDA to ACK.
[image]

Ncerzzk, 2022-01-09:
Here is my code
```scala

package mylib

import spinal.core._
import spinal.lib._
import spinal.lib.bus.amba3.apb._
import spinal.lib.com.i2c._
import spinal.lib.fsm._
import spinal.lib.io.{InOutWrapper, ReadableOpenDrain}

import scala.util.Random
import spinal.sim._
import spinal.core.sim._
import spinal.lib.com.i2c.sim.OpenDrainInterconnect

object DutTests {
  def main(args: Array[String]): Unit = {
    SimConfig.withWave.withVerilator.compile {
      val a=new MyTopLevel
      a.i2c_apb.io.apb.simPublic()
      a.a.init.ok.simPublic()
      a
    }.doSim{ dut =>
      def sleep_half_period=sleep(300)
      def sleep_period = sleep(600)
      val sdaInterconnect=new OpenDrainInterconnect
      val sclInterconnect=new OpenDrainInterconnect
      sdaInterconnect.addHard(dut.i2c.sda)
      sclInterconnect.addHard(dut.i2c.scl)
      val sda = sdaInterconnect.newSoftConnection()
      val scl = sclInterconnect.newSoftConnection()

      def master_send_start()={
        scl.write(true)
        sleep_half_period
        sda.write(false)
        sleep_half_period
        scl.write(false)
      }

      def master_send_byte(data:Int)={
        for (i <- 7 downto 0){
          if((data & 1<<i)!=0){
            sda.write(true)
          }else{
            sda.write(false)
          }
          sleep_half_period
          scl.write(true )
          sleep_half_period
          scl.write(false)
        }
        sleep_half_period
        scl.write(true )
        sleep_half_period
        scl.write(false)
        sleep_half_period
        scl.write(true )
        sleep_half_period
        scl.write(false)
        sleep_half_period
        scl.write(true )
        sleep_half_period
        scl.write(false)
        sleep_half_period
        scl.write(true )
        sleep_half_period
        scl.write(false)
[... 305 lines cut ...]
    I2cSlaveMemoryMappedGenerics(
      ctrlGenerics = I2cSlaveGenerics(
        samplingWindowSize = 3,
        samplingClockDividerWidth = 10 bits,
        timeoutWidth = 20 bits
      ),
      addressFilterCount = 1
    )
  )
  val a = new PWM()
  a.apb <> i2c_apb.io.apb

  a.int := i2c_apb.io.interrupt
  i2c <> i2c_apb.io.i2c

}

//Generate the MyTopLevel's Verilog
object MyTopLevelVerilog {
  def main(args: Array[String]) {
    //InOutWrapper
    SpinalVerilog(new PWM).printPruned()
  }
}
```

Dolu1990, 2022-01-09:
Hi,

Are you sure the slave should ack a address which isn't targeted for him ?

I mean, what's about the case where there is multiple slaves on the line, the master would not be able to know if a slave feel concerned about the address, which would be a issue i guess ?

Ncerzzk, 2022-01-09:
> Hi,
> 
> Are you sure the slave should ack a address which isn't targeted for him ?
> 
> I mean, what's about the case where there is multiple slaves on the line, the master would not be able to know if a slave feel concerned about the address, which would be a issue i guess ?

Hi, I think the slave should answer when the address hit, while not answer when the address is not targeted for him.

While when I use the I2C_Ctrl as slave, the slave answered  even when the address is not targeted for him. So I think it's strange and open the issue. It's it by designed?

Dolu1990, 2022-01-12:
It is by design.

From : 
https://interrupt.memfault.com/blog/i2c-in-a-nutshell#acknack

"A NACK after an address is sent means no slave responded to that address"

Ncerzzk, 2022-01-12:
I'm sorry, but you may misunderstand my issue...

I agree that **the slave should send NACK when the address from master is incorrect**, 
but in the i2c lib(I2c_Ctrl), it seems to send ACK no matter address is correct or not when address filter is enable(as the pic I showed).

If the address filter is not enable, users should handle the tx_ack manually, and it's no problem. 

While in address filtering mode, I didn't find the place changing the value of tx_ack, which means that no matter hit or not, the slave would send the same tx_ack value.(so if we set ack_value=0(ACK) before, it would send ACK in HIT and NOT HIT).

```scala
val addressFilter = if(genAddressFilter) new Area {

      val addresses = Vec(Reg(I2cAddress()), addressFilterCount)
      ...

      val state        = RegInit(U"00")
      val byte0, byte1 = Reg(Bits(8 bits))
      val byte0Is10Bit = byte0(7 downto 3) === 0x1E

      when(rxData.event){
        switch(state){
          is(0){
            byte0 := rxData.value
            state := 1
          }
          is(1){
            byte1 := rxData.value
            state := 2
          }
        }
      }

      when(frameReset){
        state := 0
      }

      val hits = addresses.map(address => address.enable && Mux(!address.is10Bit,(byte0 >> 1) === address.value(6 downto 0) && state =/= 0, (byte0(2 downto 1) ## byte1) === address.value && state === 2))
      txAck.forceAck setWhen(byte0Is10Bit && state === 1 && addresses.map(address => address.enable && address.is10Bit && byte0(2 downto 1) === address.value(9 downto 8)).orR)
      busCtrlWithOffset.read(hits.asBits, 0x80, 0)
      busCtrlWithOffset.read(byte0.lsb,   0x84, 0)

      when(hits.orR.rise()){
        txAck.valid := False
      }
    } else null
```

Dolu1990, 2022-01-12:
Hoo right, sorry XD

So, i think the issue is that the txAck is written on the apb bus after the reset.

Instead it should only be written when you want to take over. on the second byte.

Does that make sense ?

[image]

Ncerzzk, 2022-01-12:
> Hoo right, sorry XD
> 
> So, i think the issue is that the txAck is written on the apb bus after the reset.
> 
> Instead it should only be written when you want to take over. on the second byte.
> 
> Does that make sense ?
> 
> [image]

Yes, since we enabled the address filter, of course we hope the hardware would handle the address hit ACK/NACK nicely.

Ncerzzk, 2022-01-12:
> 
Thank you very much! After remove the init ackTx register action at init, it works normally now.
[image]
[image]

Thanks again for your time.

closed 2022-01-12 by Ncerzzk

Dolu1990, 2022-01-12:
no worries ^^
Sorry for the confusion