SpinalHDL/SpinalHDL #581: workspacePath appears to be ignored by memory files

piegamesde, 2022-01-13:
Running the simulation puts some files like `SimulationName.v_toplevel_memory_ramBlocks_0_ram_symbol0.bin` into my directory. Can I somehow configure them away? I've already set `workspacePath`, yet they still appear.

I see that these files also exist in my simulation workspace path (under `SimulationName/rtl`), and they are identical to those in my working directory. So I guess this is a bug?

Dolu1990, 2022-01-17:
Should be all good now ^^

closed 2022-01-17 by Dolu1990

dnltz, 2022-01-26:
@Dolu1990  - Just tried the most recent version of SpinalHDL and want to let you know the rtl patching does not apply to files added with "addRTLPath". This might - for example - make trouble if someone wants to include a SPI NOR verilog model with code inside.

Dolu1990, 2022-02-01:
hmmm, i think in such case, one workaround is for the user to use absolute file path ?

commit 8631157d (2022-01-17): Fix #581, verilator backend no more copy rom bin files to the current directory.
This is done by patching the rtl files with some absolute path