SpinalHDL/SpinalHDL #583: Convenient and free bundle interconnection

jijingg, 2022-01-15:
Sometimes we don't want create a Class  of Bundle ， but Bundle some exits Wire  the connect each other
Similar to the usage in Verilog
```verilog
wire [15:0]  a;
wire [7:0]  b;
wire [4:0]  c;
wire [2:0]  d;
wire [9:0]  e;
wire [7:0]  f;
wire [13:0]  g;
assign {a, b, c, d} = 32'hFABC_32DD;
assign {a, b, c, d} = data[31:0] ;
assign {a, b, c, d} = {e, f, g} ;
```
what syntax we need in SpinalHDL：  
```scala
Bundle(a, b, c, d) := B(0xFABC_32DD, 32 bit)
Bundle(a, b, c, d) := io.data
Bundle(a, b, c, d) := Bundle(e, f ,g)
```
 here is an PR #582 for this feature . 

@Dolu1990  do you have any other suggestion ?

Dolu1990, 2022-01-17:
I just commented on #582 about the existing :

```scala
(a, b, c, d) := B(1832, 32 bit)
```

Dolu1990, 2022-01-17:
That does what you are aiming for right ?

jijingg, 2022-01-17:
> That does what you are aiming for right ?

```
(a, b, c, d) := B(1832, 32 bit)
```
yes,  tuples is good

closed 2022-01-17 by jijingg

Dolu1990, 2022-01-17:
Anyway, thanks for the PR ^^