SpinalHDL/SpinalHDL #584: is there any existing solution on creating regs with its value show ahead

Readon, 2022-01-16:
some times, signal need be latched by register for persistency. however there always be one clock delay for value updated. in many cases, the value is expected to be valid under latching condition.
adding a showAhead parameter would be fit for that.

Dolu1990, 2022-01-17:
Currently, it isn't implemented. The main thing is that it would require to "synthetise" the next value of the register with muxes, instead of keeping it abstracted behind when statements.

So either it would be implemented adding SpinalHDL phases for the selected signals, either we emulate it by using some abstractions.

Readon, 2022-01-19:
as I understand there will be two cases require showAhead feature.
1. RegNextWhen(signal, cond) when cond is true the signal should be appear on wires, this can be emulated by implementation of RegNextWhen.
2. RegInit(True) setWhen(cond1) clearWhen(cond2), cond1 will trigger rising of signal cond2 will make it fall, however if the signal is necessarily one clock ahead of latch, then implementation is not that clear for me.

Dolu1990, 2022-01-22:
What's about : 

```scala
case class RegMiaou[T <: Data](hardType : HardType[T]) extends Area {
   val value = Reg(hardType)
   val next = CombInit(value)
   value := next
}

val xxx = RegMiaou(UInt(8 bits))
```` 

 ?
 
 that way we keep things explicit and "emulated" Just would need a proper name instead of RegMiaou :p.

Readon, 2022-01-23:
> What's about :
>
> ```scala
> case class RegMiaou[T <: Data](hardType : HardType[T]) extends Area {
>    val value = Reg(hardType)
>    val next = CombInit(value)
>    value := next
> }
>
> val xxx = [...]

RegNextShowAhead?
Why not just modify RegNextWhen as below?
```
object RegNextWhen {
  def apply[T <: Data](next: T, cond: Bool, init: T = null.asInstanceOf[T], showAhead: Boolean = false)(implicit loc: Location): T = {
    val out = cloneOf(T)
    val reg = Reg(next, init).setCompositeName(next, "regNextWhen", true)
    out := reg
    when(cond){
      reg := next
      if(showAhead) out := next
    }
    out
  }
}
```

Dolu1990, 2022-01-25:
> Why not just modify RegNextWhen as below?

Because you lose the possibility of construct things with more complex conditions

Readon, 2022-02-06:
> What's about :
>
> ```scala
> case class RegMiaou[T <: Data](hardType : HardType[T]) extends Area {
>    val value = Reg(hardType)
>    val next = CombInit(value)
>    value := next
> }
>
> val xxx = [...]

Why the last line value := next is defined? It seems that the next is always equal to value?
What is the usage of RegMiaou?
in this way?
``` scala
when(cond) xxx.next := yyyy
```

Readon, 2022-02-27:
> What's about :
>
> ```scala
> case class RegMiaou[T <: Data](hardType : HardType[T]) extends Area {
>    val value = Reg(hardType)
>    val next = CombInit(value)
>    value := next
> }
>
> val xxx = [...]

A new area seems should be used with wire name inside it.
as when(cond) xxx.next := yyy
I have tried to define "object RegMiaou " instread, the output verilog defines xxx.next as register.

Readon, 2022-02-27:
> I have defined a RegAhead object as below.
> 
> ```scala
> object RegAhead {
>     def apply[T <: Data](hardType: HardType[T]): T = {
>         val value = Reg(hardType)
>         val next  = CombInit(value)
>         value := next
>         value.setCompositeName(next, "value", true)
>         next
>     }
> }
> ```
> 
> and use an individual class to test it.
> 
> ```scala
> case class RegAheadFixture() extends Component {
>      val io = new Bundle {
>         val enable = in Bool ()
>         val input  = in UInt (8 bits)
>         val output = out UInt (8 bits)
>     }
>     val dut = RegAhead(UInt(8 bits))
>     when(io.enable) {
>         dut := io.input
>     }
>     io.output := dut
> }
> ```
> 
> Here when clause is required to avoid combination overlaps.
> 
> there are still problems that:
> 
> 1. if the setCompositeName correct here, would it confuse other componets?
> 2. if the when clause in the fixture is necessary, why not just wrap them together. Is that because there are still swtich case also?

RegNextAhead or RegWire might be a better name, because it act like a wire but can persistent it's value until the next changes.

Dolu1990, 2022-02-28:
Damned sorry i missed the issue response.

>  It seems that the next is always equal to value?

Until the user conditional assign "next"

> def apply[T <: Data](hardType: HardType[T]): T = {

The issue with this, is that then the user lose access the the register, so either return a tuple with the reg and the next, either a area

> if the setCompositeName correct here, would it confuse other componets?

Seems good to me, what do you mean by confuse with others components ?

> RegNextAhead or RegWire might be a better name, because it act like a wire but can persistent it's value until the next changes.

RegNextAhead is good to me ^^

So one thing is that i'm now looking maybe a way to have this natively supported by the SpinalHDL compilation, i will let you know if it's done successfuly ^^

Readon, 2022-02-28:
> Damned sorry i missed the issue response.
> 
> > It seems that the next is always equal to value?
> 
> Until the user conditional assign "next"
Which lead to a longer express to use, that why I tried to use object to implement it.

> 
> > def apply[T <: Data](hardType: HardType[T]): T = {
> 
> The issue with this, is that then the user lose access the the register, so either return a tuple with the reg and the next, either a area
IMO, If the register access is preferred, RegNext might be a better choice for user?

> 
> > if the setCompositeName correct here, would it confuse other componets?
> 
> Seems good to me, what do you mean by confuse with others components ?
Yes, I do not know if setCompositeName would avoid name conflict between components automatically.

> RegNextAhead is good to me ^^
>
> So one thing is that i'm now looking maybe a way to have this natively supported by the SpinalHDL compilation, i will let you know if it's done successfuly ^^
If yes, that would be best. There are also showAhead FIFO in some FPGA toolchain.

Dolu1990, 2022-02-28:
> Yes, I do not know if setCompositeName would avoid name conflict between components automatically

Ahh, as long as the weak arguement is set to true, (as you did) that's fine. 

So, anyway, i pushed the native support of the feature in spinalhdl :)
It do not use any emulation layer as we were planning to do, instead it transform the spinalhdl graph.
check the demo : 
https://github.com/SpinalHDL/SpinalHDL/commit/21ccbf6c92baebd95a323c8490b8808bcfbdcda0#diff-c8d8334432f582379e88477911f56e592c50f98a235d1bacf49a434b16021311R51

So overall, that solution should be the less intrusive.

Remains how we want to name the API. the regNextValue maybe missunderstand with the actual RegNext API.

So i think i will change it to myReg.getAheadValue

Dolu1990, 2022-02-28:
So for : 

```scala
      val conds = in(Vec.fill(8)(Bool()))
      val a, b, c, d, e = out(Reg(UInt(8 bits)) init(0))

      when(conds(0)){
        a := 1
        when(conds(1)){
          a := 2
          b := 11
        } otherwise {
          a := 3
          c := 21
        }
      }
      when(conds(2)){
        a := 4
        b := 12
        c := 22
        d := 31
      }

      val x = out(a.getAheadValue)
      val y = out(a.getAheadValue)
      val z = out(b.getAheadValue)

      when(d.getAheadValue() === 0){
        e := 1
      }
```

Note the generated verilog : 

```verilog
module unamed (
  input               conds_0,
  input               conds_1,
  input               conds_2,
  input               conds_3,
  input               conds_4,
  input               conds_5,
  input               conds_6,
  input               conds_7,
  output reg [7:0]    a,
  output reg [7:0]    b,
  output reg [7:0]    c,
  output reg [7:0]    d,
  output reg [7:0]    e,
  output reg [7:0]    x,
  output     [7:0]    y,
  output reg [7:0]    z,
  input               clk,
  input               reset
);

  reg        [7:0]    d_aheadValue;
  wire                when_CoreMiscTester_l56;

  always @(*) begin
    d_aheadValue = d;
    if(conds_2) begin
      d_aheadValue = 8'h1f;
    end
  end

  always @(*) begin
    z = b;
    if(conds_0) begin
      if(conds_1) begin
        z = 8'h0b;
      end
    end
    if(conds_2) begin
      z = 8'h0c;
    end
  end

  always @(*) begin
    x = a;
    if(conds_0) begin
      x = 8'h01;
      if(conds_1) begin
        x = 8'h02;
      end else begin
        x = 8'h03;
      end
    end
    if(conds_2) begin
      x = 8'h04;
    end
  end

  assign when_CoreMiscTester_l56 = (d_aheadValue == 8'h0);
  assign y = x;
  always @(posedge clk or posedge reset) begin
    if(reset) begin
      a <= 8'h0;
      b <= 8'h0;
      c <= 8'h0;
      d <= 8'h0;
      e <= 8'h0;
    end else begin
      if(conds_0) begin
        if(!conds_1) begin
          c <= 8'h15;
        end
      end
      if(conds_2) begin
        c <= 8'h16;
      end
      if(when_CoreMiscTester_l56) begin
        e <= 8'h01;
      end
      a <= x;
      b <= z;
      d <= d_aheadValue;
    end
  end

endmodule

```

Readon, 2022-03-01:
Good work!

closed 2022-03-01 by Readon