SpinalHDL/SpinalHDL #585: Did I make a mistake?

Echo-b, 2022-01-16:
When I try to write a circuit using spinalHDL,the function of this circuit is to select one of in1 and in2 based on the res signal but the compiled Verilog code is not what I want it to be by using `sbt` 
```
//the src code
import spinal.core._
import spinal.lib._

class MUX21 extends Component{
  val io = new Bundle{
    val in1 = in Bool()
    val in2 = in Bool()
    val sel = in Bool()
    val res = out Bool()
  }
  if(io.sel == true){
    io.res := io.in1
  }else{
    io.res := io.in2
  }
}

object MUX {
  def main(args:Array[String]){
    SpinalVerilog(new MUX21)
  }
}

//the compiled Verilog code
module MUX21 (
  input               io_in1,
  input               io_in2,
  input               io_sel,
  output              io_res
);

  assign io_res = io_in2;

endmodule

//i want
module MUX21 (
  input               io_in1,
  input               io_in2,
  input               io_sel,
  output              io_res
);
    assign io_res = (io.sel)?io.in1:io.in2;

endmodule

```

jiegec, 2022-01-17:
You need to use `when` and `otherwise`. Please go through the tutorial

Dolu1990, 2022-01-17:
then, io.sel === True in the when statement

Echo-b, 2022-01-17:
> You need to use `when` and `otherwise`. Please go through the tutorial

thank you very much for your advice,I think that I must to go through the tutorial carefully.

Echo-b, 2022-01-17:
> then, io.sel === True in the when statement

Thank you, I have solved the problem

closed 2022-01-17 by Dolu1990