SpinalHDL/SpinalHDL #591: RegNext does not work with implicit parameters

ronan-lashermes, 2022-01-21:
Hi,
I am still evaluating the language and found another bug (?), or at least a missing feature: RegNext(X) does not work if X constructor requires an implicit parameter.
## Reproducing example
```scala
package debug

import spinal.core._

trait Xlen {
    def xlen: Int
}

case class MemoryAddress()(implicit xlenConfig: Xlen) extends Bundle {
    val address = UInt(xlenConfig.xlen bits)
}

class DebugMemory(implicit config: Xlen) extends Component {
    val io = new Bundle {
        val inputAddress = in(MemoryAddress())
    }   

    val someAddress = RegNext(io.inputAddress) // -> ERROR *****************************
    //val someAddress = Reg(MemoryAddress()) // -> This alternative is working
    //someAddress := io.inputAddress
}
```
The testbench file below
```scala
package tests

import spinal.core._
import spinal.sim._
import spinal.core.sim._
import debug._

case class XlenConfig(val xlen: Int) extends Xlen {}

object DebugMemoryTest {

    def main(args: Array[String]): Unit = {
       
        SimConfig.withWave.doSim(new DebugMemory()(XlenConfig(32))){dut =>
            //Fork a process to generate the reset and the clock on the dut
            dut.clockDomain.forkStimulus(period = 10)

            for(i <- 0 to 100) {
                dut.clockDomain.waitRisingEdge()
            }
        }
    }
}
```

Error is
```
[error] *** Spinal can't clone class debug.MemoryAddress datatype
```

Dolu1990, 2022-01-23:
Hi

Right, so the workaround is : 

```scala
case class MemoryAddress()(implicit xlenConfig: Xlen) extends Bundle {
  val address = UInt(xlenConfig.xlen bits)

  override def clone = MemoryAddress()
}
```

Basicaly, RegNext need to create type copy of the given signal, but can't handle the implicit thing by himself.

In general, one alternative to implicit is to use ScopeProperty : 

```scala
  object Xlen extends ScopeProperty[Int]

  case class MemoryAddress() extends Bundle {
    val address = UInt(Xlen.get bits)
  }

  class DebugMemory() extends Component {
    val io = new Bundle {
      val inputAddress = in(MemoryAddress())
    }

    val someAddress = RegNext(io.inputAddress)
  }

  Xlen.set(32)
  SimConfig.withWave.doSim(new DebugMemory()){dut =>
    //Fork a process to generate the reset and the clock on the dut
    dut.clockDomain.forkStimulus(period = 10)

    for(i <- 0 to 100) {
      dut.clockDomain.waitRisingEdge()
    }
  }
```

ScopeProperty are a bit like global variable, but are local to the current thread, and can be applied on local portion of the code.

I just pushed some doc about it : 

https://github.com/SpinalHDL/SpinalDoc-RTD/commit/013fd712bde559397e631e2e2c122132d1c414c0

That example work in the dev branch, with 1.6.1 it will not work i think.

will release 1.6.2 soon

ronan-lashermes, 2022-01-24:
Ok, thank you that solves it !
For the record, I tried to override the clone function but made a mistake: I assigned the address **content** in the new cloned MemoryAddress. This result in a assignment overlap error. I will update the doc for the clone override.

closed 2022-01-24 by ronan-lashermes

Dolu1990, 2022-01-25:
>  I will update the doc for the clone override.

Thanks :D