SpinalHDL/SpinalHDL #592: Can I define a register in When statement?

Ncerzzk, 2022-01-22:
I found that It's possible to define a signal in when/switch statement here: https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Semantic/when_switch.html#local-declaration

and then define a reg in When statement, and then assign it to a new value, while the generated hdl code seems to be not correct:
```scala
class Test extends Component{
    val active =False
    val c = False
    when(active){    
        val state=RegInit(U("000"))
        when(state===0){
            state:=1
            c:=True
        }
    }
}
```

generated:

```verilog
module Test (
  input               clk,
  input               reset
);
  wire                _zz_2;
  wire                active;
  reg                 c;
  reg        [2:0]    _zz_1;

  assign _zz_2 = (_zz_1 == 3'b000);
  assign active = 1'b0;
  always @ (*) begin
    c = 1'b0;
    if(active)begin
      if(_zz_2)begin
        c = 1'b1;
      end
    end
  end

  always @ (posedge clk or posedge reset) begin
    if (reset) begin
      _zz_1 <= 3'b000;
    end else begin
      if(_zz_2)begin
        _zz_1 <= 3'b001;   // Here,  the "active" signal seems disappear
      end
    end
  end
endmodule
```

But if I defined the "state" reg out of When statement, it behave normally:

```verilog
  always @ (posedge clk or posedge reset) begin
    if (reset) begin
      state <= 3'b000;
    end else begin
      if(active)begin
        if(_zz_1)begin
          state <= 3'b001;
        end
      end
    end
  end
```

Is it correct? It seems that we can not define a reg inside the when statement?

Dolu1990, 2022-01-23:
Hi,

it is the expected behaviour.

Basicaly, when you define new signal/reg, they will only see the world through the scope they are defined in, so ignoring the when(active) in your code is the expected behaviour.

Reason of that is mostly to allow calling functions which abstract thing and generate hardware, even within some when statement

ex 

```scala
when(xxx){
   when(myBundle.myFunction()){ //Here i want myFunction to not care about the xxx, as that's a function only relative to myBundle)
   
   }
}
```

Ncerzzk, 2022-01-23:
Thank you for the explanation, I got it.

closed 2022-01-23 by Ncerzzk

typingArtist, 2022-01-28:
To me the question is: Is the code example valid? Then why doesn’t the implementation follow POLA and generate the expected output? If the code is not valid, it should not create output at all and instead raise an error. The generated code doesn’t make sense IMHO.

Dolu1990, 2022-01-28:
> To me the question is: Is the code example valid ?

Yes it is ^^

> generate the expected output

Is it a typo ? "his" expected output ? Or the expected output is the same than his expected output ?

> Then why doesn’t the implementation follow POLA 

Here is a practical example why : 

```scala
when(something){
   when(signal.rise()){  //The rise function will  implement a register sampling each cycle the signal value to compare it with the current value
   
   }
}
```

So, basicaly, it is a choice. The current implementation allow to call abstracted behaviour within conditional context without being impacted by the parent context.

This allow a range of usages.
I'm not sure the other way around would have many usecase ?