SpinalHDL/SpinalHDL #593: How to extend the BaseTypeFactory with new defined XFactory

yenchih, 2022-01-22:
Hello there,
I just define a new type of data structure, and wish to add it to module IO.
However, currently the BaseTypeFactory only extended with the data type defined in the spinalhdl.
``` in BaseType.scala
trait BaseTypeFactory extends BoolFactory with BitsFactory with UIntFactory with SIntFactory with VecFactory with SFixFactory with UFixFactory
```
Is there any way that I can add the new designed factory into BaseTypeFactory?

Thanks a lot.
BR 
Joe

Dolu1990, 2022-01-23:
hmm currently, you coud be able to : 

```scala
val io = new Bundle {
  val xxx = in(MyNewType)
}
```

That BaseTypeFactory is mostly there to allow the "in MyNewType()" syntax

I'm not aware of a way to workaround it.

yenchih, 2022-01-23:
OK, Thanks. What you mention is actually the problem I want to solve. 
I did try the in(MyNewType()) syntax and it did work perfectly. 
So the issue here is simply about how to make the syntax "in MyNewType()" work.
Though it is not an urgent issue, still be great to know if there is any solution available.
Thanks a lot :)

closed 2022-01-23 by yenchih