SpinalHDL/SpinalHDL #595: doubt about StreamJoin and similar fundamental components on combination logic of valid and ready.

Readon, 2022-01-26:
As discussed in [AXI Handshaking Rules](https://zipcpu.com/blog/2021/08/28/axi-rules.html), several rules for valid/ready signal in Axi4 bus is concluded, which seems also fit the Stream related component's design.
There is one rule that "On master and slave interfaces, there must be no combinatorial paths between input and output signals." make confusion with StreamJoin.

in StreamJoin the code
``` scala
   ....
    combined.valid := sources.map(_.valid).reduce(_ && _)
    sources.foreach(_.ready := combined.fire)
   ....
```
the ready signal is assigned to a combined fire which is a logic combination of master valids && target ready. This makes master valid contribute to the slave's ready.
The combination logic would cause a long critical path in the design which fail STA offten.
It would also be great to validate this while elaborating.

Dolu1990, 2022-01-27:
Hi :)

> There is one rule that "On master and slave interfaces, there must be no combinatorial paths between input and output signals."

My opinon about this AXI rule, if applied to interconnect/bridges, is that is tyranny, lazyness or nativity, as it would require buffering things everywhere, costing latency / area usage / suboptimal design.

The phylosophy in the spinal.lib.Stream is to always implement things as minimal as possible, a let people pipeline things where they want with the m2sStage / s2mStage / halfPipe functions

I would say, i general guideline is to not have a streamX.valid + streamX.payload depending of the streamX.ready.

The only case i had where that guideline make sense to be transgressed maybe the StreamFork, in some cases.

Readon, 2022-01-27:
good policy. then it might be reasonable to support m2s pipe like function to buses to help with cutting logic path.

Dolu1990, 2022-02-21:
> I am planning on the </< operator to Axi4 bus, which represent

One reason why it wasn't implemented, is that it is kind of too specific, and often overkill.

Typicaly, you can use the halfPipe instead of ar and aw to reduce area (half bandwidth)

Maybe we should pick a more general aproache, as your pipeReady suggest, but a more options ? 

For instance for the Stream class there is : 

```scala
  def pipelined(m2s : Boolean = false,
                s2m : Boolean = false,
                halfRate : Boolean = false) : Stream[T] = {
    (m2s, s2m, halfRate) match {
      case (false,false,false) => this.combStage()
      case (true,false,false) =>  this.m2sPipe()
      case (false,true,false) =>  this.s2mPipe()
      case (true,true,false) =>   this.s2mPipe().m2sPipe()
      case (false,false,true) =>  this.halfPipe()
    }
  }
```

So having a single fonction in axi which cover it for each link

Readon, 2022-02-21:
> 

Then pipeReady and pipePayload would be better. In some case, they can be cascaded to get payload and ready piped also.
On other side, they can be used individually. 

also, halfRate arg can be given to represent an attitude of optimization, performance or area?
``` scala
pipePayload(toWriteCmd=false, toWriteData=false, toWriteRsp=false, toReadCmd=false, toReadRsp=false, halfRate=false): Axi4

//usage
axiOut << axiIn.pipeReady(toWriteCmd=true).pipePayload(toWriteCmd=true)
```

Dolu1990, 2022-02-21:
but what's about 

```scala
def pip(awM2s : Boolean, awS2m : Boolean, awHalfPipe, awM2s : Boolean,  ...
``` 

it would be all in one ^^

> also, halfRate arg can be given to represent an attitude of optimization, performance or area?

Sure : 

m2s => payload bits FF, only cut valid + payload path
s2m => payload bits FF + payload bits mux, only cut ready path
halfPipe => 1+payload bits FF, cut all path, half bandwidth

Readon, 2022-02-27:
> ```scala
> awM2s : Boolean, awS2m : Boolean, awHalfPipe,
> ```

I tried to implement once, the problem is there are too many arguments should be take care.
``` scala
def pipelined(
            awM2s: Boolean = false,
            awS2m: Boolean = false,
            awHalfRate: Boolean = false,
            wM2s: Boolean = false,
            wS2m: Boolean = false,
            wHalfRate: Boolean = false,
            bM2s: Boolean = false,
            bS2m: Boolean = false,
            bHalfRate: Boolean = false,
            arM2s: Boolean = false,
            arS2m: Boolean = false,
            arHalfRate: Boolean = false,
            rM2s: Boolean = false,
            rS2m: Boolean = false,
            rHalfRate: Boolean = false
        ): Axi4 = {
            val out = Axi4(from.config)
            out.aw << from.aw.pipelined(awM2s, awS2m, awHalfRate)
            out.w << from.w.pipelined(wM2s, wS2m, wHalfRate)
            out.b.pipelined(bM2s, bS2m, bHalfRate) >> from.b
            out.ar << from.ar.pipelined(arM2s, arS2m, arHalfRate)
            out.r.pipelined(rM2s, rS2m, rHalfRate) >> from.r
            out
        }
```

How about define a enumerate with five option, then there would be only five argments.
``` scala
object PipeType extends SpinalEnum {
  val noPipe, m2sPipe, s2mPipe, fullPipe,halfPipe = newElement()
}
pipeline (awConfig : PipeType = PipeType.noPipe .....) : Axi4 {
     out.aw << from.aw.pipelined(awConfig.getTuple())
}
```

Dolu1990, 2022-02-28:
> the problem is there are too many arguments should be take care

A problem for scala itself ? or a problem because it look like a borderline usage ^^ ? 

> the problem is there are too many arguments should be take care

The enumeration solution sound good for me.
Just do not use Hardware enumeration (SpinalEnum) to have a elaboration time enumeration.

Instead : 

```scala
trait PipeType
object PipeType{
   val NONE, M2S, S2M, FULL, HALF = new PipeType
}
```

Also, i'm open for it to be implemented as an alternative Stream.pipeline function, which is then used in the Axi stuff :D

Readon, 2022-02-28:
> > the problem is there are too many arguments should be take care
> 
> A problem for scala itself ? or a problem because it look like a borderline usage ^^ ?

It seems that there should be only at most five arguments each time we use the Axi4 bus, in the "split version".
The only problem is the argument list is too long when define such function.

> The enumeration solution sound good for me. Just do not use Hardware enumeration (SpinalEnum) to have a elaboration time enumeration.
>
> Instead :
>
> ```scala
> trait PipeType
> object PipeType{
>    val [...]

is that possible to add enum like.
``` scala
trait PipeType
object PipeType {
    val None = (false, false, false)
    val M2S = (true, false, false)
    val S2M = (false, true, false)
    val FULL = (true, true, false)
    val HALF = (false, false, true)
}
```

so that the stream can be pipelined with one more style
``` scala
def pipelined(config: PipeType): Stream[T] {
     val new = cloneOf(this)
     new << (this.pipelined _).tupled(config)
}
```

Dolu1990, 2022-02-28:
> is that possible to add enum like.

Probably, yes, but now that you bring that, i think the best is just to go full in, and provide a even more flexibly way : 

```scala
    trait StreamPipe{
      def apply[T <: Data](m : Stream[T]) : Stream[T]
    }

    object StreamPipe{
      val NONE = new StreamPipe{
        override def apply[T <: Data](m: Stream[T]) = m.combStage()
      }
      val M2S = new StreamPipe{
        override def apply[T <: Data](m: Stream[T]) = m.m2sPipe()
      }
      ...
    }

    class Stream ...{
      def pipelined(pipe: StreamPipe) = pipe(this)
      def pipelined(m2s : Boolean = false,
                    s2m : Boolean = false,
                    halfRate : Boolean = false) : Stream[T] = {
        (m2s, s2m, halfRate) match {
          case (false,false,false) => StreamPipe.NONE(this)
          case (true,false,false) =>  StreamPipe.M2S(this)
          case (false,true,false) =>  StreamPipe.S2M(this)
          case (true,true,false) =>   StreamPipe.FULL(this)
          case (false,false,true) =>  StreamPipe.HALF(this)
        }
      }
    }
```

This way you can define new pipelining ways, and provide them as parameter too. It make some sence in the Stream class, but a lot more sense in the AXI class.

Would that be good ?

Readon, 2022-02-28:
> 

That would be great!
However, I have searched the pipelined function in the source code, where the AvalonMM and Wishbone bus both have pipelined function seems does not consistent with Bmb bus and Stream.
Is it necessary to keep this consistency?

BTW, there are functions that should called only from a master, such as the toAxi4ReadOnly for Axi4 and m2sPipe for Stream. 
How about add this as a traits or explicitly add the state to the table of functions?

Readon, 2022-02-28:
> 

``` scala
def pipelined(aw: StreamPipe, w: StreamPipe, b: StreamPipe, ar: StreamPipe, r: StreamPipe): Axi4 = {
            val out = cloneOf(from)
            out.aw << from.aw.pipelined(aw)
            out.w << from.w.pipelined(w)
            out.b.pipelined(b) >> from.b
            out.ar << from.ar.pipelined(ar)
            out.r.pipelined(r) >> from.r
            out
        }
```
Is the name of StreamPipe fit better than StreamPipeKind? and the name of arguments aw should be renamed to awKind? awType?

Maybe, it's best to distribute a naming guideline for the lib.

Readon, 2022-03-01:
> > where the AvalonMM and Wishbone bus both have pipelined function seems does not consistent
> 
> Right, so mostly, those two buses aren't Steam compliant

Then the strategy is to ignore those two buses?

> 
> > Is it necessary to keep this consistency?
> 
> Between stream and Axi ?

I think it's all Stream based buses.

> 
> > How about add this as a traits
> 
> This can be usefull yes, to have for instance Flow and Stream both able to being staged.
> 
> > explicitly add the state to the table of functions?
> 
> which state ? which table ?

That is another story I will open a new issue.

Readon, 2022-03-04:
It's partially implemented by #623.
Further work on Bmb will break the existing implementation, so that part should be hanged on currently.

closed 2022-03-04 by Readon