SpinalHDL/SpinalHDL #597: SystemDebugger has no reset for JTAG area

dnltz, 2022-01-27:
This clocked area has no reset and the JtagFsm uses `randBoot`. Turning this feature off leads to a undefined start value.

https://github.com/SpinalHDL/SpinalHDL/blob/5d819d2873da452a73541c0c4e315e2808dc9449/lib/src/main/scala/spinal/lib/system/debugger/SystemDebugger.scala#L78

My idea is to add an optional `ClockDomainConfig` parameter to `SystemDebuggerConfig`. If this parameter is set, the clock domain will use the parent reset with the given config. Since this CD is driven by tck, which will not toggle at the start, the parent's clock config can't be used. This is why the jtag CD should be configurable.

Other solution: Add trst and add this as reset signal. Since the jtag protocol defines how a TAP should be resetted, a static ClockDomainConfig can be used.

dnltz, 2022-01-27:
@Dolu1990  - What do you think?

Dolu1990, 2022-01-28:
I think this is the exact same case than for the counters of the other issue, this randBoot is there to avoid X propagation. But in itself, it isn't usefull from a design perspective.

Basicaly, the JTAG as a specific sequance which will make the FSM go in reset state, no mather which which state it is. If i remember well, that's 5 TMS high cycles.

The idea of jtag is that it should work even if the system has no clock / no reset.

> Other solution: Add trst and add this as reset signal. Since the jtag protocol defines how a TAP should be resetted, a static ClockDomainConfig can be used.

As long it is optional, that's fine to me.

More generaly, X state propagation is pissing me off, as it is a simulation feature which endup changing how we design things XD

So the question is : 
What would be the Verilog/VHDL feature which would allow to handle both RTL and simulation ?

dnltz, 2022-01-31:
Ah yes, you're right. Jtag works fine without randBoot! I only get connection errors when using symbiflow. Vivado works fine. Super strange ...

Mhh I think randBoot is fine. It might be dangerous in verification steps because it generates slightly different RTL.

About this topic: Since it works without trst, I would like to focus on other things. Keep to chat about this first :)

closed 2022-01-31 by dnltz

Dolu1990, 2022-02-04:
> Mhh I think randBoot is fine. It might be dangerous in verification steps because it generates slightly different RTL.

Yes i agree, idealy we would need to randomize the signal for real, this would allow verification steps to catch issues.

Maybe using things as $urandom in verilog