SpinalHDL/SpinalHDL #598: [Bug] There should be a check on every signal/instance that they don't share name with module

rkhat, 2022-01-28:
If you use setDefinitionName with the same name of a signal/instance, you can get unsynthesizable code. You can also do it with setName on a signal/instance.

Example below where the module and instance are both named test1.

SpinalHDL code:
```
case class Test1() extends Component {
  setDefinitionName("test1")
  val din   = in  Bool()
  val dout  = out Bool()
  dout := din
}

case class Test0() extends Component {
  setDefinitionName("test0")
  val din   = in  Bool()
  val dout  = out Bool()
  val test1 = Test1()
  test1.din <> din
  test1.dout <> dout
}
```

Generated verilog code:
```
`timescale 1ns/1ps 

module test0 (
  input               din,
  output              dout
);

  wire                test1_dout;

  test1 test1 (
    .din     (din         ), //i
    .dout    (test1_dout  )  //o
  );
  assign dout = test1_dout;

endmodule

module test1 (
  input               din,
  output              dout
);

  assign dout = din;

endmodule
```

rkhat, 2022-01-28:
Nevermind it is fixed on dev

jijingg, 2022-01-28:
> same name of a signal/instance, you can get unsynthesizable code

Why not?  Which synthesis-tool you used? Although we don't recommend it， the same name of moudle/instance should be synthesizable .

ramielkhatibPQS, 2022-01-28:
You are actually right. I just assumed it was an error when I transitioned from VHDL to Verilog. And, for VHDL, you put `work.modulename` to avoid that error over there.

rkhat, 2022-01-28:
You are actually right. I just assumed it was an error when I transitioned from VHDL to Verilog. And, for VHDL, you put `work.modulename` to avoid that error over there.

closed 2022-01-28 by rkhat