SpinalHDL/SpinalHDL #599: Feature request: switch case factorization

ronan-lashermes, 2022-01-28:
# Motivation

Get more compact code when using the *switch* construct.
Example: extracting immediate values in RISC-V instructions.
**Today:**
```scala
switch(aluop) {
    is(ALUOp.add) {
        immediate := instruction.immI.signExtend
    }
    is(ALUOp.slt) {
        immediate := instruction.immI.signExtend
    }
    is(ALUOp.sltu) {
        immediate := instruction.immI.signExtend
    }
    is(ALUOp.sll) {
        immediate := instruction.shamt
    }
}
```
**Desired:**

```scala
switch(aluop) {
    is(ALUOp.add || ALUOp.slt || ALUOp.sltu) {
        immediate := instruction.immI.signExtend
    }
    is(ALUOp.sll) {
        immediate := instruction.shamt
    }
}
```

# Suggested implementation
You are obviously more familiar with the codebase, but it doesn’t seem too complicated with my limited experience of SpinalHDL.

- Implement a new class in Enum.scala: `SpinalEnumElementList` that contains the list of alternatives in the `is` clause.
- Add a `def ||(el: SpinalEnumElement)` to `SpinalEnumElement` and to `SpinalEnumElementList` to build the list of alternatives.
- Add a `def ===(list: SpinalEnumElementList)` to `SpinalEnumCraft` that or_reduce the `===` comparisons to all alternatives in the list.

I am not sure of the quality of the generated VHDL/Verilog code at the end.

If you are interested in this proposal and don’t see obvious errors in the suggested implementation, I can give it a try.

Dolu1990, 2022-01-28:
Hi ^^

Isn't is(ALUOp.add, ALUOp.slt, ALUOp.sltu) a working syntax ?

ronan-lashermes, 2022-01-28:
Totally working for me... Didn’t look enough in the sources. Closing the issue, but I will propose a new doc update then ^^

closed 2022-01-28 by ronan-lashermes

Dolu1990, 2022-01-28:
No worries ^^