SpinalHDL/SpinalHDL #600: ClockDomain doesn't get restored automatically from child

rkhat, 2022-01-28:
If you run below code, then the clockDomain doesn't automatically get restored after exiting the child module. You'll need to manually call the restore at the end of the child module.

```
case class TestParent() extends Component {
  val clkParent = in Bool()
  val rstParent = in Bool()

  ClockDomainStack.set(ClockDomain(clock=clkParent,reset=rstParent))
  
  println("Has reset1? " + ClockDomainStack.get.hasResetSignal)
  val testChild = TestChild()
  println("Has reset2? " + ClockDomainStack.get.hasResetSignal)
  testChild.clkChild  <> clkParent
}
                       
case class TestChild() extends Component {
  val clkChild = (in Bool()).setName("clkChild")
  val ctx = ClockDomainStack.set(ClockDomain(clock=clkChild))
  // ctx.restore()
}
```

Maybe change the Component postInitCallback to this to automatically restore it:

```
def postInitCallback(): this.type = {
  prePop()
  ClockDomain.push(clockDomain) // add this line
  DslScopeStack.set(parentScope)
  this
}
```

closed 2022-02-01 by Dolu1990

rkhat, 2022-02-18:
I just tested version 1.6.4 and ` ClockDomainStack.set(...)` is still broken in the child component when `privateNamespace` in `SpinalConfig` is enabled. The parent will port map the clock prefixed with the child's name (This signal doesn't exist lol).

The way I was renaming the clock is below. I adapted the code from `mapClockDomain` method in `BlackBox` class. I would recommend simply moving the `mapClockDomain` and `mapCurrentClockDomain` methods from `BlackBox` class into `Component` class. Then simply put the `if (parent != null)` wrapper on the method. This way both `Component` class and `BlackBox` class (by inheritance) have it.

```
  def setClockDomainName(clkName: String, rstName: String = null): Unit = {
    val clk = (in Bool()).setName(clkName)
    val rst = if (rstName != null) (in Bool()).setName(rstName) else null

    if (parent != null) {
      val context = Component.push(parent)
      pulledDataCache += (clockDomain.clock -> clk)
      clk := ClockDomainStack.get.readClockWire
      if (rstName != null) {
        pulledDataCache += (clockDomain.reset -> rst)
        rst := ClockDomainStack.get.readResetWire
      }
      context.restore()
    }
    ClockDomainStack.set(ClockDomain(clk, rst))
  }
```

Dolu1990, 2022-02-21:
Can you provide a minimal test example to reproduce ? To be sure i understand.

Thanks :)

rkhat, 2022-02-22:
I just realized the issue is not related to the privateNamespace. It can be reproduced with or without it. Here's an example:

```
case class TestParent() extends Component {
  val clkParent = in Bool()
  val rstParent = in Bool()

  ClockDomain.push(ClockDomain(clock=clkParent,reset=rstParent))
  val child = TestChild()
}
                       
case class TestChild() extends Component {
  val clkChild = in Bool()
  val rstChild = in Bool()
  ClockDomain.push(ClockDomain(clock=clkChild,reset=rstChild))
}

object Builder {
  def main(args: Array[String]) {
    SpinalConfig(mode = Verilog).generate(TestParent())
  }
}

```

Output (notice how the child's clock signals in the parent module have child prefixed).

```
module TestParent (
  input               clkParent,
  input               rstParent
);

  wire                child_clkChild;
  wire                child_rstChild;

  TestParent_TestChild child (
    .clkChild    (child_clkChild  ), //i
    .rstChild    (child_rstChild  )  //i
  );

endmodule

module TestParent_TestChild (
  input               clkChild,
  input               rstChild
);
```

rkhat, 2022-02-22:
Here's how I fixed the issue. Basically, move the 2 methods `mapClockDomain` and `mapCurrentClockDomain` from `BlackBox` class to `Component` class and add the suggested changes.

```
// These 2 functions are copy pasted from BlackBox class except the first line and last line of mapClockDomain
abstract class MyComponent extends Component {
    /**
    * Map clock domain signals (clock, reset, enable) to a clockDomain
    */
  def mapClockDomain(clockDomain: ClockDomain = ClockDomain.current, clock: Bool = null, reset: Bool = null, enable: Bool = null, resetActiveLevel: Polarity = HIGH, enableActiveLevel: Polarity = HIGH): Unit = {
    if (parent != null) {
      val context = Component.push(parent)

      if (clockDomain.hasClockEnableSignal && enable == null) SpinalError(s"Clock domain has clock enable, but blackbox is not compatible $this")

      if (enable != null) {
        pulledDataCache += (clockDomain.clockEnable -> enable)
        enable := (if(enableActiveLevel != clockDomain.config.clockEnableActiveLevel) !clockDomain.readClockEnableWire else clockDomain.readClockEnableWire)
      }

      if (reset != null) {
        if (!clockDomain.hasResetSignal) SpinalError(s"Clock domain has no reset, but blackbox need it $this")
        pulledDataCache += (clockDomain.reset -> reset)
        reset := (if(resetActiveLevel != clockDomain.config.resetActiveLevel) !clockDomain.readResetWire else clockDomain.readResetWire)
      }

      pulledDataCache += (clockDomain.clock -> clock)
      clock := clockDomain.readClockWire

      context.restore()
    }
    // you will need to add the resetActiveLevel and enableActiveLevel. I suggest maybe having a config argument instead.
    ClockDomain.push(ClockDomain(clock=clock, reset=reset, clockEnable=enable))
  }

  /** Map clock domains signal to the current ClockDomain */
  def mapCurrentClockDomain(clock: Bool, reset: Bool = null, enable: Bool = null, resetActiveLevel: Polarity = HIGH, enableActiveLevel: Polarity = HIGH): Unit = {
    mapClockDomain(ClockDomain.current, clock, reset, enable, resetActiveLevel, enableActiveLevel)
  }
}

case class TestParent() extends MyComponent {
  val clkParent = in Bool()
  val rstParent = in Bool()

  // ClockDomain.push(ClockDomain(clock=clkParent,reset=rstParent))
  mapCurrentClockDomain(clkParent, rstParent)
  val child = TestChild()
}
                       
case class TestChild() extends MyComponent {
  val clkChild = in Bool()
  val rstChild = in Bool()
  // ClockDomain.push(ClockDomain(clock=clkChild,reset=rstChild))
  mapCurrentClockDomain(clkChild, rstChild)
}
```

Dolu1990, 2022-02-24:
I would say that :

```scala
    .clkChild    (clkParent  ), //i
    .rstChild    (rstParent  )  //i
```

is wrong, why clkParent ? where was made that connection ?

rkhat, 2022-02-24:
The `ClockDomain.current` grabs the parent's clock domain, but I have a mistake. This will only work for the first call as the `ClockDomain.current` will change once you push a new clockdomain. The parent's clock domain needs to be stored during the `Component` constructor. I looked through the `Component` code and it seems it is already stored in the variable `clockDomain`

```
abstract class Component extends ... {
...
  /** Get the current clock domain (null if there is no clock domain already set )*/
  val clockDomain = ClockDomain.currentHandle
...
}
```

My suggestion is to copy the `mapClockDomain` and `mapCurrentClockDomain` as is from `BlackBox` class to `Component` class and then add a function `setCurrentClockDomain`, which creates and pushes a new clock domain that is mapped to the parent's clock domain.

```
abstract class MyComponent extends Component {
  
  /**
    * Creates and pushes a new clock domain mapped to the parent's current clock domain
    */
  def setCurrentClockDomain(clock: Bool = null, reset: Bool = null, enable: Bool = null, resetActiveLevel: Polarity = HIGH, enableActiveLevel: Polarity = HIGH): Unit = {
   // clockDomain has the parent's clock domain
    if (parent != null) {
      mapClockDomain(clockDomain, clock, reset, enable, resetActiveLevel, enableActiveLevel)
    }
    ClockDomain.push(ClockDomain(
      clock       = clock, 
      reset       = reset, 
      clockEnable = enable, 
      config      = ClockDomainConfig(resetActiveLevel=resetActiveLevel, clockEnableActiveLevel=enableActiveLevel)
    ))
  }
  
  /**
    * Map clock domain signals (clock, reset, enable) to a clockDomain
    */
  def mapClockDomain(clockDomain: ClockDomain = ClockDomain.current, clock: Bool = null, reset: Bool = null, enable: Bool = null, resetActiveLevel: Polarity = HIGH, enableActiveLevel: Polarity = HIGH): Unit = {
    val context = Component.push(parent)

    if (clockDomain.hasClockEnableSignal && enable == null) SpinalError(s"Clock domain has clock enable, but blackbox is not compatible $this")

    if (enable != null) {
      pulledDataCache += (clockDomain.clockEnable -> enable)
      enable := (if(enableActiveLevel != clockDomain.config.clockEnableActiveLevel) !clockDomain.readClockEnableWire else clockDomain.readClockEnableWire)
    }

    if (reset != null) {
      if (!clockDomain.hasResetSignal) SpinalError(s"Clock domain has no reset, but blackbox need it $this")
      pulledDataCache += (clockDomain.reset -> reset)
      reset := (if(resetActiveLevel != clockDomain.config.resetActiveLevel) !clockDomain.readResetWire else clockDomain.readResetWire)
    }

    pulledDataCache += (clockDomain.clock -> clock)
    clock := clockDomain.readClockWire

    context.restore()
  }

  /** Map clock domains signal to the current ClockDomain */
  def mapCurrentClockDomain(clock: Bool, reset: Bool = null, enable: Bool = null, resetActiveLevel: Polarity = HIGH, enableActiveLevel: Polarity = HIGH): Unit = {
    mapClockDomain(ClockDomain.current, clock, reset, enable, resetActiveLevel, enableActiveLevel)
  }
}
```

Dolu1990, 2022-02-25:
hmmm,

>  The parent's clock domain needs to be stored during the Component constructor

So you mean that when exiting a component, it should restor the initial clock domain of the parent component ? not the clock domain the component was created in ?

rkhat, 2022-02-25:
By parent clock domain, I meant the clock domain the component was created in. I should have said the parent's current clock domain.

> So you mean that when exiting a component, it should restor the initial clock domain of the parent component ?

When exiting the component, the clock domain the component was created in should be restored. That's what I suggested earlier in this issue and was fixed [in here](https://github.com/SpinalHDL/SpinalHDL/commit/1dddd072760868b4c6b0b637ee312e2f5436372f). Apparently, I already knew that `clockDomain` was the parent's clock domain but forgot about it (Brain fart :)).

This new suggestion is to be able to set a new clock domain inside the child and wire it to the clock domain the component is created in. This will allow us to code directly at the class scope without wrapping everything inside a `ClockingArea`.

Dolu1990, 2022-02-28:
> This new suggestion is to be able to set a new clock domain inside the child and wire it to the clock domain the component is created in.

But would that creating exceptions to the rules ? I mean, in general i'm trying to keep things "ortogonals".

Why wanting to create a new clockdomain inside the child, and having it waire the the clock domain the component is created in ? 

Why not directly taking the "clock domain the component is created in" directly then ?

rkhat, 2022-02-28:
The idea is to have control over the naming of the clock domain without having to wrap the whole code with a clocking area each time. Also, based on what I've seen, the clock domain inherits its name from the clock domain the component is created in. So, if you happen to rename the clock domain at the top level (for example to synchronize the reset signal), now every component down the tree has a clock domain with the name of that new clock domain.

As for why would one need control over the naming.
- Simulation: When you want to test a specific sub-component directly using Verilog/VHDL, you need to know the name of the clock domain. I wouldn't want to have to guess the clock domain's name.
- A lot of us are switching over from Verilog/VHDL. I want to be able to quickly test out the legacy code (as a drop-in replacement) which needs a matching clock domain's name.
- Same as above but basically a lot of code out there is written in Verilog/VHDL.
- I created an initial Component with its VHDL/Verilog test bench. Now, I am integrating the component into a bigger project. I'll need to keep the clock domain's name fixed and not inherited from its parent.

Dolu1990, 2022-03-01:
> The idea is to have control over the naming of the clock domain without having to wrap the whole code with a clocking area each time

Ahhhhh 

So, so far, i never had that case, mostly because when i wanted to test a specific component of my design, i was generating it, instead of reusing a bigger netlist which had that given component in.

There would be a easy way to set the name of the clock signal for a given component. Would that be good enough ?

rkhat, 2022-03-01:
Setting the names of the clock and reset should handle most but not all cases. Full control of the clock domain will handle all cases especially when you have components with no reset signals and components with multiple clock domains.

Dolu1990, 2022-03-09:
hmmm, so, i think such things should not be manualy handeled by the user, but instead be a change into the way SpinalHDL pull the block domains automaticaly through the hearchy

Dolu1990, 2022-03-16:
Should be all good now with : 

```scala
  new Component{
    val sub1 = new Component{
      val sub2 = new Component{
        ClockDomain.current.renamePulledWires(clock = "rawrr", reset = "miaou")
        val sub3 = new Component {
          val x = RegNext(False) init(True)
        }
      }
    }
  }
```

Is it good ?

rkhat, 2022-03-17:
Sorry, I was very busy the past week. Thank you for getting it working. I see what you mean by keeping it "orthogonal."

I don't think the renamePulledWires function is needed. We can directly use the underlying code. You can keep it, but, in my opinion, it adds redundancy.

```
  new Component{
    val sub1 = new Component{
      val sub2 = new Component{
        ClockDomain.current.readClockWire.setName("rawrr")
        ClockDomain.current.readResetWire.setName("miaou")
        val sub3 = new Component {
          val x = RegNext(False) init(True)
        }
      }
    }
  }
```

Also, the other issue I had before can easily be solved below after the fix in https://github.com/SpinalHDL/SpinalHDL/commit/1dddd072760868b4c6b0b637ee312e2f5436372f. If I am creating my own clocks and reset inputs, I should wire them in the parent using `ClockDomain.current`.

```scala
case class TestParent() extends Component {
  val child = TestChild()
  child.clkChild <> ClockDomain.current.readClockWire
  child.rstChild <> ClockDomain.current.readResetWire

  val s = RegNext(False)

}
                       
case class TestChild() extends Component {
  val clkChild = in Bool()
  val rstChild = in Bool()
  ClockDomain.push(ClockDomain(clkChild, rstChild))
  
  val s = RegNext(False)
}
```

Dolu1990, 2022-03-17:
> I don't think the renamePulledWires function is needed.

Right, it isn't needed, that's just some API to "ease" the process ^^

commit 1dddd072 (2022-02-01): Fix #600 Component postInitCallback now enforce the clockdomain
commit f52abbe6 (2022-03-16): Fix #600 ClockDomain Pulled signal now construct a chain of name depedences, allowing renaming at one level and having that new name propagated downstream.

Add ClockDomain.renamePulledWires