SpinalHDL/SpinalHDL #601: Feature (or doc) request : optional values in Bundle

ronan-lashermes, 2022-02-01:
#Motivation
It is useful to add extra signals in Bundle depending on config value. A common use case is to add extra signal upon a "Debug" flag.
Chisel does that with Option (cf https://www.chisel-lang.org/chisel3/docs/cookbooks/cookbook.html#how-do-i-create-an-optional-io).

Is there anything to achieve a similar feature ? It is important to have a compilation or elaboration error if we consume a signal when it should not be possible (e.g. using a debug signal in production build).

wswslzp, 2022-02-04:
Three ways: 
1. use ìfGen`
```scala
val condGen = ifGen(flag) {in Bool()}
```
2. use `generate`
```scala
val condGen = flag generate {in Bool()}
```
3. use `null`
```scala
val condGen = if (flag) in Bool() else null
```

Dolu1990, 2022-02-04:
Hi,

Ahhh i was writing this message while [wswslzp](https://github.com/wswslzp) was doing the same ^^

So
I'm not realy fan of " val out2 = if (flag) Some(Output(UInt(12.W))) else None"  as it is very verbose. Also option in general, i'm not realy in love with them, because there is most of the time some explicit flag / parameters associated (the "flag" in this case), which make things kind of redondant.

But option are supported aswell in SpinalHDL : 

```scala
    val a = Some(Bool())
    val x = new Bundle {
      val b = Some(Bool())
    }
    val y = new Area {
      val c = Some(Bool())
    }
```

```scala
  wire                a;
  wire                x_b;
  wire                y_c;
```

Also, a null entering anypoint in the SpinalHDL API will generate a error, so it is "safe" to use

Dolu1990, 2022-02-04:
Done ^^

closed 2022-02-04 by Dolu1990

ronan-lashermes, 2022-02-04:
Perfect, that’s totally what I was looking for. Thanks

dxsm, 2022-02-06:
Is there any plan to perfect the spinalhdl document? I think there are many nice features are not recorded. Then guide the contributor to supplement the document with the checklist below like chisel for new PR. If this can be done, a virtuous circle can be formed.

**Contributor Checklist**
￼[ ] Did you add Scaladoc to every public function/method?
[ ]￼ Did you add at least one test demonstrating the PR?
[ ]￼ Did you delete any extraneous printlns/debugging code?
[ ]￼ Did you specify the type of improvement?
￼[ ] Did you add appropriate documentation in docs/src?
￼[ ] Did you state the API impact?
[ ]￼ Did you specify the code generation impact?
￼[ ] Did you request a desired merge strategy?
[ ]￼ Did you add text to be included in the Release 
[ ]Notes for this change?
**Type of Improvement**
bug fix
**API Impact**
TruthTable will accept unknown width input now.

**Backend Code Generation Impact**
None

**Desired Merge Strategy**
Squash: The PR will be squashed and merged (choose this if you have no preference.
Release Notes
TruthTable will accept unknown width input now.

**Reviewer Checklist (only modified by reviewer)**
[ ]￼ Did you add the appropriate labels?
[ ]￼ Did you mark the proper milestone (Bug fix: 3.4.x, [small] API extension: 3.5.x, API modification or big change: 3.6.0)?
[ ]￼ Did you review?
[ ]￼ Did you check whether all relevant Contributor checkboxes have been checked?
[ ]￼ Did you mark as Please Merge?

Dolu1990, 2022-02-07:
> Is there any plan to perfect the spinalhdl document?

So, not realy, excepted PR are welcome.

>  I think there are many nice features are not recorded.

That's right. Probably at this stage "we" should take all the scala files which implement general stuff and check. 

> Then guide the contributor to supplement the document with the checklist below like chisel for new PR

As long as it is not too heavy handed and discouraging sure ^^, as many time, contributions are made by people not confortable with the whole git flow / read the doc.

Dolu1990, 2022-02-24:
@dxsm I pinned a checklist issue : 
https://github.com/SpinalHDL/SpinalHDL/issues/619