SpinalHDL/SpinalHDL #602: ConcurrentModificationException because of Scopt

piegamesde, 2022-02-02:
I updated my `scopt` dependency and this somehow triggered a ConcurrentModificationException within the internals of SpinalHDL:

```
Exception in thread "main" java.util.ConcurrentModificationException: mutation occurred during iteration
	at scala.collection.mutable.MutationTracker$.checkMutations(MutationTracker.scala:43)
	at scala.collection.mutable.CheckedIndexedSeqView$CheckedIterator.hasNext(CheckedIndexedSeqView.scala:47)
	at scala.collection.IterableOnceOps.foreach(IterableOnce.scala:563)
	at scala.collection.IterableOnceOps.foreach$(IterableOnce.scala:561)
	at scala.collection.AbstractIterable.foreach(Iterable.scala:926)
	at spinal.core.internals.SwitchStatement.$anonfun$normalizeInputs$1(Statement.scala:457)
	at spinal.core.internals.SwitchStatement.$anonfun$normalizeInputs$1$adapted(Statement.scala:457)
	at scala.collection.IterableOnceOps.foreach(IterableOnce.scala:563)
	at scala.collection.IterableOnceOps.foreach$(IterableOnce.scala:561)
	at scala.collection.AbstractIterable.foreach(Iterable.scala:926)
	at spinal.core.internals.SwitchStatement.bitVectorNormalize$1(Statement.scala:457)
	at spinal.core.internals.SwitchStatement.normalizeInputs(Statement.scala:508)
	at spinal.core.internals.PhaseNormalizeNodeInputs.$anonfun$impl$112(Phase.scala:1327)
	at spinal.core.internals.PhaseNormalizeNodeInputs.$anonfun$impl$112$adapted(Phase.scala:1323)
	at spinal.core.internals.ScopeStatement.$anonfun$walkStatements$1(Statement.scala:150)
	at spinal.core.internals.ScopeStatement.$anonfun$walkStatements$1$adapted(Statement.scala:148)
	at spinal.core.internals.ScopeStatement.foreachStatements(Statement.scala:143)
	at spinal.core.internals.ScopeStatement.walkStatements(Statement.scala:148)
	at spinal.core.internals.PhaseContext.$anonfun$walkStatements$1(Phase.scala:129)
	at spinal.core.internals.PhaseContext.$anonfun$walkStatements$1$adapted(Phase.scala:129)
	at spinal.core.internals.GraphUtils$.walkAllComponents(Misc.scala:268)
	at spinal.core.internals.PhaseContext.walkStatements(Phase.scala:129)
	at spinal.core.internals.PhaseNormalizeNodeInputs.impl(Phase.scala:1323)
	at spinal.core.internals.PhaseContext.doPhase(Phase.scala:181)
	at spinal.core.internals.SpinalVerilogBoot$.$anonfun$singleShot$12(Phase.scala:2564)
	at spinal.core.internals.SpinalVerilogBoot$.$anonfun$singleShot$12$adapted(Phase.scala:2562)
	at scala.collection.IterableOnceOps.foreach(IterableOnce.scala:563)
	at scala.collection.IterableOnceOps.foreach$(IterableOnce.scala:561)
	at scala.collection.AbstractIterable.foreach(Iterable.scala:926)
	at spinal.core.internals.SpinalVerilogBoot$.$anonfun$singleShot$7(Phase.scala:2562)
	at spinal.core.ScopeProperty$.sandbox(ScopeProperty.scala:66)
	at spinal.core.internals.SpinalVerilogBoot$.singleShot(Phase.scala:2499)
	at spinal.core.internals.SpinalVerilogBoot$.apply(Phase.scala:2494)
	at spinal.core.Spinal$.apply(Spinal.scala:371)
	at spinal.core.SpinalConfig.generateVerilog(Spinal.scala:164)
	at spinal.core.sim.SpinalSimConfig.compileCloned(SimBootstraps.scala:633)
	at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:617)
	…
```

I can reproduce this by merely adding `implementation 'com.github.scopt.scopt:scopt_2.13:f5ecbf3b8b'` as dependency to my project that uses Spinal (version 1.6.1).

Also, this made me wonder: why does Spinal even depend on a CLI argument parser? If it has some CLI parts, would it be possible to factor these out in a way such that the library part does not depend on `scopt`?

Dolu1990, 2022-02-04:
Look like you got the issue fixed by : 
https://github.com/SpinalHDL/SpinalHDL/commit/5e6896ae32d9fb2902e22490b5518e542d67f3af

Did you had that commit in the SpinalHDL you used ?

piegamesde, 2022-02-04:
No, I haven't tested that commit. I've seen that issue and the fix, and couldn't imagine how that issue could be triggered by adding a dependency elsewehere and decided that these probably weren't related. I'll give it a try.

closed 2022-02-08 by piegamesde