SpinalHDL/SpinalHDL #608: Add setReadyWhen for Stream

rkhat, 2022-02-09:
There seems to be no way to derive an output stream from an input stream such that:
```
output.valid := input.valid
output.payload := input.payload
input.ready := output.ready && externalCond
```
There is a `continueWhen` which blocks both valid and ready and `clearValidWhen` which blocks valid only. There is `setBlocked`  but this won't allow chaining like this:
```
val output = input.m2sPipe().setReadyWhen(externalCond)
```

Here is the function I am suggesting to add to Stream:
```
def setReadyWhen(cond : Bool): Stream[T] = {
  val next = Stream(payloadType).setCompositeName(this, "setReadyWhen", true)
  next.valid := this.valid
  next.payload := this.payload
  this.ready := next.ready && cond
  next
}
```

Also, in my opinion, `clearValidWhen` should be replaced with `setValidWhen` so that there is `continueWhen`, `setValidWhen`, and `setReadyWhen` all three have positive conditions.

rkhat, 2022-02-09:
Actually take out `.setCompositeName` because SpinalHDL will automatically use the output's name. Same for `clearValidWhen`/`setValidWhen`

Dolu1990, 2022-02-10:
Ahh you want to duplicate transactions of a stream right ? Kind of a forking it in a serial way ?

rkhat, 2022-02-10:
Yes, pretty much. I have a multi-stage stream. The second stage needs to keep duplicating the transaction until a certain condition is met. It is kinda similar to the `repeat` function in the `stream` class but with a resettable variable counter.

Dolu1990, 2022-02-14:
Added as forkSerial :)

closed 2022-02-14 by Dolu1990