SpinalHDL/SpinalHDL #609: Should Data with zero bit an error?

pwang7, 2022-02-11:
It looks like SpinalHDL allow zero bit data, such as zero bit Bits/UInt, and then SpinalHDL does not generate zero bit data in the generated Verilog code.

Sometime, it's an implicit error to define a zero bit Bits/UInt. Therefore, should SpinalHDL give an error when it encounters zero bit data?

Dolu1990, 2022-02-11:
Hi,

It is done on purpose, zero width wire can be very usefull to avoid having to manualy handle case where some configuration "require" it.

I would say, the benefit of zero wire overweight the cost.

Dolu1990, 2022-02-11:
One thing which would be still a possibility is to have a option to report zero width bits

pwang7, 2022-02-11:
> One thing which would be still a possibility is to have a option to report zero width bits

Thanks for your reply!
Having zero width bits reported would be good as well.
How to implement this?

Dolu1990, 2022-02-14:
Done ^^

So SpinalVerilog(...).printZeroWidth()

closed 2022-02-14 by Dolu1990

pwang7, 2022-02-14:
> Done ^^
> 
> So SpinalVerilog(...).printZeroWidth()

Great!