SpinalHDL/SpinalHDL #611: Assign default init value in vec of reg

ronan-lashermes, 2022-02-14:
Related to #127 .
What is the idiomatic way to init a default value from a generic data type ?
Example with a ShiftRegister :
```scala
import spinal.core._

case class ShiftRegister[T <: Data](dataType: HardType[T], depth: Int) extends Component {
  val io = new Bundle {
    val input  = in (dataType())
    val output = out(dataType())
  }

  val regs = Vec(Reg(dataType()) init **?**, depth)

  for (i <- 1 to (depth-1)) {
      regs(i) := regs(i-1)
  }

  regs(0) := io.input
  io.output := regs(depth-1)
}
```

My `dataType` is a `Flow[Something]`. Without initialization, I may have random `valid` bits set at reset.
I tried a `clone()` from an added `resetValue: T` in constructor, but it returns a `Data` and not a `T`.

PS: by the way, a shift register in the lib may be a nice thing to have.

Dolu1990, 2022-02-14:
I would say that in general : 

```scala
val regs = Vec.fill(depth)(Reg(dataType) init(xxx))
```

In your case, where the init value is a parameter i would go with : 

```scala
case class ShiftRegister[T <: Data](dataType: HardType[T], depth: Int, initFunc : T => Unit) extends Component {
val regs = Vec.fill(depth)(Reg(dataType))
regs.foreach(initFunc)
```

ronan-lashermes, 2022-02-15:
Ok, thanks. I will try to add something in the doc on Vec of Reg.

ronan-lashermes, 2022-02-25:
Actually I made a mistake when I tried your solution, and it misses something to work. (I will have to edit the added documentation).
Code example
```scala
case class ShiftRegister[T <: Data](dataType: HardType[T], depth: Int, initFunc: T => Unit) extends Component {
   val io = new Bundle {
      val input  = in (dataType())
      val output = out(dataType())
   }

   val regs = Vec.fill(depth)(Reg(dataType()))
   regs.foreach(initFunc)

   for (i <- 1 to (depth-1)) {
         regs(i) := regs(i-1)
   }

   regs(0) := io.input
   io.output := regs(depth-1)
}

object SRConsumer {
   def initIdleFlow[T <: Data](flow: Flow[T]): Unit = {
      flow.setIdle()
   }
}

class SRConsumer() extends Component {
   //...
   val sr = ShiftRegister(Flow(UInt(8 bits)), 4, SRConsumer.initIdleFlow[UInt])
}
```

This cannot work since `setIdle` is called, but not during reset. I tried
```scala
val regs = Vec.fill(depth)(Reg(dataType()))
    when(ClockDomain.current.isResetActive) {
        regs.foreach(initFunc)
    }
    .otherwise {
        for (i <- 1 to (depth-1)) {
            regs(i) := regs(i-1)
        }

        regs(0) := io.input
    }
```
but the simulation shows that does not work.
So I am still trying to find a solution for this.

Dolu1990, 2022-02-25:
I would say, 

```scala
   def initIdleFlow[T <: Data](flow: Flow[T]): Unit = {
      flow.setIdle()
   }
```

Isn't the good usage.

But instead : 

```scala
   def initIdleFlow[T <: Data](flow: Flow[T]): Unit = {
      flow.valid init(False)
   }
```

ronan-lashermes, 2022-02-25:
Ok, this is working. I am updating the docs. Thanks !

closed 2022-02-25 by ronan-lashermes

Dolu1990, 2022-02-25:
Thanks for the doc ^^