SpinalHDL/SpinalHDL #616: add io signal by component.current.getAllIO

openRV, 2022-02-23:
I tried to add signal by `component.current.getAllIO : mutable.Set[BaseType]`, with a template variable inferring BaseType like this:
``` scala
    def fn [T <: BaseType] (type:T) {
        val ios = Component.current.getAllIO
        ios += in(type)       // signal 1
        ios += out(type)    // signal 2
    }
```
and by `println(ios)`
I just found that signal 1 and signal 2 in generated ios are the same.

while changed type to such as UInt(32 bits), that just work fine, and correctly added 2 different in and out signal.
I would like to know whether it's a bug or I have some other way to do this.

I would like to add io signals from mutable Vector, is it possible?

Dolu1990, 2022-02-23:
Hi

```scala
    def fn [T <: BaseType] (type:HardType[T]) {
       in(type())       // signal 1
       out(type())    // signal 2
    }
    ```

should be good normaly

in() and out() both are modifiers to the parameter, they do not create anything.
Also they add the given arguements to the Component.current list already :)

openRV, 2022-03-02:
of course it can， but I'd like to add customized signals, which can't infer num, type or bitwidth in advance.
such as an abstract class of all pipeline regs (or maybe can I use trait to do that?)

Dolu1990, 2022-03-04:
Do you have some example of the API usage ? I'm not exactly sure of the goal you are aiming for.

openRV, 2022-03-09:
just research purpose, there is an example following
```scala

    // decalration could be:
    val PipelineReg = Pipeline( Bits(2 bits) , Int(32 bits) )
    // like a bundle of RegNext of different type

    // maybe I can get io values like this:
    val a = Bits(2 bits)
    val b = Bits(2 bits)
    // for example, just a pulse length delay
    a := PipelineReg.io.first( in )
    b := PipelineReg.io.second( out )
    // it performance like b := RegNext(a)
```

I have no idea whether it's possible

Dolu1990, 2022-03-09:
Ahhh, so you want to do some auto pipelining ?

For instance, NaxRiscv has a pipelining API : 
https://spinalhdl.github.io/NaxRiscv-Rtd/main/NaxRiscv/abstraction/index.html#pipeline

About your example, what is the "in" and "out" in the last part ? 
PipelineReg.io.first( in )
PipelineReg.io.second( out )

 ?

openRV, 2022-03-09:
yes, that's just what I want.  Thank you so much.

I'm not quite sure about that, but maybe a symbol that can help me get the input or output signal of the auto generated Pipeline module? If it's possible to do this in this way or something else?

Dolu1990, 2022-03-11:
So, one thing, is that when i was doing the pipeline API for VexRiscv and Nax, i avoided using Component to encapsulate the pipeline.

Instead i was using a Area, that way it provide a namespace while preserving full access to the signal (component can be cumbersome)

openRV, 2022-03-16:
Ok, I've learned that. But if there is any document of such inner component like area in SpinalHDL ?

Dolu1990, 2022-03-16:
Sure : 
https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Structuring/area.html

openRV, 2022-03-23:
I see , thank you so much

closed 2022-03-23 by openRV