SpinalHDL/SpinalHDL #617: mem.bin  issue

jijingg, 2022-02-23:
1.  path naming bug on Windows
Linux/Windows Separator "/" misusing  when generator
```verilog
  initial begin
    $readmemb("D:\nutstore\xxxx\.\simWorkspace\Top\rtl/Top.v_toplevel_uMain_uTwiddleGroup_uWRom_roms_1_rom_7_mem.bin",mem);
  end
```
lead error :
```
%Error: D:/nutstore/Gitlab/proj/macLab/tmp/job_1/Top.v:12209:15: Unknown escape sequence: \G
12209 |     $readmemb("D:\nutstore\Gitlab\proj\macLab\.\simWorkspace\Top\rtl/Top.v_toplevel_uMain_uTwiddleGroup_uWRom_roms_5_rom_7_mem.bin",mem);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~
```

2.  mem.bin  ‘s naming 
as we talk in gitter ,  current  naming of mem.bin with instance hierarchy ,  `top.v_toplevel_uMain_xxxxxxxxxRom_mem.bin`,
i think the the Component name is enough,  Otherwise, instantiation of the same ROM in different hierarchies will result in the generation of duplicate rom.bin files

Dolu1990, 2022-02-24:
>  the Component name is enough

The issue is that you can have the rom having a different content depending component parameters.

The only proper way to handle it (i think), is to check for a previously emited .bin which has the same content, and refer to it if found.

Dolu1990, 2022-02-24:
I pushed a fix, 

Here is a example :

```scala
  SpinalConfig(targetDirectory = "rawrrr", romReuse = true).generateVerilog(new Component{
    val romInit1 = (0 to 7).map(BigInt(_))
    val romInit2 = (0 to 7).map(e => BigInt(e+1))
    val romA = Mem.fill(8)(UInt(8 bits)) initBigInt(romInit1)
    val romB = Mem.fill(8)(UInt(8 bits)) initBigInt(romInit2)
    val romC = Mem.fill(8)(UInt(8 bits)) initBigInt(romInit1)
    val romD = Mem.fill(8)(UInt(8 bits)) initBigInt(romInit1)
    val romE = Mem.fill(8)(UInt(8 bits)) initBigInt(romInit1)

    val x = slave(romD.writePortWithMask(4))
    val y = slave(romD.writePortWithMask(4))
  }
```

Note that romReuse is disabled by default, i'm not sure if it should be enabled by default, as it may be surprising that sometime some rom are generated in .bin, sometime not.

Any opinon ?

commit 854ab2da (2022-02-24): Merge pull request #618 from jijingg/membin

fix #617 of OS seperator char issue
commit af555adc (2022-02-24): fix #617 of OS seperator char issue