SpinalHDL/SpinalHDL #619: Contribution checklist

Dolu1990, 2022-02-24:
Contributor Checklist

- ￼[ ] Did you add appropriate documentation in docs/src?
- ￼[ ] Did you state the API impact?
- [ ]￼ Did you specify the code generation impact?
- ￼[ ] Did you request a desired merge strategy?
- [ ]￼ Did you add text to be included in the Release

Readon, 2022-02-24:
what about the test code, it seems also important to avoid bugs.

Dolu1990, 2022-02-24:
@Readon added ^^

Readon, 2022-02-24:
> @Readon added ^^

TDD is fantastic. ^^

numero-744, 2022-10-15:
What about replacing this pinned issue by a [PR template?](https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/creating-a-pull-request-template-for-your-repository)

numero-744, 2022-10-17:
Agreed for the CONTRIBUTOR file, to help / guide contributors, it would also describe file structure (#902) and other things?

A PR template could also be helpful for both contributors and maintainers, to really act as a checklist.

I'll open a PR for both CONTRIBUTOR and PR template.

PR #909 (2022-11-14): docs: add help for contributors
Closes #619 and #902
PR #964 (2022-11-21): fix: StreamFragmentWidthAdapter(x, y, true) widthOf(y) = 2 * widthOf(x)
Root cause is that counter is in range [0,1], number of elements in buffer Vector is 1, thus the number of bits differs in whenIndexed(), only for the case of widthOf(y) = 2 * widthOf(x). Triggering assert in whenIndexed().

Closes #963

# Context, Motivation & Description

# Impact on code generation

# Checklist

- [x] Unit tests were added
- [x] API changes are or will be documented:
  - using Scaladoc comments: `/** */`?
  - on [RTD](https://github.com/SpinalHDL/SpinalDoc-RTD)?
  - thanks to a [new tracking issue on RT [... cut]
PR #974 (2022-11-23): fix cocotb problem while runing on system with different language.

Closes #973 

# Context, Motivation & Description

fix cocotb charset error on different language environment.

# Impact on code generation

# Checklist

- [x] Unit tests were added // not available.
- [x] API changes are or will be documented:  // not needed.
  - using Scaladoc comments: `/** */`?
  - on [RTD](https://github.com/SpinalHDL/SpinalDoc-RTD)?
  - thanks to a [new tracking issue on RTD](https://github.com/SpinalHDL/SpinalDoc-RTD/issues/new?title=Document%20XXX&body=Do%20not%20merge%20until%20SpinalHDL/SpinalHDL%23X [... cut]
PR #969 (2022-11-24): Add support for mill build.

Closes #878, partially.

# Context, Motivation & Description

Support build SpinalHDL by Mill, it could compile/test/publishLocal the existing modules.
Build through mill can be much faster than sbt, and sbt would be really unavailable sometimes due to network problem.

#### compile library
mill __.compile

#### test library
mill __.test

#### run individual app to play
mill tester.runMain spinal.xxxxx.xxxxx

#### publish to local, work as dev version
mill __.publishLocal

This might be the first step to create a standalone working environment on Windows.
The secon [... cut]
closed 2022-11-14 by Dolu1990