SpinalHDL/SpinalHDL #625: how to pull internal signal to another module for debug report?

dxsm, 2022-02-28:
[image]

Hi,
As above, B module is used for gather all the internal signals to group or add some simple logic, then report to software. How to pull the  internal sig1/sig2 from module A1/A2 to module B, then do some simple logic?

Dolu1990, 2022-02-28:
Hi,

You can use the pull everywhere for any kind of pulling, including from inside B to pull A1/A2 stuff, the only thing is that you need the reference to the signal you want to pull, and be in the B context

```scala
//Inside B
refToAA1.pull()
```

dxsm, 2022-02-28:
Thanks,but I tried 
```scala
class A extends Module {
  val sig1 = UInt(16 bits)
  sig1 := 0
}

//class B extends Module {
//  val sig2 = UInt(16 bits)
//  sig2 := 0
//}

class top extends Module {
  var ua = new A()
//  var ub = new B()

  var ua_sig1 = ua.sig1.pull()
}
```
generates
```verilog
module top (
);

  A ua (
  );

endmodule

module A (
);

  wire       [15:0]   sig1;

  assign sig1 = 16'h0;

endmodule
```

1. Why signal `ua_sig1` does not generate?
2. Module A and B has the same level, does that mean I can not pull signal inside A to B context?

Dolu1990, 2022-02-28:
> Why signal ua_sig1 does not generate?

Because by default the pull function pull things with an anonymus name, which mean that if the wire is never used, it will be removed from the netlist. Try to add a     val x = CombInit(ua_sig1) in the toplevel, there you should get a better visibility

> Module A and B has the same level, does that mean I can not pull signal inside A to B context?

You can :

```scala
    class A extends Module {
      val sig1 = UInt(16 bits)
      sig1 := 0
    }

    class B(ref : UInt) extends Module {
      val sig2 = ref.pull()
    }

    var ua = new A()
    var ub = new B(ua.sig1)
```

dxsm, 2022-03-02:
Thanks, it works.

closed 2022-03-02 by dxsm