SpinalHDL/SpinalHDL #627: Methods can only be called from instance of Master should be deal with explicitly.

Readon, 2022-03-01:
There are functions that should called only from a master, such as the toAxi4ReadOnly for Axi4 and m2sPipe for Stream.
How about add a check just like assert(this.isMaster) to those method.
or explicitly add the state MasterOnly to the Functions definition tables in the Documentation?

As the first time I use functions of stream and buses there always be a mistake to call above functions from slave instance.

Dolu1990, 2022-03-01:
> There are functions that should called only from a master,

It depend what you mean by master :

- if you mean a component master interface
- a interface inside a component which controle another interface

And in both cases, those method can be used. (So not only for io masters)

> How about add a check just like assert(this.isMaster) to those method.

For the reason above it shouldn't be enforced

> or explicitly add the state MasterOnly to the Functions definition tables in the Documentation?

Now in general, i'm using the naming : 

- toXxxx when you need to call it from a master (toAxi, ..)
- fromXxx when you need to call it from a slave (fromAxi)

Overall it is very rare to have functions that you need to call on a slave, right ?

> or explicitly add the state MasterOnly to the Functions definition tables in the Documentation?

Depending what is defined by master, this can be confusing too. (see the other issue)

maybe some general explenation of the paradigme in the doc would be better ?

Readon, 2022-03-01:
> It depend what you mean by master :
>
> * if you mean a component master interface
> * a interface inside a component which controle another interface
>
> And in both cases, those method can be used. (So [...]

Now I understand the naming policy of them. 

> 
> > or explicitly add the state MasterOnly to the Functions definition tables in the Documentation?
> 
> Depending what is defined by master, this can be confusing too. (see the other issue)
> 
> maybe some general explenation of the paradigme in the doc would be better ?

IMO, an explanation in doc would be good and enough.

Dolu1990, 2022-03-04:
Do you feel you can document it ?
Maybe around https://github.com/SpinalHDL/SpinalDoc-RTD/blob/ee599f6a529168ed2b6f9868a0403afb6c5a981a/source/SpinalHDL/Data%20types/bundle.rst#masterslave ?

Where do you think it would be the best ?

Readon, 2022-03-04:
> Do you feel you can document it ? Maybe around https://github.com/SpinalHDL/SpinalDoc-RTD/blob/ee599f6a529168ed2b6f9868a0403afb6c5a981a/source/SpinalHDL/Data%20types/bundle.rst#masterslave ?
> 
> Where do you think it would be the best ?

I have submitted the docs in the repos. It's time to move to that PR.

closed 2022-03-04 by Readon