SpinalHDL/SpinalHDL #628: MaskMapping Bug?

kleinai, 2022-03-01:
https://github.com/SpinalHDL/SpinalHDL/blob/2ab339a36cfd2d3f8b902e1254dfc65d895e7a42/lib/src/main/scala/spinal/lib/bus/misc/Misc.scala#L50-L55

Is `(address & base) === mask` correct here? Shouldn't the inverted mask be applied then compared to the base address?

Furthermore, `removeOffset` will not work because inverting `mask` directly will result in a negative number (I tried it myself).

This works as a possible fix:
```
override def hit(address: UInt): Bool = (address & ~U(mask).resize(address.getWidth)) === base
```

Although I'm sure there's a more elegant solution.

Dolu1990, 2022-03-01:
> Is (address & base) === mask correct here? Shouldn't the inverted mask be applied then compared to the base address?

You are right, it should be inverted XD
Good finding ^^

> inverting mask directly will result in a negative number 

Ahhh, right

> Although I'm sure there's a more elegant solution.

Better to directly create the literal with the correct bit width :)

```scala
override def hit(address: UInt): Bool = (address & ~U(mask, widthOf(address) bits)) === base
```

Dolu1990, 2022-03-01:
Will you create a pull request ?

PR #630 (2022-03-03): Correct MaskMapping hit() function
Fixes #628
closed 2022-03-03 by Dolu1990