SpinalHDL/SpinalHDL #629: IVerilog simulation assertion failure

piegamesde, 2022-03-01:
Because of #607, I thought about exchanging Vrilator for IVerilog and see if it helps. Adding `simConfig.withIVerilog` and then running it again gave me the following:

```scala
[Progress] IVerilog compilation started
[Progress] IVerilog compilation done in 25.691 ms
Exception in thread "main" java.lang.AssertionError: assertion failed: Analyze step of verilog files failed
	at scala.Predef$.assert(Predef.scala:279)
	at spinal.sim.IVerilogBackend.analyzeRTL(VpiBackend.scala:445)
	at spinal.sim.VpiBackend.delayed_compilation$lzycompute(VpiBackend.scala:120)
	at spinal.sim.VpiBackend.delayed_compilation(VpiBackend.scala:76)
	at spinal.sim.VpiBackend.instanciate_(VpiBackend.scala:133)
	at spinal.sim.VpiBackend.instanciate(VpiBackend.scala:156)
	at spinal.sim.VpiBackend.instanciate(VpiBackend.scala:168)
	at spinal.sim.SimVpi.<init>(SimVpi.scala:15)
	at spinal.core.sim.SpinalSimConfig$$anon$8.newSimRaw(SimBootstraps.scala:758)
	at spinal.core.sim.SimCompiled.doSimApi(SimBootstraps.scala:412)
	at spinal.core.sim.SimCompiled.doSim(SimBootstraps.scala:393)
```

Note that with Verilator the same code runs just fine. Either way, the error message does not tell me what exactly is wrong with the code so I can't really debug it.

piegamesde, 2022-03-03:
The relevant code appears to be https://github.com/SpinalHDL/SpinalHDL/blob/dev/sim/src/main/scala/spinal/sim/VpiBackend.scala#L434-L445

We should capture the output of the spawned process and print it in case of an error. I also found this comment while browsing the code: https://github.com/SpinalHDL/SpinalHDL/blob/dev/sim/src/main/scala/spinal/sim/VpiBackend.scala#L324-L328

piegamesde, 2022-03-03:
I tried the GHDL backend, and the error messages are similarly helpful:

```
Exception in thread "main" java.lang.AssertionError: assertion failed: Analyze step of vhdl files failed
	at scala.Predef$.assert(Predef.scala:279)
	at spinal.sim.GhdlBackend.analyzeRTL(VpiBackend.scala:254)
	at spinal.sim.VpiBackend.delayed_compilation$lzycompute(VpiBackend.scala:120)
	at spinal.sim.VpiBackend.delayed_compilation(VpiBackend.scala:76)
	at spinal.sim.VpiBackend.instanciate_(VpiBackend.scala:133)
	at spinal.sim.VpiBackend.instanciate(VpiBackend.scala:156)
	at spinal.sim.VpiBackend.instanciate(VpiBackend.scala:168)
	at spinal.sim.SimVpi.<init>(SimVpi.scala:15)
	at spinal.core.sim.SpinalSimConfig$$anon$7.newSimRaw(SimBootstraps.scala:732)
	at spinal.core.sim.SimCompiled.doSimApi(SimBootstraps.scala:412)
	at spinal.core.sim.SimCompiled.doSim(SimBootstraps.scala:393)
	at watca.HSMSim$.main(HSMSim.scala:208)
	at watca.HSMSim.main(HSMSim.scala)
```

When a dependency is missing/cannot be found:

```
Exception in thread "main" java.lang.AssertionError: assertion failed: Compilation of VpiPlugin.o failed
	at scala.Predef$.assert(Predef.scala:279)
	at spinal.sim.GhdlBackend.compileVPI(VpiBackend.scala:228)
	at spinal.sim.VpiBackend.delayed_compilation$lzycompute(VpiBackend.scala:119)
	at spinal.sim.VpiBackend.delayed_compilation(VpiBackend.scala:76)
	at spinal.sim.VpiBackend.instanciate_(VpiBackend.scala:133)
	at spinal.sim.VpiBackend.instanciate(VpiBackend.scala:156)
	at spinal.sim.VpiBackend.instanciate(VpiBackend.scala:168)
	at spinal.sim.SimVpi.<init>(SimVpi.scala:15)
	at spinal.core.sim.SpinalSimConfig$$anon$7.newSimRaw(SimBootstraps.scala:732)
	at spinal.core.sim.SimCompiled.doSimApi(SimBootstraps.scala:412)
	at spinal.core.sim.SimCompiled.doSim(SimBootstraps.scala:393)
	at watca.HSMSim$.main(HSMSim.scala:208)
	at watca.HSMSim.main(HSMSim.scala)
```

Dolu1990, 2022-03-04:
Hi ^^

I just pushed a change in dev which will print all messages from ghdl/iverilog if there is a failure, this should help. Can you give a try ? to understand why it fail on your setup.

piegamesde, 2022-03-07:
Hm, I tried out 3d31d28725b0d14fe751ee8f260830ebc32707b0 and without changing anything to my code, I get:

```
Exception in thread "main" java.lang.NoClassDefFoundError: scala/Serializable
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:756)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:468)
	at java.net.URLClassLoader.access$100(URLClassLoader.java:74)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:369)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:363)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:362)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:418)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:352)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:351)
	at spinal.core.sim.package$.SimConfig(package.scala:37)
Caused by: java.lang.ClassNotFoundException: scala.Serializable
	at java.net.URLClassLoader.findClass(URLClassLoader.java:382)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:418)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:352)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:351)
	... 15 more
```

Although I should note that besaid commit is from `dev`, and usually only the `master` branch works for me for some reason.

Dolu1990, 2022-03-09:
How did you used the dev branch ? directly form sources ? or you did a publishLocal ?

ypmen, 2022-03-17:
Hi. I meet the same error with GHDL. Please see below my example,
counter.scala:
```
package mylib

import spinal.core._
import spinal.lib._
import spinal.core.sim._

class counter() extends BlackBox {
    val io = new Bundle {
        val clk = in Bool()
        val rst = in Bool()
        val cout = out Bits(8 bits)
    }

    noIoPrefix()

    mapCurrentClockDomain(io.clk, io.rst)
}

case class counter_top() extends Component {
    val io = new Bundle {
        val cout = out Bits(8 bits)
    }

    noIoPrefix()

    val counter_inst = new counter()

    io.cout <> counter_inst.io.cout
}

object CounterTester {
    def main(args: Array[String]): Unit = {

        val compiled = SimConfig.withGhdl.withWave.addRtl("/home/ypmen/vscode/spinalhdl/blackbox_vhdl/counter.vhd").compile(counter_top())
        //val compiled = SimConfig.withWave.addRtl("counter.v").compile(counter_top())
        
        compiled.doSim{ dut =>
            dut.clockDomain.forkStimulus(period = 10)

            for (i <- 0 until 10) {
                dut.clockDomain.waitSampling()
                println(dut.io.cout.toInt)
            }
        }
    }
}
```
counter.vhd:
```
library ieee;
    use ieee.std_logic_1164.all;
    use ieee.std_logic_unsigned.all;

entity counter is
    port (
        clk  :in  std_logic;                      -- Input clock
        rst  :in  std_logic;                       -- Input reset
        cout  :out std_logic_vector (7 downto 0)  -- Output of the counter
    );
end entity;

architecture rtl of counter is
    signal count :std_logic_vector (7 downto 0);
begin
    process (clk, rst) begin
        if (rst = '1') then
            count <= (others=>'0');
        elsif (rising_edge(clk)) then
                count <= count + 1;
        end if;
    end process;
    cout <= count;
end architecture;
```
build.sbt:
```
ThisBuild / version := "1.0"
ThisBuild / scalaVersion := "2.13.8"
ThisBuild / organization := "org.example"

val spinalVersion = "1.6.4"
val spinalCore = "com.github.spinalhdl" %% "spinalhdl-core" % spinalVersion
val spinalLib = "com.github.spinalhdl" %% "spinalhdl-lib" % spinalVersion
val spinalIdslPlugin = compilerPlugin("com.github.spinalhdl" %% "spinalhdl-idsl-plugin" % spinalVersion)

lazy val mylib = (project in file("."))
  .settings(
    name := "counter_top",
   libraryDependencies ++= Seq(
      spinalCore, 
      spinalLib, 
      spinalIdslPlugin
     )
  )

fork := true
```

Dolu1990, 2022-03-21:
@ypmen Your issue was a bit different, it was due to GHDL not having the -fsynopsys option (to enable std_logic_unsigned)

tanhongze, 2022-03-26:
Does anybody know the reason of this message? How to solve this problem?
```
[error] Exception in thread "main" spinal.sim.VpiException: Attempt to access an already closed simulation
[error]         at spinal.sim.vpi.JNISharedMemIfaceJNI.SharedMemIface_close(Native Method)
[error]         at spinal.sim.vpi.SharedMemIface.close(SharedMemIface.java:124)
[error]         at spinal.sim.SimVpi.end(SimVpi.scala:145)
[error]         at spinal.sim.SimManager.runWhile(SimManager.scala:334)
[error]         at spinal.sim.SimManager.run(SimManager.scala:227)
[error]         at spinal.core.sim.SimCompiled.doSimApi(SimBootstraps.scala:433)
[error]         at spinal.core.sim.SimCompiled.doSim(SimBootstraps.scala:389)
[error]         at spinal.core.sim.SpinalSimConfig.doSim(SimBootstraps.scala:613)
[error]         at centaur.test.MyTopLevelSim$.main(MyTopLevelSim.scala:14)
[error]         at centaur.test.MyTopLevelSim.main(MyTopLevelSim.scala)
[error] Nonzero exit code returned from runner: 1
```

tanhongze, 2022-03-26:
I tried to run my code with IVerilog, got a excetion like this.

tanhongze, 2022-03-26:
Well, I found the problem by myself.
My code triggered a assertion with `FAILURE` severity.
And it just stoped without print the error message.

So, why it can't print its error message?

piegamesde, 2022-05-23:
iVerilog:

```
 No such file or directoryerror: Unable to find the root module "__simulation_def" in the Verilog source.     : Perhaps ``-s __simulation_def'' is incorrect?error: Unable to find the root module "HSMRiscv" in the Verilog source.     : Perhaps ``-s HSMRiscv'' is incorrect?2 error(s) during elaboration.
 ```

The first error is clearly caused by a file path containing whitespace characters, and a lack of escaping thereof. I tried GHDL and it ran into the same issue.